espressif/esp-iot-solution #322: LEDC_FADE_TOO_FAST

ThatBigPrint, 2023-12-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF 5.1.1

### Espressif SoC revision.

Esp32 pico d4

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom (confirmed working hardware

### Steps to reproduce.

Just running the lightbulb example... maybe its related to the esp32 needing to wait for tasks i have tried to use both hardware timers on and off same result 

### Build Logs.

```plain
this is the error when we try to change state interestingly it seems once its gets past the first few seconds of booting it works more reliable but as soon as the LEDC_FADE_TOO_FAST shows in the logs its no longer responsive and the light needs to be power cycled 

W (9410) Mesh/Firebase: Path: /
I (9410) lightbulb: lightbulb_set_switch will update on/off status: on -> off
W (9419) ledc: LEDC FADE TOO FAST
I (9422) Mesh/Firebase: power: 0
I (11252) Mesh/Firebase: DATA=event: patch
data: {"path":"/","data":{"power":1}}

W (11253) Mesh/Firebase: Path: /
I (11254) lightbulb: lightbulb_set_switch will update on/off status: off -> on
I (11262) lightbulb: set cct:0 brightness:100
I (11267) lightbulb: convert cold:0 warm:255
I (11271) lightbulb: hal write value [white1:0 white2:255], channel_mask:24 fade_ms:1000
E (11304) hal_manage: hal_set_channel_group(868): Can't get mutex
E (11304) lightbulb: lightbulb_set_cctb(1042): set hal channel group fail
I (11307) Mesh/Firebase: power: 1
I (11451) Mesh/Firebase: System information, channel: 11, layer: 1, self mac: 4c:75:25:a3:88:e4, parent bssid: c4:e5:32:85:a7:e9, parent rssi: -71, free heap: 110920
I (11456) Mesh/Firebase: child node number: 0
I (12175) Mesh/Firebase: DATA=event: patch
data: {"path":"/","data":{"power":0}}

W (12176) Mesh/Firebase: Path: /
I (12177) lightbulb: lightbulb_set_switch will update on/off status: on -> off
E (12209) hal_manage: hal_set_channel_group(868): Can't get mutex
I (12209) Mesh/Firebase: power: 0
I (13404) Mesh/Firebase: DATA=event: patch
data: {"path":"/","data":{"power":1}}

W (13406) Mesh/Firebase: Path: /
I (13406) lightbulb: lightbulb_set_switch will update on/off status: off -> on
I (13414) lightbulb: set cct:0 brightness:100
I (13419) lightbulb: convert cold:0 warm:255
I (13424) lightbulb: hal write value [white1:0 white2:255], channel_mask:24 fade_ms:1000
E (13457) hal_manage: hal_set_channel_group(868): Can't get mutex
E (13457) lightbulb: lightbulb_set_cctb(1042): set hal channel group fail
I (13460) Mesh/Firebase: power: 1
I (14324) Mesh/Firebase: DATA=event: patch
data: {"path":"/","data":{"power":0}}

W (14325) Mesh/Firebase: Path: /
I (14326) lightbulb: lightbulb_set_switch will update on/off status: on -> off
E (14358) hal_manage: hal_set_channel_group(868): Can't get mutex
I (14358) Mesh/Firebase: power: 0
```

shixinke-orion, 2024-01-02:
What frequency did you set for the PWM? Can you try disable the hardware fade and test again?
```
(Top) → Component config → LightBulb Driver Config → Drivers
[*] Enable PWM interface
[ ]     Use hardware fade API
...
...
```

ThatBigPrint, 2024-01-02:
I'm only using it at 150hz and I have tried that

shixinke-orion, 2024-01-03:
This is a bit strange, as I can't reproduce the issue in my environment. It might be related to the application code. Could you provide a debug-level log?

ThatBigPrint, 2024-01-03:
I'm in airport at the min... Is it true on the esp32 including the d4 I
can't call other code until the fade is finished? Or can I pin this to
separate core? I've noticed sometimes dependant on connection time it
interups JSON parsing