espressif/esp-iot-solution #324: Inquiry and Suggestion: Driving a Buzzer with ESP-IDF and ESP IoT Solution

hayschan, 2024-01-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello Espressif Team,

I am working on a project using the ESP-IDF and the ESP IoT Solution, and I have encountered a challenge in driving a buzzer. After reviewing the ESP IoT Solution repository, I noticed that there isn't a specific component library for buzzers. The closest related library seems to be the LED indicator library.

I have a few questions and a suggestion:

1. **Recommended Way to Drive a Buzzer in ESP-IDF:** 
   Could you advise on the recommended method to drive a buzzer using ESP-IDF? Specifically, guidance on settings for the divider, clock, duty cycle, and melody pitch which is related to frequency would be highly beneficial.

2. **LED Indicator Component Library for Buzzers:** 
   Is it possible to use the LED indicator component library for driving a buzzer? If so, could you provide some insights or examples on how to adapt it for this purpose?

3. **Suggestion - Buzzer Component in ESP IoT Solution:**
   Considering the common use of buzzers in IoT applications, it would be great to have a dedicated component library for buzzers in the ESP IoT Solution. This addition could simplify the development process for many users who intend to use buzzers in their projects.

I appreciate your time and support in this matter and look forward to your guidance and insights.

Best Regards,
Hays Chan

hayschan, 2024-01-16:
My current method of driving a buzzer is using the `LEDC` API.

lijunru-hub, 2024-01-16:
Hi Hayschan,

The buzzer indication in this application does have some similarities with the LED indicator. It allows the buzzer to produce long and short beeps, double beeps, triple beeps, etc.

Below, I will provide instructions on how to use the LED indicator for two types of buzzers.

Active Buzzer: It beeps when powered. You can use the GPIO driver in the LED indicator, set it as a GPIO indicator light.

Example:

```c
static const blink_step_t double_beeps[] = {
    {LED_BLINK_HOLD, LED_STATE_ON, 500},
    {LED_BLINK_HOLD, LED_STATE_OFF, 500},
    {LED_BLINK_HOLD, LED_STATE_ON, 500},
    {LED_BLINK_HOLD, LED_STATE_OFF, 500},
    {LED_BLINK_STOP, 0, 0},
};
```

Passive Buzzer: You need to use the LED indicator's LEDC driver, but currently, the LEDC frequency does not support dynamic modification. It needs to be fixed in the menuconfig. If you don't need buzzer music, I believe this should be sufficient.

Example:

```c
static const blink_step_t double_beeps[] = {
    {LED_BLINK_BRIGHTNESS, LED_STATE_50_PERCENT, 500},
    {LED_BLINK_HOLD, LED_STATE_OFF, 500},
    {LED_BLINK_BRIGHTNESS, LED_STATE_50_PERCENT, 500},
    {LED_BLINK_HOLD, LED_STATE_OFF, 500},
    {LED_BLINK_STOP, 0, 0},
};
```

Certainly, examples and component documentation will be provided later. If you have any questions, feel free to ask.

closed 2024-01-25 by leeebo