espressif/esp-iot-solution #325: 120x160摄像头只能看到部分画面

jeff-getlucky, 2024-01-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

160x120的摄像头是正常的，包括http stream和lcd显示。
但120x160只能显示一点点内容。
用demo获取的截图
[image]
原图如下（！！！用电脑网页看都是空白，下载到本地或者使用手机浏览才能看到图片。！！！）
capture 1
[image]
capture 2
[image]
capture 3
[image]

这里是一些日志
I (5888) USB_STREAM: Recover Stream Task
I (5917) USB_STREAM: Action: ACTION_PIPE_DFLT_DISABLE
I (5917) USB_STREAM: Action: ACTION_PORT_RECOVER
I (5917) USB_STREAM: Action: ACTION_DEVICE_DISCONN
I (5918) USB_STREAM: Waiting USB Connection
I (6179) USB_STREAM: line 151 HCD_PORT_EVENT_CONNECTION
I (6180) USB_STREAM: Action: ACTION_DEVICE_CONNECT
I (6230) USB_STREAM: Resetting Port
I (6290) USB_STREAM: Setting Port FIFO, 1
I (6291) USB_STREAM: USB Speed: full-speed
I (6291) USB_STREAM: ENUM Stage START, Succeed
I (6292) USB_STREAM: ENUM Stage GET_SHORT_DEV_DESC, Succeed
I (6304) USB_STREAM: Default pipe endpoint MPS update to 64
I (6304) USB_STREAM: ENUM Stage CHECK_SHORT_DEV_DESC, Succeed
I (6306) USB_STREAM: ENUM Stage SET_ADDR, Succeed
I (6317) USB_STREAM: ENUM Stage CHECK_ADDR, Succeed
I (6318) USB_STREAM: ENUM Stage GET_FULL_DEV_DESC, Succeed
*** Device descriptor ***
bcdUSB 2.00
bDeviceClass 0xef
bDeviceSubClass 0x2
bDeviceProtocol 0x1
bMaxPacketSize0 64
idVendor 0x2bdf
idProduct 0x101
bNumConfigurations 1
I (6320) USB_STREAM: ENUM Stage CHECK_FULL_DEV_DESC, Succeed
I (6321) USB_STREAM: ENUM Stage GET_SHORT_CONFIG_DESC, Succeed
I (6322) USB_STREAM: ENUM Stage CHECK_SHORT_CONFIG_DESC, Succeed
I (6323) USB_STREAM: ENUM Stage GET_FULL_CONFIG_DESC, Succeed
*** Configuration descriptor ***
wTotalLength 433
bNumInterfaces 2
bConfigurationValue 1
*** Interface Association Descriptor: Video ***
        *** Interface descriptor ***
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x1
                *** Endpoint descriptor ***
                bEndpointAddress 0x83   EP 3 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 16
                bInterval 8
        *** Interface descriptor ***
        bInterfaceNumber 1
        bAlternateSetting 0
[... 67 lines cut ...]
I (7559) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 18, width = 120, height = 160, length = 1022, ptr = 0
I (7583) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 19, width = 120, height = 160, length = 1022, ptr = 0
I (7592) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 20, width = 120, height = 160, length = 1022, ptr = 0
I (7601) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 21, width = 120, height = 160, length = 1022, ptr = 0
I (7623) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 22, width = 120, height = 160, length = 1022, ptr = 0
I (7632) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 23, width = 120, height = 160, length = 1022, ptr = 0
I (7641) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 24, width = 120, height = 160, length = 1022, ptr = 0
I (7664) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 25, width = 120, height = 160, length = 1022, ptr = 0
I (7673) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 26, width = 120, height = 160, length = 1022, ptr = 0
I (7682) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 27, width = 120, height = 160, length = 1022, ptr = 0
I (7705) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 28, width = 120, height = 160, length = 1022, ptr = 0
I (7714) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 29, width = 120, height = 160, length = 1022, ptr = 0
I (7723) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 30, width = 120, height = 160, length = 1022, ptr = 0
I (7747) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 31, width = 120, height = 160, length = 1022, ptr = 0
I (7755) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 32, width = 120, height = 160, length = 1022, ptr = 0
I (7778) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 33, width = 120, height = 160, length = 1022, ptr = 0
I (7787) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 34, width = 120, height = 160, length = 1022, ptr = 0
I (7796) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 35, width = 120, height = 160, length = 1022, ptr = 0
I (7824) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 36, width = 120, height = 160, length = 1022, ptr = 0
I (7832) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 37, width = 120, height = 160, length = 1022, ptr = 0
I (7864) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 38, width = 120, height = 160, length = 1022, ptr = 0
I (7904) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 39, width = 120, height = 160, length = 1022, ptr = 0

我发现他的length比160x120小，正常数据应该是3000左右，而他是固定的1022。
感谢帮助

jeff-getlucky, 2024-01-18:
[image]
手机上能看到图片

jeff-getlucky, 2024-01-18:
@leeebo 巨佬方便的时候帮我看看

jeff-getlucky, 2024-01-18:
> 看起来是摄像头问题，一帧图像有点过于小了。切个大一点的分辨率看看。

尝试了240x360和640x360  都是1022长度  奇了怪了

I (25474) USB_STREAM: line 151 HCD_PORT_EVENT_CONNECTION
I (25475) USB_STREAM: Action: ACTION_DEVICE_CONNECT
I (25525) USB_STREAM: Resetting Port
I (25585) USB_STREAM: Setting Port FIFO, 1
I (25585) USB_STREAM: USB Speed: full-speed
I (25586) USB_STREAM: ENUM Stage START, Succeed
I (25587) USB_STREAM: ENUM Stage GET_SHORT_DEV_DESC, Succeed
I (25589) USB_STREAM: Default pipe endpoint MPS update to 64
I (25589) USB_STREAM: ENUM Stage CHECK_SHORT_DEV_DESC, Succeed
I (25591) USB_STREAM: ENUM Stage SET_ADDR, Succeed
I (25602) USB_STREAM: ENUM Stage CHECK_ADDR, Succeed
I (25603) USB_STREAM: ENUM Stage GET_FULL_DEV_DESC, Succeed
*** Device descriptor ***
bcdUSB 2.00
bDeviceClass 0xef
bDeviceSubClass 0x2
bDeviceProtocol 0x1
bMaxPacketSize0 64
idVendor 0x2bdf
idProduct 0x101
bNumConfigurations 1
I (25616) USB_STREAM: ENUM Stage CHECK_FULL_DEV_DESC, Succeed
I (25617) USB_STREAM: ENUM Stage GET_SHORT_CONFIG_DESC, Succeed
I (25618) USB_STREAM: ENUM Stage CHECK_SHORT_CONFIG_DESC, Succeed
I (25619) USB_STREAM: ENUM Stage GET_FULL_CONFIG_DESC, Succeed
*** Configuration descriptor ***
wTotalLength 433
bNumInterfaces 2
bConfigurationValue 1
*** Interface Association Descriptor: Video ***
        *** Interface descriptor ***
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x1
                *** Endpoint descriptor ***
                bEndpointAddress 0x83   EP 3 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 16
                bInterval 8
        *** Interface descriptor ***
        bInterfaceNumber 1
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x2
        *** Class-specific VS Interface Descriptor ***
        bNumFormats 3
        *** VS Format MJPEG Descriptor ***
        bFormatIndex 0x2
        bNumFrameDescriptors 3
        bDefaultFrameIndex 1
        *** VS MJPEG Frame Descriptor ***
        bFrameIndex 0x1
        wWidth 120
        wHeigh 160
        FrameInterval[0] 400000
        *** VS MJPEG Frame Descriptor ***
        bFrameIndex 0x2
        wWidth 240
        wHeigh 320
        FrameInterval[0] 333333
        *** VS MJPEG Frame Descriptor ***
        bFrameIndex 0x3
        wWidth 640
[... 30 lines cut ...]
I (25792) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 3, width = 640, height = 360, length = 1022, ptr = 0
I (25815) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 4, width = 640, height = 360, length = 1022, ptr = 0
I (26436) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 5, width = 640, height = 360, length = 1022, ptr = 0
I (26459) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 6, width = 640, height = 360, length = 1022, ptr = 0
I (26468) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 7, width = 640, height = 360, length = 1022, ptr = 0
I (26492) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 8, width = 640, height = 360, length = 1022, ptr = 0
I (26500) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 9, width = 640, height = 360, length = 1022, ptr = 0
I (26509) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 10, width = 640, height = 360, length = 1022, ptr = 0
I (26533) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 11, width = 640, height = 360, length = 1022, ptr = 0
I (26542) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 12, width = 640, height = 360, length = 1022, ptr = 0
I (26550) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 13, width = 640, height = 360, length = 1022, ptr = 0
I (26578) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 14, width = 640, height = 360, length = 1022, ptr = 0
I (26587) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 15, width = 640, height = 360, length = 1022, ptr = 0
I (26595) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 16, width = 640, height = 360, length = 1022, ptr = 0
I (26620) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 17, width = 640, height = 360, length = 1022, ptr = 0
I (26628) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 18, width = 640, height = 360, length = 1022, ptr = 0
I (26653) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 19, width = 640, height = 360, length = 1022, ptr = 0
I (26661) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 20, width = 640, height = 360, length = 1022, ptr = 0
I (26670) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 21, width = 640, height = 360, length = 1022, ptr = 0
I (26678) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 22, width = 640, height = 360, length = 1022, ptr = 0
I (26700) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 23, width = 640, height = 360, length = 1022, ptr = 0
I (26709) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 24, width = 640, height = 360, length = 1022, ptr = 0
I (26732) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 25, width = 640, height = 360, length = 1022, ptr = 0
I (26741) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 26, width = 640, height = 360, length = 1022, ptr = 0
I (26749) uvc_mic_spk_demo: uvc callback! frame_format = 7, seq = 27, width = 640, height = 360, length = 1022, ptr = 0

leeebo, 2024-01-19:
@jeff-getlucky LOG 开 verbose，看一下包解析和组包过程

jeff-getlucky, 2024-01-19:
以下是摄像头信息，我还发现frame_index是生效的，但format_index没生效，我设置的1，但log里提示设置成了2
I (17823) USB_STREAM: ENUM Stage GET_FULL_DEV_DESC, Succeed
*** Device descriptor ***
bcdUSB 2.00
bDeviceClass 0xef
bDeviceSubClass 0x2
bDeviceProtocol 0x1
bMaxPacketSize0 64
idVendor 0x2bdf
idProduct 0x101
bNumConfigurations 1
I (17836) USB_STREAM: ENUM Stage CHECK_FULL_DEV_DESC, Succeed
I (17837) USB_STREAM: ENUM Stage GET_SHORT_CONFIG_DESC, Succeed
I (17838) USB_STREAM: ENUM Stage CHECK_SHORT_CONFIG_DESC, Succeed
I (17839) USB_STREAM: ENUM Stage GET_FULL_CONFIG_DESC, Succeed
*** Configuration descriptor ***
wTotalLength 433
bNumInterfaces 2
bConfigurationValue 1
*** Interface Association Descriptor: Video ***
        *** Interface descriptor ***
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x1
                *** Endpoint descriptor ***
                bEndpointAddress 0x83   EP 3 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 16
                bInterval 8
        *** Interface descriptor ***
        bInterfaceNumber 1
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0xe (Video)
        bInterfaceSubClass 0x2
        *** Class-specific VS Interface Descriptor ***
        bNumFormats 3
        *** VS Format MJPEG Descriptor ***
        bFormatIndex 0x2
        bNumFrameDescriptors 3
        bDefaultFrameIndex 1
        *** VS MJPEG Frame Descriptor ***
        bFrameIndex 0x1
        wWidth 120
        wHeigh 160
        FrameInterval[0] 400000
        *** VS MJPEG Frame Descriptor ***
        bFrameIndex 0x2
        wWidth 240
        wHeigh 320
        FrameInterval[0] 333333
        *** VS MJPEG Frame Descriptor ***
        bFrameIndex 0x3
        wWidth 640
        wHeigh 360
        FrameInterval[0] 333333
                *** Endpoint descriptor ***
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
I (17846) USB_STREAM: Actual VS Interface(MPS <= 512) found, interface = 1, alt = 0
I (17847) USB_STREAM:   Endpoint(BULK) Addr = 0x81, MPS = 64
I (17847) USB_STREAM: Actual MJPEG format index, format index = 2, contains 3 frames
I (17848) USB_STREAM: Actual Frame: 7, width*heigh: 120*160, frame index = 1
I (17848) USB_STREAM: ENUM Stage CHECK_FULL_CONFIG_DESC, Succeed
I (17850) USB_STREAM: ENUM Stage SET_CONFIG, Succeed
I (17852) uvc_mic_spk_demo: UVC: get frame list size = 3, current = 0
I (17852) uvc_mic_spk_demo:     frame[0] = 120x160
I (17853) uvc_mic_spk_demo:     frame[1] = 240x320
I (17853) uvc_mic_spk_demo:     frame[2] = 640x360
I (17853) uvc_mic_spk_demo: Device connected
I (17854) USB_STREAM: Probe Format(2), Frame(1) 120*160, interval(400000)
I (17854) USB_STREAM: Probe payload size = 2048
bFormatIndex: 2
bFrameIndex: 1
dwFrameInterval: 400000
dwMaxPayloadTransferSize: 16384
bInterfaceNumber: 0
I (17885) USB_STREAM: dwMaxPayloadTransferSize set = 2048, probed = 16384
I (17886) USB_STREAM: Sample processing task start
I (17906) USB_STREAM: USB Device Connected

jeff-getlucky, 2024-01-19:
usb device tree

    =========================== USB Port1 ===========================

Connection Status        : 0x01 (Device is connected)
Port Chain               : 1-9-1

      ========================== Summary =========================
Vendor ID                : 0x2BDF (Hangzhou Hikvision Digital Technology Co., Ltd.)
Product ID               : 0x0101
USB Version              : 2.00
Port maximum Speed       : High-Speed
Device maximum Speed     : High-Speed
Device Connection Speed  : High-Speed
Self powered             : yes
Demanded Current         : 2 mA
Used Endpoints           : 3

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : USB Composite Device
Device Path              : \\?\USB#VID_2BDF&PID_0101#G52989620#{a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Kernel Name              : \Device\USBPDO-7
Device ID                : USB\VID_2BDF&PID_0101\G52989620
Hardware IDs             : USB\VID_2BDF&PID_0101&REV_0409 USB\VID_2BDF&PID_0101
Driver KeyName           : {36fc9e60-c465-11cf-8056-444553540000}\0035 (GUID_DEVCLASS_USB)
Driver                   : \SystemRoot\System32\drivers\usbccgp.sys (Version: 10.0.19041.1949  Date: 2022-09-15)
Driver Inf               : C:\Windows\inf\usb.inf
Legacy BusType           : PNPBus
Class                    : USB
Class GUID               : {36fc9e60-c465-11cf-8056-444553540000} (GUID_DEVCLASS_USB)
Service                  : usbccgp
Enumerator               : USB
Location Info            : Port_#0001.Hub_#0002
Location IDs             : PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(9)#USB(1), ACPI(_SB_)#ACPI(PC00)#ACPI(XHCI)#ACPI(RHUB)#ACPI(HS09)#USB(1)
Container ID             : {b0723e64-425d-5e0b-b5cf-2dd2dedd7a02}
Manufacturer Info        : (标准 USB 主控制器)
Capabilities             : 0x94 (Removable, UniqueID, SurpriseRemovalOK)
Status                   : 0x0180600A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
Address                  : 1
HcDisableSelectiveSuspend: 0
EnableSelectiveSuspend   : 0
SelectiveSuspendEnabled  : 0
EnhancedPowerMgmtEnabled : 0
IdleInWorkingState       : 0
WakeFromSleepState       : 0
Power State              : D0 (supported: D0, D2, D3, wake from D0, wake from D2)
 Child Device 1          : UVC Camera (USB 视频设备)
  Device Path            : \\?\USB#VID_2BDF&PID_0101&MI_00#7&1772ab12&0&0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\global (STATIC_KSCATEGORY_VIDEO_CAMERA)
  Kernel Name            : \Device\00000194
  Device ID              : USB\VID_2BDF&PID_0101&MI_00\7&1772AB12&0&0000
  Class                  : Camera
  Driver KeyName         : {ca3e7ab9-b4c3-4ae6-8251-579ef933890f}\0002 (GUID_DEVCLASS_CAMERA)
  Service                : usbvideo
  Location               : 0000.0014.0000.009.001.000.000.000.000
  LocationPaths          : PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(9)#USB(1)#USBMI(0)  PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(9)#USB(1)#USB(1)  ACPI(_SB_)#ACPI(PC00)#ACPI(XHCI)#ACPI(RHUB)#ACPI(HS09)#USB(1)#USBMI(0)  ACPI(_SB_)#ACPI(PC00)#ACPI(XHCI)#ACPI(RHUB)#ACPI(HS09)#USB(1)#USB(1)

        +++++++++++++++++ Registry USB Flags +++++++++++++++++
HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\usbflags\2BDF01010409
 osvc                    : REG_BINARY 00 00

        ---------------- Connection Information ---------------
Connection Index         : 0x01 (Port 1)
Connection Status        : 0x01 (DeviceConnected)
Current Config Value     : 0x01 (Configuration 1)
Device Address           : 0x09 (9)
Is Hub                   : 0x00 (no)
Device Bus Speed         : 0x02 (High-Speed)
[... 1022 lines cut ...]
dwBytesPerLine           : 0x00 (0 bytes)
adwFrameInterval[1]      : 0x00051615 (33.3333 ms -> 30.000 fps)
Data (HexDump)           : 1E 24 11 01 00 F0 00 40 01 00 00 7D 00 00 00 7D   .$.....@...}...}
                           00 15 16 05 00 01 00 00 00 00 15 16 05 00         ..............

        ------- VS Color Matching Descriptor Descriptor -------
bLength                  : 0x06 (6 bytes)
bDescriptorType          : 0x24 (Video Streaming Interface)
bDescriptorSubtype       : 0x0D (Color Matching)
bColorPrimaries          : 0x01 (BT.709, sRGB)
bTransferCharacteristics : 0x01 (BT.709)
bMatrixCoefficients      : 0x04 (SMPTE 170M)
Data (HexDump)           : 06 24 0D 01 01 04                                 .$....

        ----------------- Endpoint Descriptor -----------------
bLength                  : 0x07 (7 bytes)
bDescriptorType          : 0x05 (Endpoint Descriptor)
bEndpointAddress         : 0x81 (Direction=IN EndpointID=1)
bmAttributes             : 0x02 (TransferType=Bulk)
wMaxPacketSize           : 0x0040 (max 64 bytes)
bInterval                : 0x00 (never NAKs)
Data (HexDump)           : 07 05 81 02 40 00 00                              ....@..

      -------------------- String Descriptors -------------------
none

leeebo, 2024-01-24:
> frame_index是生效的，但format_index没生效

`frame_index`，和 `format_index` 一般不需要手动设置，你只需要配置分辨率和帧率就可以了，驱动会自动找到对应的 `frame_index`，和 `format_index`。只有在驱动找不到该项参数，才会用你指定的值

leeebo, 2024-01-24:
```
V (34498) USB_STREAM: zlp=0, lstp=1, req_len=2048, payload_len=1024, first=0x02, second=0x81
D (34499) USB_STREAM: bogus packet: len = 1024 02 81...2c 00

V (34499) USB_STREAM: Pipe(UVC): XFER_DONE
V (34500) USB_STREAM: zlp=0, lstp=1, req_len=2048, payload_len=1024, first=0x02, second=0x81
D (34500) USB_STREAM: bogus packet: len = 1024 02 81...5d 08

V (34500) USB_STREAM: Pipe(UVC): XFER_DONE
V (34501) USB_STREAM: zlp=0, lstp=1, req_len=2048, payload_len=1024, first=0x02, second=0x81
D (34501) USB_STREAM: bogus packet: len = 1024 02 81...d2 b3
```

这个摄像头，没有按照协商的大小返回数据包。为了兼容这个摄像头，你可以这样修改：

在 `menuconfig ->components -> USB Stream -> UVC Stream Config -> uvc bulk segment size in each urb transfer` 将 `CONFIG_NUM_BULK_BYTES_PER_URB` 从 2048 改为 1024

jeff-getlucky, 2024-01-25:
> ```
> V (34498) USB_STREAM: zlp=0, lstp=1, req_len=2048, payload_len=1024, first=0x02, second=0x81
> D (34499) USB_STREAM: bogus packet: len = 1024 02 81...2c 00
>
> V (34499) USB_STREAM: Pipe(UVC): [...]

成功了，谢谢巨佬，爱你。

closed 2024-01-25 by jeff-getlucky

jeff-getlucky, 2024-01-25:
@leeebo  实在是抱歉 ， 这个账号是公司的，可以帮我删除这个issue吗？

leeebo, 2024-01-25:
Github issue 无法删除，如果有隐私保护需求，删除你的评论即可

jeff-getlucky, 2024-01-25:
> Github issue 无法删除，如果有隐私保护需求，删除你的评论即可

查了一下只有项目拥有者才可以删除，那好吧，已经很麻烦你们了，谢谢。