espressif/esp-iot-solution #331: led_indicator: LED not off on Stop

diplfranzhoepfinger, 2024-01-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

https://components.espressif.com/components/espressif/led_indicator
https://docs.espressif.com/projects/esp-iot-solution/en/latest/display/led_indicator.html

if i define a looped Blink: 
like this one: 

https://github.com/espressif/esp-iot-solution/blob/6359293fbe0727767bd3431c26e33044c3144057/components/led/led_indicator/src/led_indicator_blink_default.c#L15

then i call `led_indicator_stop`, then it is that the LED is on or off randomly.
depending if it was in the on or off state when the stop came in. 

the Customer of a stop await that the LED is off in any case. 

any solution ? 

Thanks, 
Franz

diplfranzhoepfinger, 2024-01-27:
i found a possible Solution: 

after Init start a lowest Prio looped OFF, and never stop it. 

when the blinking ends the lowest prio off jumps in.

lijunru-hub, 2024-01-29:
You are correct. The 'stop' control logic doesn't actually turn off the light; it simply stops controlling the light. If you want to turn off the light when stopping, you can use the led_indicator_set_on_off function. Manually set the light to the off state during the 'stop' operation.

closed 2024-10-25 by leeebo