espressif/esp-iot-solution #333: 获取usb camera描述符错误

jeff-getlucky, 2024-02-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

下面是用pc软件查询到的信息
>                ----- Video Streaming MJPEG Frame Type Descriptor -----
>        bLength                  : 0x2E (46 bytes)
>        bDescriptorType          : 0x24 (Video Streaming Interface)
>        bDescriptorSubtype       : 0x07 (MJPEG Frame Type)
>        bFrameIndex              : 0x0B
>        bmCapabilities           : 0x00
>        wWidth                   : 0x0280 (640)
>        wHeight                  : 0x01E0 (480)
>        dwMinBitRate             : 0x000BB800 (768000 bps -> 96 KB/s)
>        dwMaxBitRate             : 0x0BB80000 (196608000 bps -> 24.576 MB/s)
>        dwMaxVideoFrameBufferSize: 0x00096000 (614400 bytes)
>        dwDefaultFrameInterval   : 0x00028B0A (16.6666 ms -> 60.000 fps)
>        bFrameIntervalType       : 0x05 (5 discrete frame intervals supported)
>        adwFrameInterval[1]      : 0x00028B0A (16.6666 ms -> 60.000 fps)
>        adwFrameInterval[2]      : 0x00030D40 (20.0000 ms -> 50.000 fps)
>        adwFrameInterval[3]      : 0x00051615 (33.3333 ms -> 30.000 fps)
>        adwFrameInterval[4]      : 0x0007A120 (50.0000 ms -> 20.000 fps)
>        adwFrameInterval[5]      : 0x000F4240 (100.0000 ms -> 10.000 fps)
>        Data (HexDump)           : 2E 24 07 0B 00 80 02 E0 01 00 B8 0B 00 00 00 B8   .$..............
>                                   0B 00 60 09 00 0A 8B 02 00 05 0A 8B 02 00 40 0D   ..`...........@.
>                                   03 00 15 16 05 00 20 A1 07 00 40 42 0F 00         ...... ...@B..

其中默认fps是60
>        dwDefaultFrameInterval   : 0x00028B0A (16.6666 ms -> 60.000 fps)
但esp32s3遍历的却是这样的
>       *** VS Format MJPEG Descriptor ***
>       bLength 0xb
>       bDescriptorType 0x24
>       bDescriptorSubType 0x6
>       bFormatIndex 0x1
>       bNumFrameDescriptors 1
>       bmFlags 0x1
>       bDefaultFrameIndex 1
>       bAspectRatioX 0
>       bAspectRatioY 0
>       bmInterlaceFlags 0x0
>       bCopyProtect 0
>       *** VS MJPEG Frame Descriptor ***
>       bLength 0x2a
>       bDescriptorType 0x24
>       bDescriptorSubType 0x7
>       bFrameIndex 0x1
>       bmCapabilities 0x0
>       wWidth 640
>       wHeigh 480
>       dwMinBitRate 768000
>       dwMaxBitRate 196608000
>       dwMaxVideoFrameBufSize 614400
>       dwDefaultFrameInterval 42666538
>       bFrameIntervalType 0
>       dwMinFrameInterval 42666501
>       dwMaxFrameInterval 85333248
>       dwFrameIntervalStep 102400000
读到的fps是
>       dwDefaultFrameInterval 42666538
转换成16进制是0x028B0A2A
而正确的应该是0x00028B0A
请问这里怎么修改

leeebo, 2024-02-06:
@jeff-getlucky 对于同一个摄像头，不同的 USB 模式会有不同的描述符配置，PC 上拿到的 Highspeed 描述符，S3 下摄像头只能工作在 Full-speed 模式，拿到的是 Full-speed 描述符。

现在无法确认是 S3 平台的解析问题。

可以在摄像头连接到 PC 之前加一个  USB 隔离器，PC 上获取 Full-speed 描述符进一步对比

jeff-getlucky, 2024-02-06:
> @jeff-getlucky 对于同一个摄像头，不同的 USB 模式会有不同的描述符配置，PC 上拿到的 Highspeed 描述符，S3 下摄像头只能工作在 Full-speed 模式，拿到的是 Full-speed 描述符。
> 
> 现在无法确认是 S3 平台的解析问题。
> 
> 可以在摄像头连接到 PC 之前加一个 USB 隔离器，PC 上获取 Full-speed 描述符进一步对比

但单独看esp32s3解析到的fps是4266.6538ms->0.234 fps
不可能呀

leeebo, 2024-02-06:
@jeff-getlucky dwDefaultFrameInterval 是一个 word，4 个 Bytes，如果你能修改摄像头固件，查一下 Fullspeed 的描述符，这个变量数据类型是不是选错了

jeff-getlucky, 2024-02-19:
usb隔离器到了，确实是因为摄像头不支持full speed。

closed 2024-02-19 by jeff-getlucky