espressif/esp-iot-solution #338: led_indicator: Different indicator LEDs on same strip

joeltroughton, 2024-03-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Is there any way to have different LEDs on an addressable strip acting as their own indicators? 

I can see how this might work with GPIO-connected LEDs by assigning each LED its own `led_indicator_handle_t`, but I can't seem to figure out how to accomplish the same thing when the LEDs are on a strip. An example might be to have index 0 blinking red while index 1 is breathing green.

Would be great, if possible!

lijunru-hub, 2024-03-04:
For information on indexing and support for WS2812 LED strips, please refer to the documentation: https://docs.espressif.com/projects/esp-iot-solution/en/latest/display/led_indicator.html#controlling-index
Alternatively, you can check out the example: https://github.com/espressif/esp-iot-solution/tree/master/examples/indicator/ws2812_strips.
Currently, up to 126 LED positions are supported, with the default starting at position 0. Setting it to 127 controls all LEDs.

lijunru-hub, 2024-05-27:
If you have no further questions or issues, we will proceed to close this issue. If you need any more help, please feel free to reopen this issue or create a new one.

Thank you for your understanding.

closed 2024-05-29 by leeebo