espressif/esp-iot-solution #339: usb cdc 4g module issue

achao1104, 2024-03-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

idf5.1 esp32s3-WROOM-1-N4R8 and master branch of esp-iot-solution

I have tried endpoint address in/out 0x84/0x3 0x85/0x4 0x83/0x2 0x81/0x1 
however, the issue is like below, and please help.

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x18e8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xda4
load:0x403cc700,len:0x3084
entry 0x403c9934
I (27) boot: ESP-IDF HEAD-HASH-NOTFOUND 2nd stage bootloader
I (27) boot: compile time Feb 21 2024 12:01:58
I (27) boot: Multicore bootloader
I (31) boot: chip revision: v0.2
I (35) qio_mode: Enabling default flash chip QIO
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (45) boot.esp32s3: SPI Mode       : QIO
I (50) boot.esp32s3: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot:  3 storage          Unknown data     01 82 00110000 00100000
I (100) boot: End of partition table
I (105) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=2b904h (178436) map
I (140) esp_image: segment 1: paddr=0003b92c vaddr=3fc9aa00 size=04564h ( 17764) load
I (144) esp_image: segment 2: paddr=0003fe98 vaddr=40374000 size=00180h (   384) load
I (146) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=a1e80h (663168) map
I (255) esp_image: segment 4: paddr=000e1ea8 vaddr=40374180 size=1678ch ( 92044) load
I (282) boot: Loaded app from partition at offset 0x10000
I (283) boot: Disabling RNG early entropy source...
I (294) cpu_start: Multicore app
I (294) cpu_start: Pro cpu up.
I (294) cpu_start: Starting app cpu, entry point is 0x40375884
0x40375884: call_start_cpu1 at D:/project/datacollection/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system/port/cpu_start.c:157

I (0) cpu_start: App cpu up.
I (312) cpu_start: Pro cpu start user code
I (312) cpu_start: cpu freq: 240000000 Hz
I (312) cpu_start: Application information:
I (315) cpu_start: Project name:     usb_cdc_4g_module
I (321) cpu_start: App version:      26ec17b8
I (326) cpu_start: Compile time:     Mar  4 2024 15:18:53
I (332) cpu_start: ELF file SHA256:  e974879d59b27fe9...
I (338) cpu_start: ESP-IDF:          HEAD-HASH-NOTFOUND
I (344) cpu_start: Min chip rev:     v0.0
I (349) cpu_start: Max chip rev:     v0.99
I (354) cpu_start: Chip rev:         v0.2
I (358) heap_init: Initializing. RAM available for dynamic allocation:
I (366) heap_init: At 3FCA2578 len 00047198 (284 KiB): DRAM
I (372) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (379) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (385) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (392) spi_flash: detected chip: generic
I (396) spi_flash: flash io: qio
[... 195 lines cut ...]
I (26452) modem_board: DTE reconnect, reconnecting ...

I (27462) modem_board: reconnect after 5s...
I (28462) modem_board: reconnect after 4s...
I (29462) modem_board: reconnect after 3s...
I (30462) modem_board: reconnect after 2s...
I (31462) modem_board: reconnect after 1s...
I (31462) modem_board: Modem state STAGE_SYNC, Start
E (31493) esp-modem-dte: esp_dte_handle_line(79): handle line failed
E (31516) esp-modem-dte: esp_dte_handle_line(79): handle line failed
I (31516) modem_board: Network Auto reconnecting ...
W (31518) 4g_main: Modem Board Event: Network disconnected
E (31524) led_indicator: ./managed_components/espressif__led_indicator/src/led_indicator.c:461 (led_indicator_stop):invalid p_handle
I (31518) modem_board: Modem state STAGE_SYNC, Success!
I (31642) modem_board: Modem state STAGE_CHECK_SIM, Start
W (31665) esp_modem_dce: esp_modem_dce_generic_command(31): Command:AT+CPIN?
...failed
W (31689) esp_modem_dce: esp_modem_dce_generic_command(31): Command:AT+CPIN=1234
...failed
W (31713) esp_modem_dce: esp_modem_dce_generic_command(31): Command:AT+CPIN?
...failed
W (31713) modem_board: Get SIM card state failed
W (31716) modem_board: Modem state STAGE_CHECK_SIM, Failed, retry1, after 3000ms...
I (34724) modem_board: Modem state STAGE_CHECK_SIM, Start
W (34747) esp_modem_dce: esp_modem_dce_generic_command(31): Command:AT+CPIN?

leeebo, 2024-03-06:
@achao1104 Did you config the IO for 4G reset?  please retry if not.

And please note not all 4G modules support PPP protocol, for example, some modules only support ECM or custom protocols, please contact the supplier to confirm whether PPP dial-up is supported.

achao1104, 2024-03-07:
> @achao1104 Did you config the IO for 4G reset? please retry if not.
> 
> And please note not all 4G modules support PPP protocol, for example, some modules only support ECM or custom protocols, please contact the supplier to confirm whether PPP dial-up is supported.

the 4g module is air795ug which is using the same chipset as air724ug. so I think it can support PPP.

leeebo, 2026-10-26:
@achao1104 @DanDevine Could you each retry with the latest esp-iot-solution master and iot_usbh_modem component using your exact modem? These reports involve different devices (Air795UG and SIMCOM A7670E), so please include the modem model and firmware, ESP-IDF/component revisions, USB interface/endpoints, sdkconfig, and whether PPP registration and traffic/ping work. The current example includes A7670E VID/PID 1E0E:9011 on interface 5, but we do not yet have hardware evidence that this fixes either report.