espressif/esp-iot-solution #340: ESP32S2使用USB_CDC_4G_MODULE时，连接WiFi芯片重启

EscapeDai7, 2024-03-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-v4.4

### Espressif SoC revision.

ESP32-S2N4R2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-KIT

### Power Supply used.

USB

### What is the expected behavior?

连接WiFi应该获取IP地址

### What is the actual behavior?

ESP32-S2的LOG中提示重启

### Steps to reproduce.

给ESP32-S板子上电并且提供额外供电给4G模组，4G模组正常联网并且生成WiFi

### Debug Logs.

```plain
I (11689) modem_board: Modem state STAGE_SYNC, Start
I (11709) modem_board: Network Auto reconnecting ...      
W (11709) 4g_main: Modem Board Event: Network disconnected
I (11709) modem_board: Modem state STAGE_SYNC, Success!   
I (11815) modem_board: Modem state STAGE_CHECK_SIM, Start
I (11818) modem_board: SIM Card Ready
I (11818) 4g_main: Modem Board Event: SIM Card Connected
I (11820) modem_board: Modem state STAGE_CHECK_SIM, Success!
I (11926) modem_board: Modem state STAGE_CHECK_SIGNAL, Start
I (11928) modem_board: Signal quality: rssi=28, ber=99
I (11928) modem_board: Modem state STAGE_CHECK_SIGNAL, Success!
I (12033) modem_board: Modem state STAGE_CHECK_REGIST, Start
I (12040) modem_board: Network registed, Operator: "46000"
I (12040) modem_board: Modem state STAGE_CHECK_REGIST, Success!
I (12143) modem_board: Modem state STAGE_START_PPP, Start
I (12156) modem_board: Modem state STAGE_START_PPP, Success!
W (12157) modem_board: Modem event! 0
I (12168) esp-netif_lwip-ppp: Connected
I (12168) modem_board: IP event! 6
I (12168) modem_board: Modem Connected to PPP Server
I (12172) modem_board: ppp ip: 10.12.3.237, mask: 255.255.255.255, gw: 10.64.64.64
I (12180) modem_board: Main DNS: 10.5.8.2
I (12185) modem_board: Backup DNS: 10.5.8.4
I (12190) esp-modem-netif: PPP state changed event 0: (NETIF_PPP_ERRORNONE)
I (12197) 4g_router_server: ssid : Can't find in NVS!
I (12203) 4g_router_server: password : Can't find in NVS!
I (12209) 4g_router_server: auth_mode : Can't find in NVS!
I (12215) 4g_router_server: channel : Can't find in NVS!
I (12221) 4g_router_server: hide_ssid : Can't find in NVS!
I (12227) 4g_router_server: bandwidth : Can't find in NVS!
I (12234) 4g_router_server: max_connection : Can't find in NVS!
I (12256) modem_board: Modem state STAGE_WAIT_IP, Start
I (12256) 4g_main: Modem Board Event: Network connected
I (12256) modem_board: Modem state STAGE_WAIT_IP, Success!
I (12267) 4g_router_server: Partition size: total: 956561, used: 182477
I (12270) 4g_router_server: Starting server on port: '80'
I (12277) 4g_router_server: Registering URI handlers
I (12282) 4g_router_server: Starting webserver
I (12288) wifi:wifi driver task: 3fff56b4, prio:23, stack:6656, core=0
I (12295) wifi:wifi firmware version: a9f5b59
I (12297) wifi:wifi certification version: v7.0
I (12301) wifi:config NVS flash: enabled
I (12305) wifi:config nano formating: disabled
I (12309) wifi:Init data frame dynamic rx buffer num: 32
I (12314) wifi:Init static rx mgmt buffer num: 5
I (12318) wifi:Init management short buffer num: 32
I (12323) wifi:Init dynamic tx buffer num: 32
I (12327) wifi:Init static rx buffer size: 1600
I (12331) wifi:Init static rx buffer num: 10
I (12335) wifi:Init dynamic rx buffer num: 32
I (12340) wifi_init: tcpip mbox: 32
I (12344) wifi_init: udp mbox: 6
I (12348) wifi_init: tcp mbox: 6
I (12351) wifi_init: tcp tx win: 5760
I (12356) wifi_init: tcp rx win: 5760
I (12360) wifi_init: tcp mss: 1440
I (12364) wifi_init: WiFi IRAM OP enabled
I (12369) wifi_init: WiFi RX IRAM OP enabled
I (12374) wifi_init: LWIP IRAM OP enabled
I (12379) phy_init: phy_version 2401,2a6dc26,Sep 26 2023,11:22:15
I (12428) wifi:mode : softAP (48:27:e2:50:f2:05)
I (12430) wifi:Total power save buffer number: 16
I (12431) wifi:Init max length of beacon: 752/752
I (12431) wifi:Init max length of beacon: 752/752
I (12436) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (12444) modem_wifi: Wi-Fi AP started
I (12453) wifi:Total power save buffer number: 16
I (12454) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (12462) modem_wifi: softap ssid: esp_4g_router password: 12345678
I (12469) modem_wifi: ap dns addr(default): 8.8.8.8
I (12475) 4g_main: changed: ap dns addr (auto): 10.5.8.2
I (12480) 4g_main: Ping addr 8.8.8.8 Restart..
I (12736) 4g_main: 64 bytes from 8.8.8.8 icmp_seq=1 ttl=111 time=250 ms

I (22486) 4g_main: Ping addr 8.8.8.8 Restart..
I (22636) 4g_main: 64 bytes from 8.8.8.8 icmp_seq=1 ttl=111 time=149 ms

I (32486) 4g_main: Ping addr 8.8.8.8 Restart..
I (32652) 4g_main: 64 bytes from 8.8.8.8 icmp_seq=1 ttl=111 time=165 ms

I (42486) 4g_main: Ping addr 8.8.8.8 Restart..
I (42652) 4g_main: 64 bytes from 8.8.8.8 icmp_seq=1 ttl=111 time=165 ms

I (44769) wifi:new:<6,0>, old:<6,1>, ap:<6,1>, sta:<255,255>, prof:6
I (44770) wifi:station: 5e:a8:6e:93:ed:10 join, AID=1, bgn, 20
I (44788) 4g_router_server: station 5e:a8:6e:93:ed:10 join, AID=1
I (44788) 4g_router_server: 5ea86e93ed10 : Can't find in NVS!

assert failed: ip_napt_init ip4_napt.c:220 (ip_portmap_table != NULL && ip_napt_table != NULL)

Backtrace: 0x40025776:0x3ffda340 0x4002cc85:0x3ffda360 0x40036e39:0x3ffda380 0x400a194c:0x3ffda4a0 0x400a1d16:0x3ffda4c0 0x40091e48:0x3ffda4e0 0x40115a4a:0x3ffda500 0x40115b54:0x3ffda560
0x40025776: call_start_cpu0 at F:/esp/v4.4/esp-idf/components/esp_system/port/cpu_start.c:590

0x4002cc85: prvReceiveGeneric at F:/esp/v4.4/esp-idf/components/esp_ringbuf/ringbuf.c:797 (discriminator 1)

0x40036e39: lmacTxFrame at ??:?

0x400a194c: esp_netif_start_api at F:/esp/v4.4/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:819

0x400a1d16: esp_netif_init_configuration at F:/esp/v4.4/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:421
 (inlined by) esp_netif_new at F:/esp/v4.4/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:579

0x40091e48: modem_wifi_set_dns at C:/Users/Dai/Desktop/esp-iot-solution-master/components/usb/iot_usbh_modem/src/usbh_modem_wifi.c:137 (discriminator 1)
```

leeebo, 2024-03-07:
```
assert failed: ip_napt_init ip4_napt.c:220 (ip_portmap_table != NULL && ip_napt_table != NULL)
```

这是由于内存不足导致的，S2 上如果想要开 web 管理界面，请增加并启用 PSRAM

EscapeDai7, 2024-03-07:
你好，我的芯片是ESP32-S2N4R2启用PSRAM还是同样情况，我在ESP32-S3内存与缓存的为8+16MB，开启PSRAM后，测试WIFI网络下行只有1Mbps，上行7Mbps左右，通过有线连接到PC端可以得到6Mbps的下行和3Mbps的上行，WiFi的影响还有这么大吗

leeebo, 2024-03-07:
@EscapeDai7 减小 [CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-reference/kconfig.html#config-spiram-malloc-alwaysinternal) 阈值，使更多的内存申请，默认放在 PSRAM

EscapeDai7, 2024-03-16:
谢谢你，经过努力调试，成功跑到了与有线连接时不相上下的速度，感谢

closed 2024-03-18 by leeebo