espressif/esp-iot-solution #341: 用USB CDC 4G Module时，卡在了Modem state STAGE_WAIT_IP步骤

tyrael-cui, 2024-03-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

我的硬件是ESP32-S2连接EC600M-CN模块，直接用了USB_CDC_MODULE中的例程。IDF版本 5.1.2。开机log如下：

> ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1     
load:0x3ffe6110,len:0x19c8
load:0x4004b000,len:0x4   
load:0x4004b004,len:0xc34 
load:0x4004f000,len:0x3428
entry 0x4004b1ec
I (23) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader
I (24) boot: compile time Mar  7 2024 15:01:58        
I (24) boot: chip revision: v0.0
I (28) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s2: SPI Speed      : 80MHz
I (38) boot.esp32s2: SPI Mode       : QIO
I (42) boot.esp32s2: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot:  3 storage          Unknown data     01 82 00110000 00100000
I (93) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=2c848h (182344) map
I (137) esp_image: segment 1: paddr=0003c870 vaddr=3ffcbed0 size=03520h ( 13600) load
I (140) esp_image: segment 2: paddr=0003fd98 vaddr=40024000 size=00280h (   640) load
I (143) esp_image: segment 3: paddr=00040020 vaddr=40080020 size=9f53ch (652604) map
I (263) esp_image: segment 4: paddr=000df564 vaddr=40024280 size=17c50h ( 97360) load
I (298) boot: Loaded app from partition at offset 0x10000
I (298) boot: Disabling RNG early entropy source...
I (309) cpu_start: Unicore app
I (309) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (309) cache: Data cache               : size 8KB, 4Ways, cache line size 32Byte
I (316) cpu_start: Pro cpu up.
I (339) cpu_start: Pro cpu start user code
I (339) cpu_start: cpu freq: 240000000 Hz
I (339) cpu_start: Application information:
I (342) cpu_start: Project name:     usb_cat1
I (347) cpu_start: App version:      1
I (351) cpu_start: Compile time:     Mar 12 2024 13:50:53
I (357) cpu_start: ELF file SHA256:  e4141d4cdf55fffb...
I (363) cpu_start: ESP-IDF:          v5.1.2-dirty
I (369) cpu_start: Min chip rev:     v0.0
I (373) cpu_start: Max chip rev:     v1.99 
I (378) cpu_start: Chip rev:         v0.0
I (383) heap_init: Initializing. RAM available for dynamic allocation:
I (390) heap_init: At 3FFD3200 len 00028E00 (163 KiB): DRAM
I (396) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (402) heap_init: At 3FF9E000 len 00001FE8 (7 KiB): RTCRAM
I (409) spi_flash: detected chip: generic
I (413) spi_flash: flash io: qio
I (418) app_start: Starting scheduler on CPU0
I (422) main_task: Started on CPU0
I (426) main_task: Calling app_main()
I (430) led_indicator: LED Indicator Version: 0.3.0
I (436) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (445) led_indicator: blink_lists is null, use default blink list
W (452) led_indicator: ./managed_components/espressif__led_indicator/src/led_indicator.c:280 (_led_indicator_create_com):LED indicator does not have the hal_indicator_set_brightness function
[... 297 lines cut ...]
E (159747) esp-modem-dte: esp_modem_dte_send_cmd(272): process command timeout
W (159747) esp_modem_dce: esp_modem_dce_generic_command(26): Command:+++ response timeout
I (159757) modem_board: Network Auto reconnecting ...
W (159758) 4g_main: Modem Board Event: Network disconnected
E (159764) led_indicator: ./managed_components/espressif__led_indicator/src/led_indicator.c:461 (led_indicator_stop):invalid p_handle
I (159777) modem_board: Modem state STAGE_SYNC, Success!
I (159883) modem_board: Modem state STAGE_CHECK_SIM, Start
I (159885) modem_board: SIM Card Ready
I (159885) 4g_main: Modem Board Event: SIM Card Connected
I (159888) modem_board: Modem state STAGE_CHECK_SIM, Success!
I (159995) modem_board: Modem state STAGE_CHECK_SIGNAL, Start
I (159997) modem_board: Signal quality: rssi=25, ber=99
I (159997) modem_board: Modem state STAGE_CHECK_SIGNAL, Success!
I (160103) modem_board: Modem state STAGE_CHECK_REGIST, Start
I (160105) modem_board: Network registed, Operator: "CHINA MOBILE"
I (160105) modem_board: Modem state STAGE_CHECK_REGIST, Success!
I (160211) modem_board: Modem state STAGE_START_PPP, Start
I (160217) modem_board: Modem state STAGE_START_PPP, Success!
W (160218) modem_board: Modem event! 0
I (160317) modem_board: Modem state STAGE_WAIT_IP, Start

========================================================
目前已经检查过EC600M模块，是正常工作的，SIM卡也可以注册上网，但就是无法获取IP。
请大家指教是什么原因造成的？
谢谢！

leeebo, 2024-03-15:
@tyrael-cui Please try with other SIM Card or [change the APN](https://github.com/espressif/esp-iot-solution/blob/master/components/usb/iot_usbh_modem/Kconfig#L106) same as the operator suggest.

tyrael-cui, 2024-03-18:
@leeebo 我设置了APN名字，打开了debug信息，发现已经连接成功了，但仍然无法获取IP。Log如下：
`
I (689) app_start: Starting scheduler on CPU0
I (694) main_task: Started on CPU0
D (698) heap_init: New heap initialised at 0x3fffc000
I (703) main_task: Calling app_main()
I (707) led_indicator: LED Indicator Version: 0.3.0
I (713) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (722) led_indicator: blink_lists is null, use default blink list
W (729) led_indicator: ./managed_components/espressif__led_indicator/src/led_indicator.c:280 (_led_indicator_create_com):LED indicator does not have the hal_indicator_set_brightness functionI (746) led_indicator: Indicator create successfully. type:GPIO mode, hardware_data:0x5, is_active_level_high:1, blink_lists:default
D (762) esp_netif_lwip: LwIP stack has been initialized
D (764) esp_netif_lwip: check: remote, if=0x0 fn=0x40119f78
0x40119f78: alloc_client_data_id at D:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_netif/lwip/esp_netif_lwip.c:166

D (770) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (776) esp_netif_lwip: esp-netif has been successfully initialized
D (782) event: running task for loop 0x3ffd8fe8
D (787) event: created task for loop 0x3ffd8fe8
D (791) event: created event loop 0x3ffd8fe8
I (796) 4g_main: ====================================
I (801) 4g_main:      ESP 4G Cat.1 Wi-Fi Router
I (806) 4g_main: ====================================
I (812) modem_board: iot_usbh_modem, version: 0.2.1
s_modem_evt_hdl ptr= 0x3ffcfc84
I (821) modem_board: Power the module first....
I (826) gpio: GPIO[9]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (835) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (844) modem_board: Power modem using io=10, level=1
I (1150) modem_board: Force reset modem board....
I (1150) modem_board: Resetting modem using io=9, level=1
I (1950) modem_board: Waiting for modem initialize ready
I (2950) modem_board: Waiting for modem 11 seconds left.....
I (3950) modem_board: Waiting for modem 10 seconds left.....
I (4950) modem_board: Waiting for modem 9 seconds left.....
I (5950) modem_board: Waiting for modem 8 seconds left.....
I (6950) modem_board: Waiting for modem 7 seconds left.....
I (7950) modem_board: Waiting for modem 6 seconds left.....
I (8950) modem_board: Waiting for modem 5 seconds left.....
I (9950) modem_board: Waiting for modem 4 seconds left.....
I (10950) modem_board: Waiting for modem 3 seconds left.....
I (11950) modem_board: Waiting for modem 2 seconds left.....
I (12950) modem_board: Waiting for modem 1 seconds left.....
D (12950) event: created event loop 0x3ffdba10
I (12950) USB_HCDC: iot_usbh_cdc, version: 0.2.2
I (12955) IOT_USBH: iot_usbh, version: 0.2.1
D (12960) intr_alloc: Connected src 48 to int 9 (cpu 0)
I (12995) IOT_USBH: USB Port=1 init succeed, fifo strategy=0
I (12995) IOT_USBH: USB Processing Start
I (12995) IOT_USBH: Waiting USB Connection
I (12999) IOT_USBH: Port power: ON Succeed
I (13004) IOT_USBH: usb port start succeed
I (13009) USB_HCDC: usbh cdc driver install succeed
I (13014) USB_HCDC: Waiting CDC Device Connection
I (13020) USB_HCDC: CDC task start
D (13024) IOT_USBH: urb alloced
D (13027) IOT_USBH: urb alloced
D (13030) IOT_USBH: urb alloced
D (13034) IOT_USBH: urb alloced
D (13037) IOT_USBH: urb alloced
D (13040) IOT_USBH: urb alloced
D (13043) IOT_USBH: urb alloced
D (13046) IOT_USBH: urb alloced
D (13050) IOT_USBH: urb alloced
D (13053) IOT_USBH: urb alloced
I (13254) IOT_USBH: line 131 PORT_EVENT_CONNECTION
I (13354) IOT_USBH: Resetting Port
I (13414) IOT_USBH: Port reset succeed
I (13414) IOT_USBH: Getting Port Speed
I (13414) IOT_USBH: USB Speed: full-speed
I (13416) IOT_USBH: Pipe init succeed, addr: 00
[... 481 lines cut ...]
D (90928) esp-netif_lwip-ppp: Phase Establish
D (90932) event: running post NETIF_PPP_STATUS:262 with handler 0x400949e8 and context 0x3ffeae80 on loop 0x3ffd8fe8
0x400949e8: on_ppp_changed at D:/Espressif/esp-iot-solution/components/usb/iot_usbh_modem/src/esp_modem_netif.c:73

D (90943) USB_HCDC: cdc write itf: 0, buf = 0x3ffedc9c, len = 47
D (90949) esp_netif_lwip: check: remote, if=0x3ffeb4a8 fn=0x4011ab6c
0x4011ab6c: esp_netif_update_default_netif_lwip at D:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_netif/lwip/esp_netif_lwip.c:327

D (90955) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffeb4a8
D (90962) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (90968) event: running post ESP_MODEM_EVENT:0 with handler 0x40091d34 and context 0x3ffeaeb4 on loop 0x3ffdba10
0x40091d34: on_modem_event at D:/Espressif/esp-iot-solution/components/usb/iot_usbh_modem/src/usbh_modem_board.c:329

W (90978) modem_board: Modem event! 0
D (90986) modem_board: Handling bits = 0120, stage = 7, retry = 0
I (90989) modem_board: Modem state STAGE_WAIT_IP, Start
D (96949) USB_HCDC: cdc write itf: 0, buf = 0x3ffedc9c, len = 47
D (102949) USB_HCDC: cdc write itf: 0, buf = 0x3ffedc9c, len = 47
D (108949) USB_HCDC: cdc write itf: 0, buf = 0x3ffedc9c, len = 47
D (114949) USB_HCDC: cdc write itf: 0, buf = 0x3ffedc9c, len = 47
D (120949) USB_HCDC: cdc write itf: 0, buf = 0x3ffedc9c, len = 47
D (126949) USB_HCDC: cdc write itf: 0, buf = 0x3ffedc9c, len = 47
D (132949) USB_HCDC: cdc write itf: 0, buf = 0x3ffedc9c, len = 47
D (138949) USB_HCDC: cdc write itf: 0, buf = 0x3ffedc9c, len = 47
`

tyrael-cui, 2024-03-21:
问题已经解决，所用的CAT1模块不支持PPP协议，但仍然在拨号命令时，返回connect，导致程序等待ip.

closed 2024-03-21 by tyrael-cui

yxmstar, 2025-06-11:
> [@yxmstar](https://github.com/yxmstar) 可以用 [ECM ](https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_ecm_4g_module) 或者 [RNDIS ](https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_rndis_4g_module) 驱动，现在都有支持

ok，thanks，我尝试一下，对于ECM或RNDIS工程需要额外配置吗？例如PID、VID、interface
我的报错信息在[#530 ] 麻烦帮忙看一下

yxmstar, 2025-06-12:
> [@yxmstar](https://github.com/yxmstar) 可以用 [ECM ](https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_ecm_4g_module) 或者 [RNDIS ](https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_rndis_4g_module) 驱动，现在都有支持

@leeebo 您好，我使用rndis启动似乎一切正常，但是4G模块无法联网，可能是哪里出问题了
------------------------------------------------------------------------------------------------------
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1918
load:0x403c8700,len:0x4
load:0x403c8704,len:0xe5c
load:0x403cb700,len:0x303c
entry 0x403c893c
I (27) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (27) boot: compile time Jun 11 2025 15:38:55
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode : QIO
I (49) boot.esp32s3: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label Usage Type ST Offset Length
I (70) boot: 0 nvs WiFi data 01 02 00009000 00006000
I (77) boot: 1 phy_init RF data 01 01 0000f000 00001000
I (85) boot: 2 factory factory app 00 00 00010000 00100000
I (92) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=27470h (160880) map
I (129) esp_image: segment 1: paddr=00037498 vaddr=3fc9e400 size=049a8h ( 18856) load
I (133) esp_image: segment 2: paddr=0003be48 vaddr=40378000 size=041d0h ( 16848) load
I (138) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=8a168h (565608) map
I (229) esp_image: segment 4: paddr=000ca190 vaddr=4037c1d0 size=121b4h ( 74164) load
I (253) boot: Loaded app from partition at offset 0x10000
I (253) boot: Disabling RNG early entropy source...
I (265) cpu_start: Multicore app
I (274) cpu_start: Pro cpu start user code
I (274) cpu_start: cpu freq: 240000000 Hz
I (274) app_init: Application information:
I (277) app_init: Project name: usb_rndis_4g_module
I (283) app_init: App version: 140c954-dirty
I (288) app_init: Compile time: Jun 11 2025 21:54:31
I (294) app_init: ELF file SHA256: 2f70135b5...
I (299) app_init: ESP-IDF: v5.3.1-dirty
I (305) efuse_init: Min chip rev: v0.0
I (310) efuse_init: Max chip rev: v0.99
I (314) efuse_init: Chip rev: v0.2
I (319) heap_init: Initializing. RAM available for dynamic allocation:
I (327) heap_init: At 3FCA63A0 len 00043370 (268 KiB): RAM
I (333) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (339) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (345) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (352) spi_flash: detected chip: generic
I (356) spi_flash: flash io: qio
W (360) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (373) sleep: Configure to isolate all GPIO pins in sleep state
I (380) sleep: Enable automatic switching of GPIO sleep configuration
I (387) main_task: Started on CPU0
I (402) main_task: Calling app_main()
I (403) USBH_CDC: iot usbh cdc version: 2.0.0
I (433) usbh_rndis: IOT USBH RNDIS Version: 0.2.0
I (433) iot_eth: IoT ETH Version: 0.1.0
I (434) usbh_rndis: USB RNDIS network interface init success
I (439) iot_eth.netif_glue: 00:00:00:00:00:00
I (443) iot_eth.netif_glue: ethernet attached to netif
E (1449) usbh_rndis: RNDIS auto detect failed
W (1449) RNDIS_4G_MODULE: Failed to start USB RNDIS driver, try again...
E (3450) usbh_rndis: RNDIS auto detect failed
W (3450) RNDIS_4G_MODULE: Failed to start USB RNDIS driver, try again...
[... 282 lines cut ...]
I (37319) phy_init: phy_version 680,a6008b2,Jun 4 2024,16:41:10
W (37361) phy_init: saving new calibration data because of checksum failure, mode(0)
I (37371) wifi:mode : softAP (30:ed:a0:ab:88:e5)
I (37372) wifi:Total power save buffer number: 32
I (37372) wifi:Init max length of beacon: 752/752
I (37373) wifi:Init max length of beacon: 752/752
I (37379) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
E (37387) ping_sock: send error=0
I (38296) iot_eth.netif_glue: Ethernet Got IP Address
I (38296) iot_eth.netif_glue: ~~~~~~~~~~~
I (38296) iot_eth.netif_glue: ETHIP:192.168.0.100
I (38301) iot_eth.netif_glue: ETHMASK:255.255.255.0
I (38306) iot_eth.netif_glue: ETHGW:192.168.0.1
I (38311) iot_eth.netif_glue: ~~~~~~~~~~~
W (39391) RNDIS_4G_MODULE: From 8.8.8.8 icmp_seq=1 timeout

W (44391) RNDIS_4G_MODULE: From 8.8.8.8 icmp_seq=1 timeout

W (49391) RNDIS_4G_MODULE: From 8.8.8.8 icmp_seq=1 timeout

W (54391) RNDIS_4G_MODULE: From 8.8.8.8 icmp_seq=1 timeout

W (59391) RNDIS_4G_MODULE: From 8.8.8.8 icmp_seq=1 timeout

W (64391) RNDIS_4G_MODULE: From 8.8.8.8 icmp_seq=1 timeout

yxmstar, 2025-06-12:
> 你使用的也是 ML307R？ 这个模组似乎有两个固件，有的固件可以直接跑 RNDIS，有的就是上不了网

是的，我问下厂商，谢谢你！