espressif/esp-iot-solution #344: knob processing accuracy is not enough for 1-pulse-1-positioning encoder

Maxwelltoo, 2024-03-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When I use the knob component in my project, I found that I needed two rotations to generate a knob callback(both right and left).

The reason is that the encoder I'm using only produces an edge with a minimum rotation.
So I thank the knob config may need to add a variable to represent the encoder type, and perform appropriate processing in the  `knob_handler()`.
I've made some changes to the code and tested it on my hardware, and finally am able to correctly generate a knob rotation event when a minimum rotation occurs.

Please let me know if you would like to handle this situation or accept a PR for this. 
Thanks.

Maxwelltoo, 2024-03-26:
[image]

This is the waveform my rotary encoder produces.

lijunru-hub, 2024-03-26:
I understand your requirement. The operation of the knob is based on treating any edge trigger on a GPIO as a rotation. However, currently, the direction of rotation is determined by another GPIO's edge trigger. On some less refined knobs, this can lead to a tactile sensation of two rotations but only one actual trigger.

In fact, the detection described above is more general for most knobs. However, by detecting the level of another GPIO instead of edge triggers, we can achieve finer counting (x2) and callbacks. I will give it a try.

我能明白你的需求，knob 的工作原理是当任意一个GPIO出现一个边沿触发视为一次旋转，但是旋转的方向目前是通过另一个GPIO的边缘触发的方向来判定的。在一些不太精细的旋钮上，就会出现手感上是两次但是实际触发一次的感觉。

大部分旋钮其实上述的检测更为通用，但是通过检测另一路的电平而不是边缘触发，就可以实现更精细的计数（x2）和回调，我会试一试的。