espressif/esp-iot-solution #346: fail to build usb cdc 4g module

tyrael-cui, 2024-04-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

esp-idf :5.1.2
VScode ：1.87.2
1 我在VScode下，以usb cdc 4g例程为基础进行开发。因为需要修改iot_usbh_modem中的代码，所以将iot_usbh_modem、iot_usbh和iot_usb_cdc这三个组件拷贝到了工程文件夹下的components目录中（此目录是自建的）
2 修改了main文件夹中idf_component.yml清单文件，内容如下：
~~~
dependencies:
  idf: ">=4.4.1"
  cmake_utilities: "0.*"
  led_indicator:
    version: "~0.3.0"
  json_parser:
    version: "1.0.3"
  iot_usbh_modem:
    version: ">=0.2.0"
#    override_path: "../../../../../components/usb/iot_usbh_modem"
    override_path: "../components/iot_usbh_modem"
~~~
3 然后开始编译，编译报错，log如下：
>
-- Found Git: D:/Espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/Espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/Espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info    
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/Espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s2        
Dependencies lock doesn't exist, solving dependencies.
Using component placed at E:\Project\ACC-10\components\iot_usbh for dependency iot_usbh(0.*)(introduced by component iot_usbh_cdc)
Using component placed at E:\Project\ACC-10\components\iot_usbh_cdc for dependency iot_usbh_cdc(0.*)(introduced by component iot_usbh_modem)
Using component placed at E:\Project\ACC-10\components\iot_usbh_modem for dependency iot_usbh_modem(>=0.2.0), specified in E:/Project/ACC-10/main\idf_component.yml
.........Using component placed at E:\Project\ACC-10\components\iot_usbh_cdc for dependency iot_usbh_cdc(0.*)(introduced by component iot_usbh_modem), specified in E:/Project/ACC-10/main\idf_component.yml
Using component placed at E:\Project\ACC-10\components\iot_usbh for dependency iot_usbh(0.*)(introduced by component iot_usbh_cdc)
...Using component placed at E:\Project\ACC-10\components\iot_usbh for dependency iot_usbh(0.*), specified in E:/Project/ACC-10/components/iot_usbh_cdc\idf_component.yml
.Using component placed at E:\Project\ACC-10\components\iot_usbh_cdc for dependency iot_usbh_cdc(0.*), specified in E:/Project/ACC-10/components/iot_usbh_modem\idf_component.yml
Using component placed at E:\Project\ACC-10\components\iot_usbh for dependency iot_usbh(0.*)(introduced by component iot_usbh_cdc), specified in E:/Project/ACC-10/components/iot_usbh_modem\idf_component.yml
....CMake Error at D:/Espressif/frameworks/esp-idf-v5.1.2/tools/cmake/build.cmake:540 (message):
  ***ERROR: Because project depends on iot_usbh_cdc (0.2.2) which doesn't match
  any versions, version solving failed.***

Call Stack (most recent call first):
  D:/Espressif/frameworks/esp-idf-v5.1.2/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "E:/Project/ACC-10/build/CMakeFiles/CMakeOutput.log".

 *  终端进程“C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B e:\Project\ACC-10\build -S e:\Project\ACC-10”已终止，退出代码: 1。 
>

***
Q1:从log里看，像是cmake无法解析iot_usbh_cdc的版本，请问这是什么原因？如何解决？
Q2:我把idf_component.yml文件中的override_path改成了esp-iot-solution的绝对地址，就可以编译通过，这是什么原因？
***

lijunru-hub, 2024-04-03:
这三个组件都可以不要手动拉过来，直接用在线拉取的方式。会自动拉取依赖
https://components.espressif.com/components/espressif/iot_usbh_modem

idf.py add-dependency "espressif/iot_usbh_modem^0.2.1"

tyrael-cui, 2024-04-03:
> 这三个组件都可以不要手动拉过来，直接用在线拉取的方式。会自动拉取依赖 https://components.espressif.com/components/espressif/iot_usbh_modem
> 
> idf.py add-dependency "espressif/iot_usbh_modem^0.2.1"

**感谢回复！**
我看编程指南里，自动拉取的会在工程文件夹下建一个managed_component来放置这些组件，但用户无法修改，可我想自己修改一下里面的代码，以便适应我的应用。
~~~
使用组件管理器添加组件在 [esp-registry](https://components.espressif.com/) 找到工程所需的组件

使用终端进入工程目录的根文件夹，执行 idf.py add-dependency "[命名空间]/<组件名>[版本号]"

这条指令将自动添加一个条目到 main/idf_component.yml，若文件不存在将自动创建

命名空间：可选，默认从 espressif (大小写不敏感)

组件名：必须，组件的名称，例如 usb_stream (大小写不敏感)

版本号：可选，默认为 * 代表任意版本。首次添加组件时，会自动下载最新版本

运行 idf.py build

备注

无需对工程 CMakeLists.txt 进行额外修改，可以直接在工程源代码中 include 组件的公共头文件、编译并链接组件的静态库

****用户无法直接修改 managed_components 中的组件，因为组件管理器会自动检测组件 Hash 值、并恢复更改。****

除了使用指令 idf.py add-dependency ，用户也可以直接修改 idf_component.yml 文件，手动修改组件条目。
~~~

lijunru-hub, 2024-04-03:
你可以让它先自动处理依赖，全部拉下来之后，把拉下来的文件直接丢到 project/components 目录中，然后就可以愉快的修改了

tyrael-cui, 2024-04-03:
> 你可以让它先自动处理依赖，全部拉下来之后，把拉下来的文件直接丢到 project/components 目录中，然后就可以愉快的修改了

好的，我试试看，谢谢！

closed 2024-05-21 by leeebo