espressif/esp-iot-solution #347: How to change usb host cdc baudrate?

Kevincoooool, 2024-04-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

This is set dtr\rts:```
#define USB_CTRL_REQ_CDC_SET_LINE_STATE(ctrl_req_ptr, itf, dtr, rts) ({  \
        (ctrl_req_ptr)->bmRequestType = 0x21;   \
        (ctrl_req_ptr)->bRequest = 0x22;    \
        (ctrl_req_ptr)->wValue = (rts ? 2 : 0) | (dtr ? 1 : 0); \
        (ctrl_req_ptr)->wIndex =  itf;    \
        (ctrl_req_ptr)->wLength = 0;   \
    })
```
But how to set baudrate?Now I test the cdc transfer speed is about 11 kb/s,that means the normal baudrate is 115200.
So if there is one way to set baudrate,that's perfect.

leeebo, 2024-04-06:
@Kevincoooool The `SET_LINE_STATE` and `SET_LINE_CODING` only used to set the UART's params of **USB-to-UART device**. For common USB CDC device, this configuration won't affect the speed of the USB side.

Kevincoooool, 2024-04-13:
> @Kevincoooool The `SET_LINE_STATE` and `SET_LINE_CODING` only used to set the UART's params of **USB-to-UART device**. For common USB CDC device, this configuration won't affect the speed of the USB side.

Yes, but i need to set baudrate for USB-to-UART device....

leeebo, 2024-04-15:
using [cdc_acm_host_line_coding_set](https://github.com/espressif/esp-usb/blob/master/host/class/cdc/usb_host_cdc_acm/include/usb/cdc_acm_host.h#L218) for baudrate and databits set.

closed 2024-05-21 by leeebo

Kevincoooool, 2024-05-27:
> @Kevincoooool using [cdc_acm_host_line_coding_set](https://github.com/espressif/esp-usb/blob/master/host/class/cdc/usb_host_cdc_acm/include/usb/cdc_acm_host.h#L218) for baudrate and databits set.

The demo can not work