espressif/esp-iot-solution #348: About examples/bluetooth/ble_conn_mgr/ble_spp/spp_server

nopnop2002, 2024-04-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I used this example using esp32s3.

https://github.com/espressif/esp-iot-solution/tree/master/examples/bluetooth/ble_conn_mgr/ble_spp/spp_server

This is mu environment:
```
$ idf.py --version
ESP-IDF v5.2.1-210-gcf7b9bed4c-dirty
```

This is my logging:
It doesn't seem to be working properly.
```
I (27) boot: ESP-IDF v5.2.1-210-gcf7b9bed4c-dirty 2nd stage bootloader
I (27) boot: compile time Apr  8 2024 10:28:03
I (28) boot: Multicore bootloader
I (32) boot: chip revision: v0.1
I (36) boot.esp32s3: Boot SPI Speed : 80MHz
I (40) boot.esp32s3: SPI Mode       : DIO
I (45) boot.esp32s3: SPI Flash Size : 2MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot: End of partition table
I (93) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=1d420h (119840) map
I (123) esp_image: segment 1: paddr=0002d448 vaddr=3fc98700 size=02bd0h ( 11216) load
I (125) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=57420h (357408) map
I (193) esp_image: segment 3: paddr=00087448 vaddr=3fc9b2d0 size=013e8h (  5096) load
I (194) esp_image: segment 4: paddr=00088838 vaddr=40374000 size=14688h ( 83592) load
I (225) boot: Loaded app from partition at offset 0x10000
I (225) boot: Disabling RNG early entropy source...
I (236) cpu_start: Multicore app
I (246) cpu_start: Pro cpu start user code
I (246) cpu_start: cpu freq: 160000000 Hz
I (246) cpu_start: Application information:
I (249) cpu_start: Project name:     spp_server
I (255) cpu_start: App version:      1
I (259) cpu_start: Compile time:     Apr  8 2024 10:27:56
I (265) cpu_start: ELF file SHA256:  b2d74571c...
I (270) cpu_start: ESP-IDF:          v5.2.1-210-gcf7b9bed4c-dirty
I (277) cpu_start: Min chip rev:     v0.0
I (282) cpu_start: Max chip rev:     v0.99
I (287) cpu_start: Chip rev:         v0.1
I (292) heap_init: Initializing. RAM available for dynamic allocation:
I (299) heap_init: At 3FC9EC70 len 0004AAA0 (298 KiB): RAM
I (305) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (311) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (317) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (324) spi_flash: detected chip: gd
I (328) spi_flash: flash io: dio
W (332) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (345) sleep: Configure to isolate all GPIO pins in sleep state
I (352) sleep: Enable automatic switching of GPIO sleep configuration
I (359) coexist: coex firmware version: 77cd7f8
I (364) coexist: coexist rom version e7ae62f
I (369) main_task: Started on CPU0
I (379) main_task: Calling app_main()
??I (409) app_main: BLE server UART_task started

I (409) blecm_nimble: BLE Connection Management: v0.1.1

I (409) BLE_INIT: BT controller compile version [30b57c4]
I (419) BLE_INIT: Bluetooth MAC: f4:12:fa:43:5c:56
I (419) phy_init: phy_version 640,cd64a1a,Jan 24 2024,17:28:12
I (469) blecm_nimble: BLE Host Task Started
I (469) blecm_nimble: No characteristic(0x2a00) found
I (469) blecm_nimble: No characteristic(0x2a01) found
I (479) blecm_nimble: No characteristic(0x2a05) found
I (479) blecm_nimble: No characteristic(0x2b3a) found
I (489) blecm_nimble: No characteristic(0x2b29) found
I (489) app_main: Callback for read
I (499) NimBLE: GAP procedure initiated: stop advertising.

I (509) NimBLE: GAP procedure initiated: advertise;
I (509) NimBLE: disc_mode=2
I (509) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (519) NimBLE:

I (529) main_task: Returned from app_main()
```

lisir233, 2024-04-21:
Different from classic Bluetooth, there is no SPP protocol specified in the BLE standard protocol, so the BLE SPP in this example is based on the vendor implementation, please use the spp_client example to connect with this example.

nopnop2002, 2024-04-26:
@lisir233 

Thanks for your comment.
I confirmed that I was able to connect after running the spp_client example.
But it's strange that my Android can't find spp_server.

[image]

lisir233, 2024-04-27:
As you can see from the logs, ESP32 has correctly established a connection with the phone, but because the phone's app doesn't have features like 0x2a00, it doesn't work properly, which is to be expected. Can you briefly describe the functionality you want to achieve with this routine?

nopnop2002, 2024-04-29:
>Can you briefly describe the functionality you want to achieve with this routine?

Thank you for your comment.

I want to connect to SPP_SERVER using my Android.

AlejandroHuerta, 2024-05-16:
I'm having the same issue with an esp32-c6-devkit-1n. I'm using the examples as is except for disabling the brownout detector. I'm running the client and server on two copies of the devices. I'm using esp-idf v5.2.1. I'm all out of ideas at this point so any help would be appreciated.

```
CLIENT

I (388) app_main: BLE client UART task started

I (388) blecm_nimble: BLE Connection Management: v0.1.1

I (388) BLE_INIT: Using main XTAL as clock source
I (398) BLE_INIT: ble controller commit:[217f1bf]
I (398) phy_init: phy_version 250,e14681b,Jan 24 2024,17:43:11
I (448) phy: libbtbb version: 939f79c, Jan 24 2024, 17:43:26
I (448) blecm_nimble: BLE Host Task Started
I (448) NimBLE: GAP procedure initiated: stop advertising.

I (448) NimBLE: GAP procedure initiated: discovery;
I (458) NimBLE: own_addr_type=0 filter_policy=0 passive=1 limited=0 filter_duplicates=1 
I (468) NimBLE: duration=forever
I (468) NimBLE:

I (478) main_task: Returned from app_main()

SERVER

I (513) blecm_nimble: BLE Host Task Started
I (513) blecm_nimble: No characteristic(0x2a00) found
I (523) blecm_nimble: No characteristic(0x2a01) found
I (523) blecm_nimble: No characteristic(0x2a05) found
I (533) blecm_nimble: No characteristic(0x2b3a) found
I (533) blecm_nimble: No characteristic(0x2b29) found
I (543) app_main: Callback for read
I (553) NimBLE: GAP procedure initiated: stop advertising.

I (553) NimBLE: GAP procedure initiated: advertise;
I (553) NimBLE: disc_mode=2
I (563) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (573) NimBLE:

I (573) main_task: Returned from app_main()

AlejandroHuerta, 2024-05-24:
After having the same issue with esp32-c3s I figured there must be something else going on and it turns out that if you do not provide enough power to the board, usb to pc alone isn't enough, that Bluetooth does not work. After connecting more power to the board via the second usb port everything is working. Posting this in case anyone else runs into the same issue.

lisir233, 2024-05-27:
Hi， @AlejandroHuerta ，Thank you for your message.，I'm trying to reproduce your question.I'm running the client and server on two copies of the devices，and I use esp-idf v5.2.1 as same,On my device it seems to work fine.
From the logs you provided, it seems that the Server is connected to another device instead of the Client, please check if there are other devices, such as your cell phone, that are unintentionally connected to the Server.

lisir233, 2024-05-27:
> > Can you briefly describe the functionality you want to achieve with this routine?
> 
> Thank you for your comment.
> 
> I want to connect to SPP_SERVER using my Android.

Hi, @nopnop2002  This example is only for data exchange between two ESP32s using BLE, if you want to use the ESP32 to interact with your phone, you may need to build an additional Android APP to simulate the functionality of the ESP32 Client.

AlejandroHuerta, 2024-05-27:
> Hi， @AlejandroHuerta ，Thank you for your message.，I'm trying to reproduce your question.I'm running the client and server on two copies of the devices，and I use esp-idf v5.2.1 as same,On my device [...]

Like I mentioned in my second post. Everything is working now after I provided more power to the device. I was previously only using power from USB and I disabled the brownout detection. This was a mistake.

lhespress, 2024-12-06:
@nopnop2002 Just as @lisir233 point that, any update about your Android APP?

nopnop2002, 2024-12-07:
@lhespress 

>any update about your Android APP?

No.

wujiangang, 2024-12-09:
Based on the information currently provided, we have identified the issue as being related to the Android app. @nopnop2002, could you please review this on your end? We plan to close this issue for now, but if you have any further updates, please feel free to reopen it.

closed 2024-12-09 by wujiangang