espressif/esp-iot-solution #351: usb cdc 4g module Auxiliary AT port

ZLYTT, 2024-04-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-3225-g5a40bb8746

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3

### Power Supply used.

USB

### What is the expected behavior?

How to support auxiliary AT port, that is, need to send AT command, do not need to exit PPP network, can send AT command

### What is the actual behavior?

After successful ppp dialing, the network connection status has been printed

### Steps to reproduce.

idf.py -p PORT monitor

### Debug Logs.

```plain
[图片]
```

leeebo, 2024-04-20:
@ZLYTT To achieve it, please can config the auxiliary AT port as [the secondary AT port](https://github.com/espressif/esp-iot-solution/blob/master/components/usb/iot_usbh_modem/Kconfig#L42)

ZLYTT, 2024-04-24:
> @ZLYTT To achieve it, please config the auxiliary AT port as [the secondary AT port](https://github.com/espressif/esp-iot-solution/blob/master/components/usb/iot_usbh_modem/Kconfig#L42)
[image]
Hello, I searched for this configuration in the menuconfig , and it was shown in red. Is it configured here? Is this configured

leeebo, 2024-04-25:
@ZLYTT Choose Modem Board -> Customer

leeebo, 2024-04-26:
> Choose Modem Board ->

sorry for the mistake, please choose `User defined`

closed 2024-05-21 by leeebo