espressif/esp-iot-solution #352: I had an error running the usb\device\usb_webcam> instance and I pulled down all the code. There will still be this error, I don't know what went wrong

garyToo, 2024-04-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I had an error running the usb\device\usb_webcam> instance and I pulled down all the code. There will still be this error, I don't know where the problem is, the error occurred when I ran the  idf.py set-target esp32s3. This is the error picture. Please help me
[image]

leeebo, 2024-04-20:
@garyToo the error caused by`idf-component-manager` `v1.5.*`, which Incorrectly resolved component's dependencies from yml file, you can workaround this by running `pip install "idf-component-manager==1.4.2"` in the IDF environment

garyToo, 2024-04-21:
Thank you for your guidance. Following your advice, I deleted the build files in usb_webcam. However, I searched the entire esp-iot-solution project and couldn't find the managered_components and dependencies.lock files. I had to proceed with executing the 'idf.py set-target esp32s3' command. But it encountered a new issue 
[image]

garyToo, 2024-04-28:
@leeebo   Can you help me take a look at my issue? I'm a beginner and I've been stuck on this problem for several days

leeebo, 2024-05-15:
@garyToo 请在 export IDF 环境以后，使用 `pip list` 再确认一下你安装的 `idf-component-manager` 版本，我们使用 `idf-component-manager==1.4.2` 在 IDF 5.0 下编译没有问题

closed 2024-06-14 by zhanzhaocheng