espressif/esp-iot-solution #355: 其他功能正常，但是在网页修改wifi账号密码的时候，会无法设置

dahwgdg, 2024-05-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

是那边设置的问题吗

dahwgdg, 2024-05-06:
[Uploading sdkconfigs3配置文件.zip…]()

lijunru-hub, 2024-05-07:
请详细说明你的开发环境，运行的例程，和问题复现的过程，以及报错的 log。

dahwgdg, 2024-05-07:
使用的是VSDODE，运行的是usb-cdc-4g-module，在网页端修改wifi账号密码时有时出现死机，有时会出现如下报错
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0xfffffffe  PS      : 0x00050033  A0      : 0xfffffffe  A1      : 0x3fca9990  
A2      : 0x3fca99d0  A3      : 0x3fcc0a38  A4      : 0x6f2a660c  A5      : 0x3fcac8e4  
A6      : 0x00000000  A7      : 0x0000180c  A8      : 0x0000002b  A9      : 0x8208a961  
A10     : 0x3fca99f0  A11     : 0x3fcc0a24  A12     : 0x1476fffe  A13     : 0x3fcaca86  
A14     : 0x00000001  A15     : 0x00001800  SAR     : 0x0000000b  EXCCAUSE: 0x00000014  
EXCVADDR: 0xfffffffc  LBEG    : 0x3fcc0a10  LEND    : 0x19aeb0e7  LCOUNT  : 0x00000003  

Backtrace: 0x7ffffffb:0x3fca9990 0x7ffffffb:0x3fca99b0 |<-CORRUPTED

ELF file SHA256: c522c6f2b9e94e95

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1830
load:0x403c9700,len:0xe8c
load:0x403cc700,len:0x3110
entry 0x403c9984

lijunru-hub, 2024-05-07:
可能是 4g 模组突然的大电流导致芯片掉电，建议单独给 4g 模组供电。

dahwgdg, 2024-05-07:
但是其他功能都是正常的

dahwgdg, 2024-05-07:
芯片没有掉电

lijunru-hub, 2024-05-07:
可以试一下不接 4g 模组，修改网页配置

dahwgdg, 2024-05-07:
直接修改网页配置也是没有更改，更改完重启，下面是日志
I (52052) 4g_router_server: =========== RECEIVED DATA ==========
I (52052) 4g_router_server: {"ssid":"talk_4g_router","if_hide_ssid":"false","auth_mode":"WPA_WPA2_PSK","password":"aaaaaaaaaaaa"}
I (52060) 4g_router_server: ====================================
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4207fdea
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1830
load:0x403c9700,len:0xe8c
load:0x403cc700,len:0x3110
SHA-256 comparison failed:
Calculated: 222ad3504f24d9b3102c55d2405094db242d3d24695b99d217c05e4d934c515c
Expected: 31dc5d83c0df7ad5fe4420d3826a8499271a6a6b2779cef1c2759c154baa52d4
Attempting to boot anyway...
entry 0x403c9984

dahwgdg, 2024-05-07:
我重新git一份，只用了wifi，日志显示I (1124) 4g_router_server: ssid : Can't find in NVS!
I (1130) 4g_router_server: password : Can't find in NVS!
I (1136) 4g_router_server: auth_mode : Can't find in NVS!
I (1142) 4g_router_server: channel : Can't find in NVS!
I (1148) 4g_router_server: hide_ssid : Can't find in NVS!
I (1154) 4g_router_server: bandwidth : Can't find in NVS!
I (1160) 4g_router_server: max_connection : Can't find in NVS!

lijunru-hub, 2024-11-12:
I have tested the example, and it allows you to change the username and password via the web interface. Do you have any further questions regarding this example? If not, I will proceed to close this issue.

closed 2024-12-09 by leeebo