espressif/esp-iot-solution #358: The esp32-S3 doesn't get the 1280*720 resolution of the USB camera

GSl-Horizon, 2024-05-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

My camera supports the resolution of 1280*720, but the esp32 can not get the resolution of 1280*720 when the get list, only get the resolution of the picture.
[image]

leeebo, 2024-05-15:
@GSl-Horizon: The camera has different configuration descriptors between High-Speed and Full-Speed modes. Using ESP32-S3, you can only get the Full-speed configurations, which may have no 720P resolution.

GSl-Horizon, 2024-05-15:
> ：相机在高速模式和全速模式之间具有不同的配置描述符。使用 ESP32-S3 时，只能获得全速配置，可能没有 720P 分辨率。

我怎么配置才能获取到720P的分辨率呢？还是只能换相机？

leeebo, 2024-05-15:
> > ：相机在高速模式和全速模式之间具有不同的配置描述符。使用 ESP32-S3 时，只能获得全速配置，可能没有 720P 分辨率。
> 
> 我怎么配置才能获取到720P的分辨率呢？还是只能换相机？

Modify Camera's firmware to support 720P in Fullspeed mode

GSl-Horizon, 2024-05-16:
> > > ：相机在高速模式和全速模式之间具有不同的配置描述符。使用 ESP32-S3 时，只能获得全速配置，可能没有 720P 分辨率。
> > 
> > 
> > 我怎么配置才能获取到720P的分辨率呢？还是只能换相机？
> 
> 修改相机固件以支持全速模式下的 720P

你好，我怎么才能把USB相机拍摄的一张图片通过wifi发送到PC呢？

leeebo, 2024-05-21:
@GSl-Horizon 请阅读这里的示例说明 https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_camera_mic_spk#usb-stream-example 。该示例通过 HTTP 发送图像，可在浏览器预览 https://github.com/espressif/esp-iot-solution/blob/master/examples/usb/host/usb_camera_mic_spk/main/main.c#L24

closed 2024-05-21 by leeebo