espressif/esp-iot-solution #359: fatal error: tusb_config.h: No such file or directory.

MR-VENOM, 2024-05-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

esp32s3

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-EYE

### Steps to reproduce.

1. Clone the repo from master branch.
2. open the usb_webcam example
3. Set sdkcong and menuconfig as per the docs.
4. Build.

### Build Logs.

```plain
[1/516] Building C object esp-idf/leeebo__tinyusb_src/CMakeFiles/__idf_leeebo__tinyusb_src.dir/src/tusb.c.obj
FAILED: esp-idf/leeebo__tinyusb_src/CMakeFiles/__idf_leeebo__tinyusb_src.dir/src/tusb.c.obj 
/home/user/esp/master/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v5.2.1-dirty\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/user/disk/examples/esp-iot-solution/examples/usb/device/usb_webcam/build/config -I/home/user/disk/examples/esp-iot-solution/examples/usb/device/usb_webcam/managed_components/leeebo__tinyusb_src -I/home/user/disk/examples/esp-iot-solution/examples/usb/device/usb_webcam/managed_components/leeebo__tinyusb_src/src -I/home/user/esp/v5.2.1/esp-idf/components/newlib/platform_include -I/home/user/esp/v5.2.1/esp-idf/components/freertos/config/include -I/home/user/esp/v5.2.1/esp-idf/components/freertos/config/include/freertos -I/home/user/esp/v5.2.1/esp-idf/components/freertos/config/xtensa/include -I/home/user/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/user/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/user/esp/v5.2.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/home/user/esp/v5.2.1/esp-idf/components/freertos/esp_additions/include -I/home/user/esp/v5.2.1/esp-idf/components/esp_hw_support/include -I/home/user/esp/v5.2.1/esp-idf/components/esp_hw_support/include/soc -I/home/user/esp/v5.2.1/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/home/user/esp/v5.2.1/esp-idf/components/esp_hw_support/port/esp32s3/. -I/home/user/esp/v5.2.1/esp-idf/components/heap/include -I/home/user/esp/v5.2.1/esp-idf/components/log/include -I/home/user/esp/v5.2.1/esp-idf/components/soc/include -I/home/user/esp/v5.2.1/esp-idf/components/soc/esp32s3 -I/home/user/esp/v5.2.1/esp-idf/components/soc/esp32s3/include -I/home/user/esp/v5.2.1/esp-idf/components/hal/platform_port/include -I/home/user/esp/v5.2.1/esp-idf/components/hal/esp32s3/include -I/home/user/esp/v5.2.1/esp-idf/components/hal/include -I/home/user/esp/v5.2.1/esp-idf/components/esp_rom/include -I/home/user/esp/v5.2.1/esp-idf/components/esp_rom/include/esp32s3 -I/home/user/esp/v5.2.1/esp-idf/components/esp_rom/esp32s3 -I/home/user/esp/v5.2.1/esp-idf/components/esp_common/include -I/home/user/esp/v5.2.1/esp-idf/components/esp_system/include -I/home/user/esp/v5.2.1/esp-idf/components/esp_system/port/soc -I/home/user/esp/v5.2.1/esp-idf/components/esp_system/port/include/private -I/home/user/esp/v5.2.1/esp-idf/components/xtensa/esp32s3/include -I/home/user/esp/v5.2.1/esp-idf/components/xtensa/include -I/home/user/esp/v5.2.1/esp-idf/components/xtensa/deprecated_include -I/home/user/esp/v5.2.1/esp-idf/components/lwip/include -I/home/user/esp/v5.2.1/esp-idf/components/lwip/include/apps -I/home/user/esp/v5.2.1/esp-idf/components/lwip/include/apps/sntp -I/home/user/esp/v5.2.1/esp-idf/components/lwip/lwip/src/include -I/home/user/esp/v5.2.1/esp-idf/components/lwip/port/include -I/home/user/esp/v5.2.1/esp-idf/components/lwip/port/freertos/include -I/home/user/esp/v5.2.1/esp-idf/components/lwip/port/esp32xx/include -I/home/user/esp/v5.2.1/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/user/esp/v5.2.1/esp-idf/components/lwip/port/esp32xx/include/sys -mlongcalls  -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -O2 -fmacro-prefix-map=/home/user/disk/examples/esp-iot-solution/examples/usb/device/usb_webcam=. -fmacro-prefix-map=/home/user/esp/v5.2.1/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -std=gnu17 -Wno-old-style-declaration -Wno-format -DCFG_TUSB_OS=OPT_OS_FREERTOS -DCFG_TUSB_MCU=OPT_MCU_ESP32S3 -MD -MT esp-idf/leeebo__tinyusb_src/CMakeFiles/__idf_leeebo__tinyusb_src.dir/src/tusb.c.obj -MF esp-idf/leeebo__tinyusb_src/CMakeFiles/__idf_leeebo__tinyusb_src.dir/src/tusb.c.obj.d -o esp-idf/leeebo__tinyusb_src/CMakeFiles/__idf_leeebo__tinyusb_src.dir/src/tusb.c.obj -c /home/user/disk/examples/esp-iot-solution/examples/usb/device/usb_webcam/managed_components/leeebo__tinyusb_src/src/tusb.c
In file included from /home/user/disk/examples/esp-iot-solution/examples/usb/device/usb_webcam/managed_components/leeebo__tinyusb_src/src/tusb.c:27:
/home/user/disk/examples/esp-iot-solution/examples/usb/device/usb_webcam/managed_components/leeebo__tinyusb_src/src/tusb_option.h:202:12: fatal error: tusb_config.h: No such file or directory
  202 |   #include "tusb_config.h"
      |            ^~~~~~~~~~~~~~~
compilation terminated.
[3/516] Performing build step for 'bootloader'
[1/1] cd /home/user/disk/examples/esp-iot-solution/examples/usb/device/usb_webcam/build/bootloader/esp-idf/esptool_py && /home/user/esp/master/.espressif/python_env/idf5.2_py3.10_env/bin/python /home/user/esp/v5.2.1/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/user/disk/examples/esp-iot-solution/examples/usb/device/usb_webcam/build/bootloader/bootloader.bin
Bootloader binary size 0x5840 bytes. 0x27c0 bytes (31%) free.
[6/514] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/src/esp_efuse_utility.c.obj
ninja: build stopped: subcommand failed.

 *  The terminal process "/usr/bin/zsh '-c', 'ninja '" terminated with exit code: 1.
```

lijunru-hub, 2024-05-15:
please use
```
 pip install "idf-component-manager<1.5"
```

beniroquai, 2024-05-17:
Thanks! Which python environment would I use? I tried 
```
 /Users/bene/.espressif/python_env/idf5.2_py3.10_env/bin/pip3  install "idf-component-manager<1.5"
 ```
 But the error remains the same.

closed 2024-06-14 by zhanzhaocheng