espressif/esp-iot-solution #366: [usb_device_uac] assert failed: xTaskCreatePinnedToCore freertos_tasks_c_additions.h

chnagy, 2024-06-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 S3 DevKitC 1

### Power Supply used.

USB

### What is the expected behavior?

I try to run a basic program of usb_device_uac based on the provided example which I am able to build and flash.

source:
```

#include <stdio.h>
#include "sdkconfig.h"
#include "esp_log.h"
#include "usb_device_uac.h"

static const char *TAG = "usb_uac_main";

static esp_err_t uac_device_output_cb(uint8_t *buf, size_t len, void *arg)
{
    return ESP_OK;
}

static esp_err_t uac_device_input_cb(uint8_t *buf, size_t len, size_t *bytes_read, void *arg)
{
    return ESP_OK;
}

static void uac_device_set_mute_cb(uint32_t mute, void *arg)
{
    ESP_LOGI(TAG, "uac_device_set_mute_cb: %"PRIu32"", mute);
}

static void uac_device_set_volume_cb(uint32_t volume, void *arg)
{
    ESP_LOGI(TAG, "uac_device_set_volume_cb: %"PRIu32"", volume);
}

void app_main(void)
{
    uac_device_config_t config = {
        .output_cb = uac_device_output_cb,
        .input_cb = uac_device_input_cb,
        .set_mute_cb = uac_device_set_mute_cb,
        .set_volume_cb = uac_device_set_volume_cb,
        .cb_ctx = NULL,
    };

    uac_device_init(&config);
}
```
idf_component.yml
```

dependencies:
  espressif/usb_device_uac: "^0.1.0"
  idf: ">=5.2.2"

```
skconfig.defaults
```
CONFIG_IDF_TARGET="esp32s3"
CONFIG_SPIRAM=y
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240=y
CONFIG_FREERTOS_HZ=1000
CONFIG_FREERTOS_PLACE_SNAPSHOT_FUNS_INTO_FLASH=y
CONFIG_UAC_SPEAKER_CHANNEL_NUM=2
CONFIG_UAC_MIC_CHANNEL_NUM=2
```

### What is the actual behavior?

It crashes with the following error while executing uac_device_init()

### Steps to reproduce.

everytime the device boots

### Debug Logs.

```plain
I (31) boot: compile time Jun  4 2024 10:55:05
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.2
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (43) boot.esp32s3: SPI Mode       : DIO
I (48) boot.esp32s3: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0bea0h ( 48800) map
I (113) esp_image: segment 1: paddr=0001bec8 vaddr=3fc93400 size=02d7ch ( 11644) load
I (116) esp_image: segment 2: paddr=0001ec4c vaddr=40374000 size=013cch (  5068) load
I (122) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=21c74h (138356) map
I (154) esp_image: segment 4: paddr=00041c9c vaddr=403753cc size=0df44h ( 57156) load
I (173) boot: Loaded app from partition at offset 0x10000
I (174) boot: Disabling RNG early entropy source...
I (185) cpu_start: Multicore app
I (185) octal_psram: vendor id    : 0x0d (AP)
I (186) octal_psram: dev id       : 0x02 (generation 3)
I (188) octal_psram: density      : 0x03 (64 Mbit)
I (194) octal_psram: good-die     : 0x01 (Pass)
I (199) octal_psram: Latency      : 0x01 (Fixed)
I (204) octal_psram: VCC          : 0x01 (3V)
I (209) octal_psram: SRF          : 0x01 (Fast Refresh)
I (215) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (221) octal_psram: BurstLen     : 0x01 (32 Byte)
I (227) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (233) octal_psram: DriveStrength: 0x00 (1/1)
I (238) esp_psram: Found 8MB PSRAM device
I (243) esp_psram: Speed: 40MHz
I (978) esp_psram: SPI SRAM memory test OK
I (987) cpu_start: Pro cpu start user code
I (987) cpu_start: cpu freq: 240000000 Hz
I (987) cpu_start: Application information:
I (990) cpu_start: Project name:     test2
I (995) cpu_start: App version:      1
I (999) cpu_start: Compile time:     Jun  4 2024 10:54:55
I (1005) cpu_start: ELF file SHA256:  0f6d87f45...
I (1011) cpu_start: ESP-IDF:          v5.2.2-dirty
I (1016) cpu_start: Min chip rev:     v0.0
I (1021) cpu_start: Max chip rev:     v0.99 
I (1026) cpu_start: Chip rev:         v0.2
I (1031) heap_init: Initializing. RAM available for dynamic allocation:
I (1038) heap_init: At 3FC98130 len 000515E0 (325 KiB): RAM
I (1044) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (1050) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1056) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (1063) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (1071) spi_flash: detected chip: generic
I (1075) spi_flash: flash io: dio
W (1079) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (1093) sleep: Configure to isolate all GPIO pins in sleep state
I (1099) sleep: Enable automatic switching of GPIO sleep configuration
I (1107) main_task: Started on CPU0
I (1111) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1120) main_task: Calling app_main()

assert failed: xTaskCreatePinnedToCore freertos_tasks_c_additions.h:178 (( ( ( ( BaseType_t ) xCoreID ) >= 0 && ( ( BaseType_t ) xCoreID ) < 2 ) ? ( ( BaseType_t ) 1 ) : ( ( BaseType_t ) 0 ) ) == ( (

Backtrace: 0x40375aae:0x3fc9b960 0x4037acd5:0x3fc9b980 0x403812b5:0x3fc9b9a0 0x4037de78:0x3fc9bac0 0x42009366:0x3fc9bb00 0x42008b95:0x3fc9bb30 0x42020d8b:0x3fc9bb70 0x4037b769:0x3fc9bba0
0x40375aae: panic_abort at /home/chris/esp/v5.2.2/esp-idf/components/esp_system/panic.c:466
0x4037acd5: esp_system_abort at /home/chris/esp/v5.2.2/esp-idf/components/esp_system/port/esp_system_chip.c:93
0x403812b5: __assert_func at /home/chris/esp/v5.2.2/esp-idf/components/newlib/assert.c:81
0x4037de78: xTaskCreatePinnedToCore at /home/chris/esp/v5.2.2/esp-idf/components/freertos/esp_additions/freertos_tasks_c_additions.h:178 (discriminator 1)
0x42009366: uac_device_init at /home/chris/Dev/test2/managed_components/espressif__usb_device_uac/usb_device_uac.c:499
0x42008b95: app_main at /home/chris/Dev/test2/main/main.c:38
0x42020d8b: main_task at /home/chris/esp/v5.2.2/esp-idf/components/freertos/app_startup.c:208
0x4037b769: vPortTaskWrapper at /home/chris/esp/v5.2.2/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: 0f6d87f45

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375a24
0x40375a24: esp_restart_noos at /home/chris/esp/v5.2.2/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:159

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x178c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcbc
load:0x403cc700,len:0x2d9c
entry 0x403c9914
I (31) boot: ESP-IDF v5.2.2-dirty 2nd stage bootloader
```

chnagy, 2024-06-04:
my bad. In the skconfig CONFIG_UAC_TINYUSB_TASK_CORE was set to -1

leeebo, 2024-06-05:
@chnagy Thanks for your report, it should be a bug in the Kconfig `range`, will fix it

closed 2024-06-14 by zhanzhaocheng