espressif/esp-iot-solution #367: [espressif/keyboard_button] How to use KBD_EVENT_COMBINATION?

Junanjunan, 2024-06-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am making HID keyboard and I use that library

https://components.espressif.com/components/espressif/keyboard_button/versions/0.1.1?language=en
https://docs.espressif.com/projects/esp-iot-solution/en/latest/input_device/keyboard_button.html

And this is the example of KBD_EVENT_COMBINATION

```c
keyboard_btn_cb_config_t cb_cfg = {
    .event = KBD_EVENT_COMBINATION,
    .callback = keyboard_combination_cb1,
    .event_data.combination.key_num = 2,
    .event_data.combination.key_data = (keyboard_btn_data_t[]) {
        {5, 1},
        {1, 1},
    },
};

keyboard_button_register_cb(kbd_handle, cb_cfg, NULL);
```

I registered callback function with that cb_cfg.
But, I don't know how to trggier KBD_EVENT_COMBINATION.

Please can you tell me how to trggier combination key event with gpios?

lijunru-hub, 2024-06-12:
This driver is specifically designed for a matrix keyboard circuit. In the example, {5,1} and {1,1} represent pressing the key at index {5,1} followed by the key at index {1,1}.

{5,1} represents the key at the sixth row and second column, starting from 0.
{1,1} represents the key at the second row and second column, starting from 0.

```
    keyboard_btn_config_t cfg = {
        .output_gpios = (int[])
        {
            40, 39, 38, 45, 48, 47
        },
        .output_gpio_num = 6,
        .input_gpios = (int[])
        {
            21, 14, 13, 12, 11, 10, 9, 4, 5, 6, 7, 15, 16, 17, 18
        },
        .input_gpio_num = 15,
        .active_level = 1,
        .debounce_ticks = 2,
        .ticks_interval = 500,
        .enable_power_save = true,
    };
    keyboard_btn_handle_t kbd_handle = NULL;
    keyboard_button_create(&cfg, &kbd_handle);
```

Based on the configured matrix keys, the actual IO port for the first key is {47,14}, and the actual IO port for the second key is {39,14}.Based on the configured matrix keys, the actual IO port for the first key is {47,14}, and the actual IO port for the second key is {39,14}.

lijunru-hub, 2024-06-18:
Do you have any further questions regarding this issue? If not, we will proceed to close it. Thank you very much for your inquiry.

Junanjunan, 2024-06-19:
Thank you, but I have more questions.
I am wondering what kind of task I can apply that keyboard combination to.

lijunru-hub, 2024-06-19:
The use of combination keys can be tailored to your specific application needs. You can also refer to the keyboard program, which demonstrates hotkey detection without using combination keys. However, you can report hotkeys using combination keys if desired, though it may be less flexible.

For reference, please see the keyboard program at:
[ESP IoT Solution Keyboard Example](https://github.com/espressif/esp-iot-solution/tree/master/examples/keyboard)

Junanjunan, 2024-06-19:
Aha, It can be used for hot key. So really Thank you.

closed 2024-06-19 by Junanjunan