espressif/esp-iot-solution #368: USB CDC 4G Example getting stuck

narangmayank, 2024-06-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0

### Espressif SoC revision.

ESP32S3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Execution shouldn't stop

### What is the actual behavior?

Execution is stopping. It's getting stuck somewhere

### Steps to reproduce.

sdkconfig and app_main is provided in the attahcment. I just added a counter in main while(1) and you can see in the logs it is getting stuck at some point.

```
[sdkconfig.txt](https://github.com/user-attachments/files/15748257/sdkconfig.txt)

[app_main.txt](https://github.com/user-attachments/files/15748256/app_main.txt)

### Debug Logs.

```plain
I (27) boot: ESP-IDF v5.1-dirty 2nd stage bootloader
I (27) boot: compile time Jun  8 2024 19:02:29
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) qio_mode: Enabling default flash chip QIO
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (49) boot.esp32s3: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot:  3 storage          Unknown data     01 82 00110000 00100000
I (100) boot: End of partition table
[... 574 lines cut ...]

W (52557) 4g_main: From 8.8.8.8 icmp_seq=1 timeout

I (52561) 4g_main: Ping addr 8.8.8.8 Restart..
I (52561) 4g_main: Counter : 33

I (53561) 4g_main: Ping addr 8.8.8.8 Restart..
I (53561) 4g_main: Counter : 34

W (54557) 4g_main: From 8.8.8.8 icmp_seq=1 timeout

I (54561) 4g_main: Ping addr 8.8.8.8 Restart..
I (54561) 4g_main: Counter : 35

I (55561) 4g_main: Ping addr 8.8.8.8 Restart..
I (55561) 4g_main: Counter : 36

W (56557) 4g_main: From 8.8.8.8 icmp_seq=1 timeout

I (56561) 4g_main: Ping addr 8.8.8.8 Restart..
I (56561) 4g_main: Counter : 37

I (57561) 4g_main: Ping addr 8.8.8.8 Restart..
I (57561) 4g_main: Counter : 38
```

leeebo, 2024-06-13:
Hi @narangmayank , please contact the SIM card supplier for the APN, then config it to https://github.com/espressif/esp-iot-solution/blob/master/components/usb/iot_usbh_modem/Kconfig#L106 to have a try

narangmayank, 2024-06-15:
Hi @leeebo, I tried using `airtelgprs.com` but that didn't worked out.

```
I (27) boot: ESP-IDF v5.1-dirty 2nd stage bootloader
I (27) boot: compile time Jun 15 2024 17:03:48
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) qio_mode: Enabling default flash chip QIO
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (49) boot.esp32s3: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot:  3 storage          Unknown data     01 82 00110000 00100000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=31090h (200848) map
I (143) esp_image: segment 1: paddr=000410b8 vaddr=3fc9a500 size=04484h ( 17540) load
I (147) esp_image: segment 2: paddr=00045544 vaddr=40374000 size=0aad4h ( 43732) load
I (157) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=af92ch (719148) map
I (267) esp_image: segment 4: paddr=000ff954 vaddr=4037ead4 size=0b99ch ( 47516) load
I (285) boot: Loaded app from partition at offset 0x10000
I (286) boot: Disabling RNG early entropy source...
I (297) cpu_start: Multicore app
I (297) cpu_start: Pro cpu up.
I (297) cpu_start: Starting app cpu, entry point is 0x40375870
0x40375870: call_start_cpu1 at C:/ESP_IDE/Espressif/frameworks/esp-idf-v5.1/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (315) cpu_start: Pro cpu start user code
I (315) cpu_start: cpu freq: 240000000 Hz
I (315) cpu_start: Application information:
I (318) cpu_start: Project name:     Byte0_FW
I (323) cpu_start: App version:      0710062-dirty
I (329) cpu_start: Compile time:     Jun 15 2024 17:03:09
I (335) cpu_start: ELF file SHA256:  98afcaace0f484bb...
I (341) cpu_start: ESP-IDF:          v5.1-dirty
I (346) cpu_start: Min chip rev:     v0.0
I (351) cpu_start: Max chip rev:     v0.99
I (356) cpu_start: Chip rev:         v0.1
I (360) heap_init: Initializing. RAM available for dynamic allocation:
I (368) heap_init: At 3FCA2868 len 00046EA8 (283 KiB): DRAM
I (374) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (381) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (387) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (394) spi_flash: detected chip: generic
I (398) spi_flash: flash io: qio
I (402) sleep: Configure to isolate all GPIO pins in sleep state
I (408) sleep: Enable automatic switching of GPIO sleep configuration
I (416) app_start: Starting scheduler on CPU0
I (421) app_start: Starting scheduler on CPU1
I (421) main_task: Started on CPU0
I (431) main_task: Calling app_main()
I (435) led_indicator: LED Indicator Version: 0.3.0
I (440) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (449) led_indicator: blink_lists is null, use default blink list
W (456) led_indicator: ./components/led_indicator/src/led_indicator.c:280 (_led_indicator_create_com):LED indicator does not have the hal_indicator_set_brightness function
I (472) led_indicator: Indicator create successfully. type:GPIO mode, hardware_data:0xf, is_active_level_high:1, blink_lists:default
I (485) led_indicator: LED Indicator Version: 0.3.0
I (490) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (500) led_indicator: blink_lists is null, use default blink list
W (507) led_indicator: ./components/led_indicator/src/led_indicator.c:280 (_led_indicator_create_com):LED indicator does not have the hal_indicator_set_brightness function
I (522) led_indicator: Indicator create successfully. type:GPIO mode, hardware_data:0x11, is_active_level_high:1, blink_lists:default
I (535) led_indicator: LED Indicator Version: 0.3.0
I (541) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (550) led_indicator: blink_lists is null, use default blink list
W (557) led_indicator: ./components/led_indicator/src/led_indicator.c:280 (_led_indicator_create_com):LED indicator does not have the hal_indicator_set_brightness function
[... 576 lines cut ...]
I (50097) 4g_main: Ping addr 8.8.8.8 Restart..
I (50097) 4g_main: Counter : 32

W (51094) 4g_main: From 8.8.8.8 icmp_seq=1 timeout

I (51097) 4g_main: Ping addr 8.8.8.8 Restart..
I (51097) 4g_main: Counter : 33

I (52097) 4g_main: Ping addr 8.8.8.8 Restart..
I (52097) 4g_main: Counter : 34

W (53094) 4g_main: From 8.8.8.8 icmp_seq=1 timeout

I (53097) 4g_main: Ping addr 8.8.8.8 Restart..
I (53097) 4g_main: Counter : 35

I (54097) 4g_main: Ping addr 8.8.8.8 Restart..
I (54097) 4g_main: Counter : 36

W (55094) 4g_main: From 8.8.8.8 icmp_seq=1 timeout

I (55097) 4g_main: Ping addr 8.8.8.8 Restart..
I (55097) 4g_main: Counter : 37

```

narangmayank, 2024-07-03:
@leeebo at your end is it working fine ?

lijunru-hub, 2024-11-19:
please see https://github.com/espressif/esp-iot-solution/issues/412 
Enable the watchdog and provide more logs.