espressif/esp-iot-solution #370: [espressif/keyboard_button] How to handle previously pressed keycodes?

Junanjunan, 2024-06-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

(Really thank you for answering key combination event question last time.)

With keyboard_button_create function, gpio pins that configured become active.
After that, when I press keycodes and release keycodes, it is a little different from usual keyboard.

I will explain that ASCII instead of gpio number.
I press 'a' and after that press 'b' without release 'a'.
Now, the pressed keys are a and b.
And, if I release 'b', mcu recognize 'a' is pressed, so keyboard reports 'a' to pc.

Another usual keyboard does not operate like that,
In usual keyboards, when I release 'b' in situation 'a' and 'b' are pressed,
usual keyboards recognize all keys are released even if I don't release 'a' key.

So, I wonder that I can handle that like usual keyboards with some options or function in your library.
Or, do I have to handle it with personally?

lijunru-hub, 2024-06-12:
This component will report a series of data whenever the key state changes (any key press/release).

You can choose to handle only key press events and ignore key release events.

```c
static void keyboard_cb(keyboard_btn_handle_t kbd_handle, keyboard_btn_report_t kbd_report, void *user_data)
{
    /*!< progress with key pressed */
    if (kbd_report.key_change_num > 0) {
       
    }

    /*!< progress with key release*/
    if (kbd_report.key_change_num < 0) {
       
    }
}
```

Junanjunan, 2024-06-12:
Really so thank you about corret and fast answer!

closed 2024-06-12 by Junanjunan