espressif/esp-iot-solution #373: EPS32S3 SH8601 QSPI驱动无法正常使用

lotoohe-space, 2024-06-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

我使用官方提供的驱动：https://components.espressif.com/components/espressif/esp_lcd_sh8601
按照要求修改引脚后，屏幕无法电亮，用示波器测试每个引脚均有输出。我查看sh8601的datasheet，发现命令采用的1wire模式，但是驱动里面好像没有这样设置，我不确实否是是这里的问题：
[image]

Horion0415, 2024-06-17:
您好，我们驱动中采用的就是 1 wire 模式。请问您是否根据您的屏幕修改了初始化参数呢？一般即使 IC 型号相同，但是不同屏的初始化参数也是不同的。请检查 RST 以及背光是否正常，并确认初始化参数是否正确。

lotoohe-space, 2024-06-17:
初始化参数包括哪些呢，rst我检查了是正常的，我的屏没有单独的背光控制接口。io我检查了配置都是正确。硬件设计上有什么需要注意的么？

Horion0415, 2024-06-17:
https://github.com/espressif/esp-iot-solution/blob/e2529ce3b0aa6362663b4c53d7e9ba21ff307f49/components/display/lcd/esp_lcd_sh8601/esp_lcd_sh8601.c#L193 一般初始化参数购买屏幕时商家会提供，您需要对应修改。不同屏幕的初始化参数很有可能是不同的。另外还需要确认下背光是否正常点亮。

lotoohe-space, 2024-06-17:
好的，我知道了，谢谢

lotoohe-space, 2024-06-17:
确实是初始化命令不同，已经可以了