espressif/esp-iot-solution #374: iot_button_get_event无法获取到BUTTON_SINGLE_CLICK，BUTTON_DOUBLE_CLICK事件

vgegok, 2024-06-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

已测试的版本5.2.1/5.2.2

### Espressif SoC revision.

ESP32 S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3 DEV Module

### Steps to reproduce.

在lvgl的indev的read_cb函数中读取button事件，并且打印事件编号：
`
static void keyboard_read_cb(lv_indev_t *indev, lv_indev_data_t *data)
{
    button_event_t btn_event = iot_button_get_event(btn_handle);
    if (btn_event != BUTTON_NONE_PRESS)
    {
        ESP_LOGI(TAG, "iot_button_get_event %d", btn_event);
    }
}
`

或者使用定时器读取事件，也可以使用while(1)循环读取。

只能得到其他中间事件，无法得到BUTTON_SINGLE_CLICK和BUTTON_DOUBLE_CLICK；
`
I (756) main_task: Returned from app_main()
I (7846) TEST: iot_button_get_event 0
I (7876) TEST: iot_button_get_event 0
I (7916) TEST: iot_button_get_event 0
I (7946) TEST: iot_button_get_event 0
I (7986) TEST: iot_button_get_event 1
I (8016) TEST: iot_button_get_event 1
I (8056) TEST: iot_button_get_event 1
I (8086) TEST: iot_button_get_event 1
I (8126) TEST: iot_button_get_event 1
I (8966) TEST: iot_button_get_event 0
I (8996) TEST: iot_button_get_event 0
I (9036) TEST: iot_button_get_event 0
I (9066) TEST: iot_button_get_event 0
I (9106) TEST: iot_button_get_event 0
I (9136) TEST: iot_button_get_event 0
I (9176) TEST: iot_button_get_event 0
I (9206) TEST: iot_button_get_event 0
I (9246) TEST: iot_button_get_event 0
I (9276) TEST: iot_button_get_event 0
I (9316) TEST: iot_button_get_event 0
I (9346) TEST: iot_button_get_event 0
[... 283 lines cut ...]
I (24016) TEST: iot_button_get_event 1
I (24046) TEST: iot_button_get_event 1
I (24086) TEST: iot_button_get_event 1
I (24116) TEST: iot_button_get_event 2
I (24156) TEST: iot_button_get_event 2
I (24186) TEST: iot_button_get_event 1
I (24226) TEST: iot_button_get_event 1
I (24256) TEST: iot_button_get_event 1
I (24296) TEST: iot_button_get_event 2
I (24326) TEST: iot_button_get_event 2
I (24366) TEST: iot_button_get_event 1
I (24396) TEST: iot_button_get_event 1
I (24436) TEST: iot_button_get_event 2
I (24466) TEST: iot_button_get_event 2
I (24506) TEST: iot_button_get_event 2
I (24536) TEST: iot_button_get_event 1
I (24576) TEST: iot_button_get_event 1
I (24606) TEST: iot_button_get_event 1
I (24646) TEST: iot_button_get_event 1
I (24676) TEST: iot_button_get_event 1
`

### More Information.

我还是用MACOS构建代码，同样的效果

lijunru-hub, 2024-06-17:
我测试了这个，确实是这样，原因是单击和双击的事件很快会被 BUTTON_PRESS_REPEAT_DONE 事件给覆盖掉。

轮询的方式确实会丢事件，这个是不可避免的，建议使用 `iot_button_register_cb` 接口，通过软件回调去做

lijunru-hub, 2024-06-17:
[image]

没有轮询到 BUTTON_SINGLE_CLICK 事件

vgegok, 2024-06-17:
> 我测试这个，确实是这样，既然单击并且双击的事件很快会被 BUTTON_PRESS_REPEAT_DONE 事件给覆盖掉。
> 
> 轮询方式实例会丢失事件，这个是核心，建议使用`iot_button_register_cb`接口，通过软件回调函数

你确定是被覆盖掉了吗？我觉得根本没有发送这个事件，我尝试了一个周末都没有任何一次获得BUTTON_SINGLE_CLICK 事件；

我在lvgl中作为输入设备使用，如果使用iot_button_register_cb的话我就得实现button和lv_indev_cb两个回调函数，这样显得过于复杂。不过如你建议，我正在参考esp的官方代码进行iot_button_register_cb的测试；

但是，如果可以直接在lv_indev的read_cb中直接获取iot_button的事件，那最好不过了。

vgegok, 2024-06-17:
如果在lvgl中使用iot_button，我觉得会过于复杂，如果有多个按键，就会更繁琐，我看了官方相关代码，都很复杂，不像单纯使用gpio实现的按键那么简单。我现在的做法是，将iot_botton的单击，双击，长按等事件在lv_indev_read_cb中映射到LV_KEY_XXXX（也就是LVGL支持的某个按键的单击事件），然后在lvgl的obj事件中处理LVGL的事件就可以了。

我觉得这样才是对LVGL最友好的做法，不知道你怎么认为？

lijunru-hub, 2024-06-17:
single click 确实是被覆盖掉了。可以参考这个组件的实现 https://github.dev/bienxanh1901/esp-bsp/tree/master/components/esp_lvgl_port 函数 `lvgl_port_add_navigation_buttons`

通过 button 的事件回调修改掉一个全局变量，然后再 lvgl 的轮询回调中处理这个全局变量。

vgegok, 2024-06-17:
[](https://github.dev/bienxanh1901/esp-bsp/tree/master/components/esp_lvgl_port)
这个链接我打不开

vgegok, 2024-06-17:
可以了

closed 2024-06-28 by leeebo