espressif/esp-iot-solution #375: 下载espressif_esp_lcd_st7701_1.1.0压缩包解压后，构建项目提示错误

WCJDI, 2024-06-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF5.3

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

github上下载的软件包

### Steps to reproduce.

下载直接编译出错

### Build Logs.

```plain
PS C:\Users\wangc\Desktop\espressif_esp_lcd_st7701_1.1.0 (1)\examples\rgb_avoid_tearing> idf.py build
Executing action: all (aliases: build)
Running cmake in directory 'C:\Users\wangc\Desktop\espressif_esp_lcd_st7701_1.1.0 (1)\examples\rgb_avoid_tearing\build'
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=E:\ESP32\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 'C:\Users\wangc\Desktop\espressif_esp_lcd_st7701_1.1.0 (1)\examples\rgb_avoid_tearing'"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: E:/ESP32/Espressif/tools/idf-git/2.43.0/cmd/git.exe (found version "2.43.0.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: E:/ESP32/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: E:/ESP32/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: E:/ESP32/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Dependencies lock doesn't exist, solving dependencies.
CMake Error at E:/ESP32/Espressif/frameworks/esp-idf-v5.2.1/tools/cmake/build.cmake:544 (message):
  ERROR: The 'override_path' field in the manifest file
  '..\..\..\..\..\components\display\lcd\esp_lcd_panel_io_additions\idf_component.yml'
  does not point to a directory.  You can safely remove this field from the
  manifest if this project is an example copied from a component repository.
  The dependency will be downloaded from the ESP component registry.
  Documentation:
  https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/manifest_file.html#override-path

Call Stack (most recent call first):
  E:/ESP32/Espressif/frameworks/esp-idf-v5.2.1/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/wangc/Desktop/espressif_esp_lcd_st7701_1.1.0 (1)/examples/rgb_avoid_tearing/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the C:\Users\wangc\Desktop\espressif_esp_lcd_st7701_1.1.0 (1)\examples\rgb_avoid_tearing\build\log\idf_py_stderr_output_750584 and C:\Users\wangc\Desktop\espressif_esp_lcd_st7701_1.1.0 (1)\examples\rgb_avoid_tearing\build\log\idf_py_stdout_output_750584
PS C:\Users\wangc\Desktop\espressif_esp_lcd_st7701_1.1.0 (1)\examples\rgb_avoid_tearing>
```

### More Information.

NO

Horion0415, 2024-06-17:
Hello, before building the project, you need to set the target chip first. The specific command is: "idf.py set-target esp32s3". This component is limited to use on the ESP32S3 chip. The default chip is ESP32, which is why your build encountered an error.

leeebo, 2024-06-26:
@WCJDI 如果是包管理器拉下来的组件包，应该已经不包含  `override_path`，如果你是从 git 仓库直接复制的组件包文件夹，请手动删除`idf_component.yml` 中的 `override_path` 或修改成你本地的实际的位置