espressif/esp-iot-solution #377: USB_dongle 无法发送命令

yunfusheng33, 2024-07-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Espressif SoC revision.

esp32s3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-s3-DevkitC-1

### Power Supply used.

USB

### What is the expected behavior?

我正确编译和烧录了USB_dongle，希望能用命令去连接网络

### What is the actual behavior?

监视器打印日志正常但是我无法发送命令

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
PS D:\work\ESP\esp-project\test\usb_dongle> & set IDF_PATH='d:\work\ESP32\v5.2.2\esp-idf'
PS D:\work\ESP\esp-project\test\usb_dongle> & 'd:\work\ESP32\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe' 'd:\work\ESP32\v5.2.2\esp-idf\tools\idf_monitor.py' -p COM12 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 'd:\work\ESP\esp-project\test\usb_dongle\build\usb_dongle.elf'
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM12 instead...
--- esp-idf-monitor 1.4.0 on \\.\COM12 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3810,len:0x1788
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc5c
load:0x403cc700,len:0x2d18
entry 0x403c98f8
I (33) boot: ESP-IDF v5.2.2 2nd stage bootloader
I (33) boot: compile time Jul  1 2024 13:55:46
W (33) boot: Unicore bootloader
I (36) boot: chip revision: v0.1
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (45) boot.esp32s3: SPI Mode       : SLOW READ
I (50) boot.esp32s3: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (78) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (86) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (93) boot:  3 factory          factory app      00 00 00010000 00100000
I (101) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (108) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (116) boot: End of partition table
I (120) boot: Defaulting to factory image
I (125) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=24074h (147572) map
I (168) esp_image: segment 1: paddr=0003409c vaddr=3fc97000 size=042e4h ( 17124) load
I (178) esp_image: segment 2: paddr=00038388 vaddr=40374000 size=07c90h ( 31888) load
I (187) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=93020h (602144) map
I (331) esp_image: segment 4: paddr=000d3048 vaddr=4037bc90 size=0b2e4h ( 45796) load
I (357) boot: Loaded app from partition at offset 0x10000
I (357) boot: Disabling RNG early entropy source...
I (368) cpu_start: Unicore app
I (369) cpu_start: Single core mode
I (377) cpu_start: Pro cpu start user code
I (377) cpu_start: cpu freq: 240000000 Hz
I (377) cpu_start: Application information:
I (380) cpu_start: Project name:     usb_dongle
I (386) cpu_start: App version:      1
I (390) cpu_start: Compile time:     Jul  1 2024 13:55:18
I (396) cpu_start: ELF file SHA256:  6afa12b38127914b...
I (402) cpu_start: ESP-IDF:          v5.2.2
I (407) cpu_start: Min chip rev:     v0.0
I (412) cpu_start: Max chip rev:     v0.99 
I (417) cpu_start: Chip rev:         v0.1
I (421) heap_init: Initializing. RAM available for dynamic allocation:
I (428) heap_init: At 3FCA1250 len 000484C0 (289 KiB): RAM
I (435) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (441) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (447) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
W (454) spi_flash: Octal flash chip is using but dio mode is selected, will automatically swich to Octal mode
I (464) spi_flash: detected chip: mxic (opi)
I (469) spi_flash: flash io: opi_str
W (473) spi_flash: Detected size(32768k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (486) sleep: Configure to isolate all GPIO pins in sleep state
I (493) sleep: Enable automatic switching of GPIO sleep configuration
I (500) coexist: coex firmware version: d96c1e51f
I (505) coexist: coexist rom version e7ae62f
I (511) main_task: Started on CPU0
I (515) main_task: Calling app_main()
I (523) USB_Dongle: USB initialization
W (524) tusb_desc: No Device descriptor provided, using default.
W (530) tusb_desc: No FullSpeed configuration descriptor provided, using default.
W (538) tusb_desc: No String descriptors provided, using default.
I (545) tusb_desc: 
┌─────────────────────────────────┐
│  USB Device Descriptor Summary  │
├───────────────────┬─────────────┤
│bDeviceClass       │ 239         │
├───────────────────┼─────────────┤
│bDeviceSubClass    │ 2           │
├───────────────────┼─────────────┤
│bDeviceProtocol    │ 1           │
├───────────────────┼─────────────┤
│bMaxPacketSize0    │ 64          │
├───────────────────┼─────────────┤
│idVendor           │ 0x303a      │
├───────────────────┼─────────────┤
│idProduct          │ 0x4001      │
├───────────────────┼─────────────┤
│bcdDevice          │ 0x100       │
├───────────────────┼─────────────┤
│iManufacturer      │ 0x1         │
├───────────────────┼─────────────┤
│iProduct           │ 0x2         │
├───────────────────┼─────────────┤
│iSerialNumber      │ 0x3         │
├───────────────────┼─────────────┤
│bNumConfigurations │ 0x1         │
└───────────────────┴─────────────┘
I (713) TinyUSB: TinyUSB Driver installed
I (719) pp: pp rom version: e7ae62f
I (722) net80211: net80211 rom version: e7ae62f
I (731) wifi_init: rx ba win: 6
I (731) wifi_init: tcpip mbox: 32
I (735) wifi_init: udp mbox: 6
I (738) wifi_init: tcp mbox: 6
I (742) wifi_init: tcp tx win: 5744
I (746) wifi_init: tcp rx win: 5744
I (751) wifi_init: tcp mss: 1440
I (754) wifi_init: WiFi IRAM OP enabled
I (759) wifi_init: WiFi RX IRAM OP enabled
I (764) phy_init: phy_version 670,b7bc9b9,Apr 30 2024,10:54:13
```

lijunru-hub, 2024-07-17:
USB Dongle 示例支持两种方式发送指令
1.（默认）通过 USB CDC 接口，需要将开发板的 USB 接口接入电脑，打开串口收发工具，与显示的串口相连接即可
2. 通过 UART 进行通信，默认为 RX:5, TX:4, UART_PORT:2, 需要将这几个引脚接入 UART 转 USB 工具。
推荐使用默认的 CDC 接口进行通信。

The USB Dongle example supports two methods for sending commands:

1. (Default) Through the USB CDC interface. Connect the development board's USB interface to a computer, open a serial communication tool, and connect to the displayed serial port.
2. Through UART communication. The default settings are RX:5, TX:4, UART_PORT:2. Connect these pins to a UART to USB converter.

It is recommended to use the default CDC interface for communication.

yunfusheng33, 2024-07-17:
我已经正常通信，感谢您的回答

closed 2024-07-17 by yunfusheng33