espressif/esp-iot-solution #383: usb_webcam example no longer working because no versions of espressif/tinyusb match >=0.15.0~10

ohault, 2024-07-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.2.2-2-v5.2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-S3-CAM WROOM

### Steps to reproduce.

c:\>git clone https://github.com/espressif/esp-iot-solution.git
c:\>cd esp-iot-solution\usb\device\usb_webcam
c:\esp-iot-solution\examples\usb\device\usb_webcam>idf.py set-target esp32s3

### Build Logs.

```plain
c:\>git clone https://github.com/espressif/esp-iot-solution.git
Cloning into 'esp-iot-solution'...
remote: Enumerating objects: 26311, done.
remote: Counting objects: 100% (26311/26311), done.
remote: Compressing objects: 100% (9820/9820), done.
remote: Total 26311 (delta 15251), reused 26090 (delta 15184), pack-reused 0
Receiving objects: 100% (26311/26311), 108.92 MiB | 6.38 MiB/s, done.
Resolving deltas: 100% (15251/15251), done.
Updating files: 100% (2750/2750), done.

c:\>cd esp-iot-solution\usb\device\usb_webcam

c:\esp-iot-solution\examples\usb\device\usb_webcam>idf.py set-target esp32s3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'C:\esp-iot-solution\examples\usb\device\usb_webcam\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig will be created.
Running cmake in directory C:\esp-iot-solution\examples\usb\device\usb_webcam\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=1 C:\esp-iot-solution\examples\usb\device\usb_webcam"...
-- Found Git: C:/Espressif/tools/idf-git/2.44.0/cmd/git.exe (found version "2.44.0.windows.1")
-- sdkconfig not exist
EYES_SHOW_TARGET is set to esp32s3
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Dependencies lock doesn't exist, solving dependencies.
Using component placed at C:\esp-iot-solution\examples\usb\device\usb_webcam\eyes_show for dependency eyes_show(*), specified in C:/esp-iot-solution/examples/usb/device/usb_webcam/main\idf_component.yml
Using component placed at C:\esp-iot-solution\components\usb\usb_device_uvc for dependency usb_device_uvc(1.1.*), specified in C:/esp-iot-solution/examples/usb/device/usb_webcam/main\idf_component.yml
...................Using component placed at C:\esp-iot-solution\examples\usb\device\usb_webcam\eyes_show for dependency eyes_show(*), specified in C:/esp-iot-solution/examples/usb/device/usb_webcam/main\idf_component.yml
Using component placed at C:\esp-iot-solution\components\usb\usb_device_uvc for dependency usb_device_uvc(1.1.*), specified in C:/esp-iot-solution/examples/usb/device/usb_webcam/main\idf_component.yml
CMake Error at C:/Espressif/frameworks/esp-idf-v5.2.2-2/tools/cmake/build.cmake:544 (message):
  ERROR: Because no versions of usb_device_uvc match >1.1.0,<1.2.0

   and usb_device_uvc (1.1.0) depends on espressif/tinyusb (>=0.15.0~10), usb_device_uvc (>=1.1.0,<1.2.0) requires espressif/tinyusb (>=0.15.0~10).

  So, because no versions of espressif/tinyusb match >=0.15.0~10

   and project depends on usb_device_uvc (1.1.*), version solving failed.

Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v5.2.2-2/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:25 (project)

...................-- Configuring incomplete, errors occurred!
See also "C:/esp-iot-solution/examples/usb/device/usb_webcam/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the C:\esp-iot-solution\examples\usb\device\usb_webcam\build\log\idf_py_stderr_output_576 and C:\esp-iot-solution\examples\usb\device\usb_webcam\build\log\idf_py_stdout_output_576

c:\esp-iot-solution\examples\usb\device\usb_webcam>
```

leeebo, 2024-07-09:
```
  ERROR: Because no versions of usb_device_uvc match >1.1.0,<1.2.0

   and usb_device_uvc (1.1.0) depends on espressif/tinyusb (>=0.15.0~10), usb_device_uvc (>=1.1.0,<1.2.0) requires espressif/tinyusb (>=0.15.0~10).

  So, because no versions of espressif/tinyusb match >=0.15.0~10
```

Seems like IDF component manager parse components' version failed like #352, we did have https://components.espressif.com/components/espressif/tinyusb/versions/0.15.0~10.

@ohault Which version of `idf-component-manager` are you using? please run `pip show idf-component-manager` after you export IDF environment.

leeebo, 2024-07-09:
I tested your case also using IDF `v5.2.2` with `idf-component-manager v1.5.3` , it works as normal. 🤔

ohault, 2024-07-09:
Hi @leeebo, 

```
C:\Espressif\frameworks\esp-idf-v5.2.2-2>pip show idf-component-manager
Name: idf-component-manager
Version: 1.5.3
Summary: The component manager for ESP-IDF
Home-page: https://github.com/espressif/idf-component-manager
Author: Sergei Silnov
Author-email: sergei.silnov@espressif.com
License: Apache License 2.0
Location: C:\Espressif\python_env\idf5.2_py3.11_env\Lib\site-packages
Requires: cachecontrol, click, colorama, contextlib2, packaging, pyparsing, pyyaml, requests, requests-file, requests-toolbelt, schema, six, tqdm, urllib3
Required-by:
```

This has been done on Microsoft Windows version 10.0.19045.4598

leeebo, 2024-07-09:
@ohault Thanks for your report, I did reproduce the issue on Windows OS with the same code and `idf-component-manager v1.5.3`. The IDF component manager team is working on the fix. cc @hfudev 

The current workaround is :
1.  (in IDF environment) upgrade `idf-component-manager` to v2.0 using `pip install idf-component-manager==2.0.0.dev1`
2. modify the `espidf.constraints.v5.2` file in your tools install path, change `idf-component-manager~=1.2` to `idf-component-manager>=1.2`
3. delete the `build` folder then retry

ohault, 2024-07-09:
Hi @leeebo, thank you. I can confirm that the proposed workaround is indeed working in the while of the fix for v1.5.3 release

ohault, 2024-07-10:
Now, I have this compile error with usb_webcam.

```
C:/esp-iot-solution/examples/usb/device/usb_webcam/main/usb_webcam_main.c:21:10: fatal error: bsp/esp-bsp.h: No such file or directory
   21 | #include "bsp/esp-bsp.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
```

Full log:

```
c:\esp-iot-solution\examples\usb\device\usb_webcam>idf.py build
Executing action: all (aliases: build)
Running ninja in directory C:\esp-iot-solution\examples\usb\device\usb_webcam\build
Executing "ninja all"...
[28/1016] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
factory,app,factory,0x10000,2M,
*******************************************************************************
[993/1016] Performing configure step for 'bootloader'
-- Found Git: C:/Espressif/tools/idf-git/2.44.0/cmd/git.exe (found version "2.44.0.windows.1")
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file C:/esp-iot-solution/examples/usb/device/usb_webcam/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Bootloader project name: "bootloader" version: 1
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/bootloader/subproject/main/ld/esp32s3/bootloader.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/bootloader/subproject/main/ld/esp32s3/bootloader.rom.ld
-- Components: boot_hooks bootloader bootloader_support efuse esp_app_format esp_bootloader_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: C:/esp-iot-solution/examples/usb/device/usb_webcam/bootloader_components/boot_hooks C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/bootloader C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/bootloader_support C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/efuse C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/esp_app_format C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/esp_bootloader_format C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/esp_common C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/esp_hw_support C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/esp_rom C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/esp_system C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/esptool_py C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/freertos C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/hal C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/log C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/bootloader/subproject/main C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/bootloader/subproject/components/micro-ecc C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/newlib C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/partition_table C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/soc C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/spi_flash C:/Espressif/frameworks/esp-idf-v5.2.2-2/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/esp-iot-solution/examples/usb/device/usb_webcam/build/bootloader
[998/1016] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/usb_webcam_main.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/usb_webcam_main.c.obj
ccache C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin\xtensa-esp32s3-elf-gcc.exe -DESP_PLATFORM -DIDF_VER=\"v5.2.2\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -DUNITY_INCLUDE_CONFIG_H -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D [... cut ...]
Loaded configuration 'C:/esp-iot-solution/examples/usb/device/usb_webcam/sdkconfig'
No changes to save (for 'C:/esp-iot-solution/examples/usb/device/usb_webcam/sdkconfig')
Loading defaults file C:/esp-iot-solution/examples/usb/device/usb_webcam/sdkconfig.defaults...
C:/esp-iot-solution/examples/usb/device/usb_webcam/sdkconfig.defaults:14 CONFIG_ESP32S3_SPIRAM_SUPPORT was replaced with CONFIG_SPIRAM
C:/esp-iot-solution/examples/usb/device/usb_webcam/sdkconfig.defaults:16 CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240 was replaced with CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240
```

ohault, 2024-07-12:
Hi @leeebo,

do you confirm it is a bug or is it a missing dependencies I should have installed manually ?

Is there a possible workaround ?

lijunru-hub, 2024-07-12:
Hi @ohault 
Before compiling, did you modify any configuration items in the menuconfig? Can you show me which components are in your management_components directory?

ohault, 2024-07-12:
> Hi @ohault Before compiling, did you modify any configuration items in the menuconfig? Can you show me which components are in your management_components directory?

Hi @lijunru-hub,

I didn't change menuconfig.
I can see the following three subdirectories in `c:\esp-iot-solution\examples\usb\device\usb_webcam\managed_components`

- espressif__cmake_utilities
- espressif__esp32-camera
- espressif__tinyusb

lijunru-hub, 2024-07-12:
This is where the problem lies: the package manager did not retrieve the **esp32_s3_eye** and **eyes_show** packages. For some reason, the following lines in the idf_component.yml did not take effect. I think you can delete them.

```
    rules:
      - if: "target == $EYES_SHOW_TARGET"
```

Is your development board an esp32-s3-eye? If not, you can set the macro CAMERA_MODULE to CAMERA_MODULE_CUSTOM or another appropriate value.

ohault, 2024-07-12:
@lijunru-hub,

I'm using an ESP32 S3 CAM ESP32 S3 WROOM 1 N16R8 Development Board with OV5640 Camera.

This module is working from [ESP Launchpad for esp-iot-solution](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://dl.espressif.com/AE/esp-iot-solution/config.toml) when I use the webcam application using ESP chipset type : esp32s3-esp32s3_eye

lijunru-hub, 2024-07-16:
Just try set `CAMERA_MODULE` to `CAMERA_MODULE_CUSTOM` and set your camera IO.

ohault, 2024-07-17:
Hi @lijunru-hub, with this last command applied, it is eventually working. Thank you.

To summarize : 
- A component version checking issue exists on Windows (). Workaround : https://github.com/espressif/esp-iot-solution/issues/383#issuecomment-2217162424
- The camera module support in the usb-webcam example application lags behind the one found in camera/video_stream_server. Workaround : https://github.com/espressif/esp-iot-solution/issues/383#issuecomment-2225145668 + `set CAMERA_MODULE = CAMERA_MODULE_CUSTOM`

Actions/suggestions:
- Fixing with bug on Windows : @hfudev
- Updating the usb-webcam example code on this repository to the same level as the camera modules support in https://github.com/espressif/esp-iot-solution/tree/master/examples/camera/video_stream_server.

Expected behavior:
```
- git clone https://github.com/espressif/esp-iot-solution
- cd esp-iot-solution/examples/usb/device/usb_webcam
- idf.py set-target esp32s3
- idf.py build
- idf.py flash
```
without any errors.

hfudev, 2024-09-03:
idf-component-manager 2.0 has released for a while. I would recommend close this issue. @leeebo

closed 2024-09-18 by leeebo