espressif/esp-iot-solution #384: ESP32-S3-A7670E-4G 开发板，usb 设备先上电 或者 执行 esp_reset() 后，_usbh_processing_task 无法找到 usb 设备进入无限等待。

codingau, 2024-07-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-A7670E-4G

### Power Supply used.

USB

### What is the expected behavior?

usb 设备先上电，可以正常执行 _usbh_processing_task ，并且联网。
esp_reset() 执行重启后，可以正常执行 _usbh_processing_task ，并且联网。

### What is the actual behavior?

_usbh_processing_task 任务，晚于 usb 上电的情况下，无法正常执行。

### Steps to reproduce.

两种测试方法
1，一直给 usb 上网卡设备供电。硬重启 esp32 或者 esp_reset() 执行重启。
2，iot_usbh.c 文件，在 xTaskCreatePinnedToCore(_usbh_processing_task, ...) 代码之前，睡 10 秒。
如下：
vTaskDelay(pdMS_TO_TICKS(10000));// 测试代码
xTaskCreatePinnedToCore(_usbh_processing_task, USBH_PROC_TASK_NAME, USBH_PROC_TASK_STACK_SIZE, (void*)port_instance,
        USBH_PROC_TASK_PRIORITY, &port_instance->usb_task_handle, USBH_PROC_TASK_CORE);

### Debug Logs.

```plain
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM3 instead...
--- esp-idf-monitor 1.4.0 on \\.\COM3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1918
load:0x403c9700,len:0x4
load:0x403c9704,len:0xe5c
load:0x403cc700,len:0x3028
entry 0x403c993c
I (27) boot: ESP-IDF v5.2.2-dirty 2nd stage bootloader
I (27) boot: compile time Jul 17 2024 12:04:08        
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) qio_mode: Enabling QIO for flash chip WinBond
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (45) boot.esp32s3: SPI Mode       : QIO
I (49) boot.esp32s3: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot:  3 storage          Unknown data     01 82 00110000 00100000
I (100) boot: End of partition table
I (105) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=28460h (164960) map
I (138) esp_image: segment 1: paddr=00038488 vaddr=3fc9cd00 size=0469ch ( 18076) load
I (142) esp_image: segment 2: paddr=0003cb2c vaddr=40374000 size=034ech ( 13548) load
I (146) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=92160h (598368) map
I (243) esp_image: segment 4: paddr=000d2188 vaddr=403774ec size=15740h ( 87872) load
I (270) boot: Loaded app from partition at offset 0x10000
I (270) boot: Disabling RNG early entropy source...
I (282) cpu_start: Multicore app
I (291) cpu_start: Pro cpu start user code
I (291) cpu_start: cpu freq: 240000000 Hz
I (291) cpu_start: Application information:
I (294) cpu_start: Project name:     usb_cdc_4g_module
I (300) cpu_start: App version:      1
I (304) cpu_start: Compile time:     Jul 17 2024 12:04:11
I (310) cpu_start: ELF file SHA256:  e096b0b6d...
I (316) cpu_start: ESP-IDF:          v5.2.2-dirty
I (321) cpu_start: Min chip rev:     v0.0
I (326) cpu_start: Max chip rev:     v0.99 
I (331) cpu_start: Chip rev:         v0.1
I (335) heap_init: Initializing. RAM available for dynamic allocation:
I (342) heap_init: At 3FCA4AB8 len 00044C58 (275 KiB): RAM
I (349) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (355) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (361) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (368) spi_flash: detected chip: winbond
I (372) spi_flash: flash io: qio
W (376) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (389) sleep: Configure to isolate all GPIO pins in sleep state
I (396) sleep: Enable automatic switching of GPIO sleep configuration
I (403) main_task: Started on CPU0
I (407) main_task: Calling app_main()
I (429) 4g_main: ====================================
I (429) 4g_main:      ESP 4G Cat.1 Wi-Fi Router
I (430) 4g_main: ====================================
I (435) modem_board: iot_usbh_modem, version: 0.2.1
I (441) modem_board: Force reset modem board....
I (446) gpio: GPIO[33]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (455) modem_board: Resetting modem using io=33, level=1
I (661) modem_board: Waiting for modem initialize ready
I (5661) USB_HCDC: iot_usbh_cdc, version: 0.2.2
I (5661) IOT_USBH: iot_usbh, version: 0.2.1
I (5691) IOT_USBH: USB Port=1 init succeed, fifo strategy=0
I (5691) IOT_USBH: USB Processing Start
I (5691) IOT_USBH: usb port start succeed
I (5691) IOT_USBH: Waiting USB Connection
I (5695) USB_HCDC: usbh cdc driver install succeed
I (5700) IOT_USBH: Port power: ON Succeed
I (5705) USB_HCDC: Waiting CDC Device Connection
I (5691) USB_HCDC: CDC task start
```

leeebo, 2024-07-17:
@codingau 检测不到连接并不是 ESP32S3 端的原因，你可以让 ESP32S3 控制 4G 的 reset 引脚，这样 ESP32S3 走初始化流程时，会把 4G 重置到可以检测的状态。

codingau, 2024-07-17:
> I (455) modem_board: Resetting modem using io=33, level=1

I (455) modem_board: Resetting modem using io=33, level=1
日志显示，我配置了 GPIO = 33，LEVEL = 0。
参考资料：[https://www.waveshare.net/w/upload/e/e7/ESP32-S3-A-SIM7670X-4G.pdf](url)
[image]

codingau, 2024-07-18:
解决了，物理上的解决。通过 gpio 控制外部继电器，当 usbh_modem_board.c 内部的 s_dce == NULL 时，外部继电器吸合断开电源一下，重启。

closed 2024-07-18 by codingau