espressif/esp-iot-solution #392: [esp_simplefoc] error: 'LEDC_USE_APB_CLK' was not declared in this scope; did you mean 'LEDC_USE_XTAL_CLK'

kutu, 2024-08-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3

### Espressif SoC revision.

esp32-c6

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

esp32-c6-devkitc-1

### Steps to reproduce.

1. `idf.py create-project stepper-test`
2. `cd stepper-test`
3. `idf.py set-target esp32c6`
4. `idf.py add-dependency "espressif/esp_simplefoc^0.2.0"`
5. `idf.py build`

### Build Logs.

```plain
$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory D:\projects\esp\stepper-test\build
Executing "ninja all"...
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
fatal: not a git repository (or any of the parent directories): .git
-- ccache will be used for faster recompilation
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32c6
Processing 4 dependencies:
[4/4] idf (5.3.0)
-- Project sdkconfig file D:/projects/esp/stepper-test/sdkconfig
-- Adding linker script E:/esp-idf/components/riscv/ld/rom.api.ld
-- App "stepper-test" version: 1
-- Adding linker script D:/projects/esp/stepper-test/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script D:/projects/esp/stepper-test/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script E:/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.ld
-- Adding linker script E:/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.api.ld
-- Adding linker script E:/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.rvfp.ld
-- Adding linker script E:/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.wdt.ld
-- Adding linker script E:/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.systimer.ld
-- Adding linker script E:/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.version.ld
-- Adding linker script E:/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.phy.ld
-- Adding linker script E:/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.coexist.ld
-- Adding linker script E:/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.net80211.ld
-- Adding linker script E:/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.pp.ld
-- Adding linker script E:/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.newlib.ld
-- Adding linker script E:/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.newlib-normal.ld
[... 28 lines cut ...]
-- Adding linker script E:/esp-idf/components/bootloader/subproject/main/ld/esp32c6/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_bootloader_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table riscv soc spi_flash
-- Component paths: E:/esp-idf/components/bootloader E:/esp-idf/components/bootloader_support E:/esp-idf/components/efuse E:/esp-idf/components/esp_app_format E:/esp-idf/components/esp_bootloader_format E:/esp-idf/components/esp_common E:/esp-idf/components/esp_hw_support E:/esp-idf/components/esp_rom E:/esp-idf/components/esp_system E:/esp-idf/components/esptool_py E:/esp-idf/components/freertos E:/esp-idf/components/hal E:/esp-idf/components/log E:/esp-idf/components/bootloader/subproject/main E:/esp-idf/components/bootloader/subproject/components/micro-ecc E:/esp-idf/components/newlib E:/esp-idf/components/partition_table E:/esp-idf/components/riscv E:/esp-idf/components/soc E:/esp-idf/components/spi_flash
-- Configuring done
-- Generating done
-- Build files have been written to: D:/projects/esp/stepper-test/build/bootloader
[1/1] cmd.exe /C "cd /D D:\projects\esp\stepper-test\build\bootloader\esp-idf\es...000 bootloader 0x0 D:/projects/esp/stepper-test/build/bootloader/bootloader.bin" Bootloader binary size 0x5500 bytes. 0x2b00 bytes (34%) free.
[1019/1032] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/stepper-test.cpp.obj
cc1plus.exe: warning: command-line option '-Wimplicit-function-declaration' is valid for C/ObjC but not for C++
[1024/1032] Building CXX object esp-idf/espressif__esp_simplefoc/CMakeFiles/__idf_espressif__esp_simplefoc.dir/port/esp/esp_hal_bldc_3pwm.cpp.obj
FAILED: esp-idf/espressif__esp_simplefoc/CMakeFiles/__idf_espressif__esp_simplefoc.dir/port/esp/esp_hal_bldc_3pwm.cpp.obj
ccache E:\idf-tools\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -DESP_PLATFORM -DIDF_VER=\"HEAD-HASH-NOTFOUND\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -ID:/projects/esp/stepper-test/build/config -ID:/projects/esp/stepper-test/managed_components/espressif__esp_simplefoc/port/esp/include -ID:/projects/esp/stepper-test/managed_components/espressif__esp_simplefoc/port/angle_sensor -IE:/esp-idf/components/newlib/platform_include -IE:/esp-idf/components/freertos/config/include -IE:/esp-idf/components/freertos/config/include/freertos -IE:/esp-idf/components/freertos/config/riscv/include -IE:/esp-idf/components/freertos/FreeRTOS-Kernel/include -IE:/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -IE:/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos -IE:/esp-idf/components/freertos/esp_additions/include -IE:/esp-idf/components/esp_hw_support/include -IE:/esp-idf/components/esp_hw_support/include/soc -IE:/esp-idf/components/esp_hw_support/include/soc/esp32c6 -IE:/esp-idf/components/esp_hw_support/dma/include -IE:/esp-idf/components/esp_hw_support/ldo/include -IE:/esp-idf/components/esp_hw_support/port/esp32c6/. -IE:/esp-idf/components/esp_hw_support/port/esp32c6/include -IE:/esp-idf/components/esp_hw_support/port/esp32c6/private_include -IE:/esp-idf/components/heap/include -IE:/esp-idf/components/log/include -IE:/esp-idf/components/soc/include -IE:/esp-idf/components/soc/esp32c6 -IE:/esp-idf/components/soc/esp32c6/include -IE:/esp-idf/components/hal/platform_port/include -IE:/esp-idf/components/hal/esp32c6/include -IE:/esp-idf/components/hal/include -IE:/esp-idf/components/esp_rom/include -IE:/esp-idf/components/esp_rom/include/esp32c6 -IE:/esp-idf/components/esp_rom/esp32c6 -IE:/esp-idf/components/esp_common/include -IE:/esp-idf/components/esp_system/include -IE:/esp-idf/components/esp_system/port/soc -IE:/esp-idf/components/esp_system/port/include/riscv -IE:/esp-idf/components/esp_system/port/include/private -IE:/esp-idf/components/riscv/include -IE:/esp-idf/components/lwip/include -IE:/esp-idf/components/lwip/include/apps -IE:/esp-idf/components/lwip/include/apps/sntp -IE:/esp-idf/components/lwip/lwip/src/include -IE:/esp-idf/components/lwip/port/include -IE:/esp-idf/components/lwip/port/freertos/include -IE:/esp-idf/components/lwip/port/esp32xx/include -IE:/esp-idf/components/lwip/port/esp32xx/include/arch -IE:/esp-idf/components/lwip/port/esp32xx/include/sys -IE:/esp-idf/components/driver/deprecated -IE:/esp-idf/components/driver/i2c/include -IE:/esp-idf/components/driver/touch_sensor/include -IE:/esp-idf/components/driver/twai/include -IE:/esp-idf/components/esp_pm/include -IE:/esp-idf/components/esp_ringbuf/include -IE:/esp-idf/components/esp_driver_gpio/include -IE:/esp-idf/components/esp_driver_pcnt/include -IE:/esp-idf/components/esp_driver_gptimer/include -IE:/esp-idf/components/esp_driver_spi/include -IE:/esp-idf/components/esp_driver_mcpwm/include -IE:/esp-idf/components/esp_driver_ana_cmpr/include -IE:/esp-idf/components/esp_driver_i2s/include -IE:/esp-idf/components/esp_driver_sdmmc/include -IE:/esp-idf/components/sdmmc/include -IE:/esp-idf/components/esp_driver_sdspi/include -IE:/esp-idf/components/esp_driver_sdio/include -IE:/esp-idf/components/esp_driver_dac/include -IE:/esp-idf/components/esp_driver_rmt/include -IE:/esp-idf/components/esp_driver_tsens/include -IE:/esp-idf/components/esp_driver_sdm/include -IE:/esp-idf/components/esp_driver_i2c/include -IE:/esp-idf/components/esp_driver_uart/include -IE:/esp-idf/components/vfs/inc [... cut ...]
D:/projects/esp/stepper-test/managed_components/espressif__esp_simplefoc/port/esp/esp_hal_bldc_3pwm.cpp:513:20: error: 'LEDC_USE_APB_CLK' was not declared in this scope; did you mean 'LEDC_USE_XTAL_CLK'?
  513 |         .clk_cfg = LEDC_USE_APB_CLK
      |                    ^~~~~~~~~~~~~~~~
      |                    LEDC_USE_XTAL_CLK
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the D:\projects\esp\stepper-test\build\log\idf_py_stderr_output_13740 and D:\projects\esp\stepper-test\build\log\idf_py_stdout_output_13740
```

YanKE01, 2024-08-14:
Analysing the IDF shows that `soc_periph_ledc_clk_src_legacy_t` in C6 does not contain `LEDC_USE_APB_CLK`：
```c
typedef enum {
    LEDC_AUTO_CLK = 0,                              /*!< LEDC source clock will be automatically selected based on the giving resolution and duty parameter when init the timer*/
    LEDC_USE_PLL_DIV_CLK = SOC_MOD_CLK_PLL_F80M,    /*!< Select PLL_F80M clock as the source clock */
    LEDC_USE_RC_FAST_CLK = SOC_MOD_CLK_RC_FAST,     /*!< Select RC_FAST as the source clock */
    LEDC_USE_XTAL_CLK = SOC_MOD_CLK_XTAL,           /*!< Select XTAL as the source clock */

    LEDC_USE_RTC8M_CLK __attribute__((deprecated("please use 'LEDC_USE_RC_FAST_CLK' instead"))) = LEDC_USE_RC_FAST_CLK,   /*!< Alias of 'LEDC_USE_RC_FAST_CLK' */
} soc_periph_ledc_clk_src_legacy_t;
```
You can try to use `LEDC_AUTO_CLK `to replace the original `LEDC_USE_APB_CLK`, we will fix this issue as soon as possible.

YanKE01, 2024-09-20:
We will fix the LEDC clock source issue for the ESP32C6 and ESP32H2 in version 1.0 of the `esp_simplefoc` component.

closed 2024-10-16 by zhanzhaocheng