espressif/esp-iot-solution #396: Zero detect: valid_count overflow

Roydudu, 2024-08-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

If we have a square wave signal at 50 Hz, the “valid_count” variable would increment on each positive and negative edge, increasing 100 times per second, causing the variable to overflow in almost 11 minutes. This would result in valid edges not generating the SIGNAL_VALID event when “valid_count” is less than “valid_times” because it resets to 0.

InfiniteYuan, 2024-09-19:
@Roydudu  This will seem to increase the latency of SIGNAL_VALID event firing. Is your application sensitive to this latency in events?

Roydudu, 2024-09-23:
My application is not affected by the latency of the SIGNAL_VALID event. However, I noticed that it can miss SIGNAL_VALID events every 11 minutes when the counter overflows and starts from zero.

closed 2024-10-28 by zhanzhaocheng