espressif/esp-iot-solution #397: Example usb_camera_mic_spk does not work, prints "Setting format: 0 NOT found"

frka1204, 2024-08-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I've tested my usb camera with the example provided in _esp-iot-solution/examples/usb/host/usb_camera_mic_spk_. However it seems that the program complains about that format 0 (UVC_FORMAT_JPEG) is an unsupported format for this camera. 
When the camera is connected to a computer with a usb port run in full speed mode frames are received.

_lsusb -v_ shows this:
[lsusb_log.txt](https://github.com/user-attachments/files/16632546/lsusb_log.txt)

The program in usb_camera_mic_spk prints this information:

```
--- esp-idf-monitor 1.4.0 on /dev/ttyUSB0 2000000 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (27) boot: ESP-IDF v5.3 2nd stage bootloader
I (27) boot: compile time Aug 13 2024 01:30:17
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.1
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (27) boot.esp32s3: SPI Mode       : DIO
I (28) boot.esp32s3: SPI Flash Size : 2MB
I (28) boot: Enabling RNG early entropy source...
I (28) boot: Partition Table:
I (28) boot: ## Label            Usage          Type ST Offset   Length
I (29) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (29) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (30) boot:  2 factory          factory app      00 00 00010000 00100000
I (30) boot: End of partition table
I (30) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=12960h ( 76128) map
I (45) esp_image: segment 1: paddr=00022988 vaddr=3fc94b00 size=02e94h ( 11924) load
I (47) esp_image: segment 2: paddr=00025824 vaddr=40374000 size=0a7f4h ( 42996) load
I (57) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=28adch (166620) map
I (87) esp_image: segment 4: paddr=00058b04 vaddr=4037e7f4 size=0627ch ( 25212) load
I (100) boot: Loaded app from partition at offset 0x10000
I (100) boot: Disabling RNG early entropy source...
V (101) mmap: after coalescing, 1 regions are left
I (101) cpu_start: Multicore app
D (101) cpu_start: Pro cpu up
D (101) cpu_start: Starting app cpu, entry point is 0x403754d0
0x403754d0: call_start_cpu1 at /home/fredrik/ESP-IDF/esp-idf-v5.3/components/esp_system/port/cpu_start.c:195

D (0) cpu_start: App cpu up
V CACHE_ERR: illegal error intr clr & ena mask is: 0x3f
V CACHE_ERR: core 1 access error intr clr & ena mask is: 0x1f
D (110) clk: RTC_SLOW_CLK calibration value: 3912371
V CACHE_ERR: illegal error intr clr & ena mask is: 0x3f
V CACHE_ERR: core 0 access error intr clr & ena mask is: 0x1f
D (111) cpu_start: calling init function: 0x420022dc on core: 0
0x420022dc: __esp_system_init_fn_init_efuse_check at /home/fredrik/ESP-IDF/esp-idf-v5.3/components/efuse/src/esp_efuse_startup.c:31

D (111) cpu_start: calling init function: 0x42002420 on core: 0
0x42002420: __esp_system_init_fn_init_show_cpu_freq at /home/fredrik/ESP-IDF/esp-idf-v5.3/components/esp_system/startup_funcs.c:58

I (111) cpu_start: Pro cpu start user code
I (112) cpu_start: cpu freq: 240000000 Hz
D (112) cpu_start: calling init function: 0x4200220c on core: 0
0x4200220c: __esp_system_init_fn_init_show_app_info at /home/fredrik/ESP-IDF/esp-idf-v5.3/components/esp_app_format/esp_app_desc.c:104

I (112) app_init: Application information:
I (112) app_init: Project name:     usb_camera_mic_spk
I (113) app_init: App version:      1
I (113) app_init: Compile time:     Aug 13 2024 02:34:39
I (113) app_init: ELF file SHA256:  10cb7478b...
I (114) app_init: ESP-IDF:          v5.3
D (114) cpu_start: calling init function: 0x420022f0 on core: 0
[... 259 lines cut ...]
	bInterfaceProtocol 0x0
criptorType 0x5ace 0
		bEndpointAddress 0x81	EP 1 IN
		bmAttributesetSize 780
		bInterval 1
	*** Interface descriptor ***
	bLengAlternateSetting 7
	bNumEndpoints 1
	bInterfaceClass 0xe (Videss 0x2
	bInterfaceProtocol 0x0
	iInterface 0
		*** Endpoint d		bEndpointAddress 0x81	EP 1 IN
		bmAttributes 0x5	ISOC
		wMaxterval 1
I (646) USB_STREAM: Actual VS Interface(MPS <= 47) USB_STREAM: 	Endpoint(ISOC) Addr = 0x81, MPS = 512
	ndpoint(ISOC) Addr = 0x81, MPS = 512
I (648) USB
M: Setting format: 0 NOT foundrmat: 0 NOT found
E (649) USB_STREAM: _STREAM: Frame: 0, width*height: 65535*65535, NOT found
: ./managed_components/espressif__usb_stream/usb_stream.c:3134 matching configurations
E (650) USB_STREAM: ENUM StaDESC, Failed
E (650) USB_STREAM: ENUM Stage CHECK_FUAM: USB enum failed, retrying in 200 ms...
W (872) USB_STREAM: USB enum failed, retry now

```

leeebo, 2024-08-16:
`Setting format: 0 NOT found` means your camera has no `MJPEG` format.

Only MJPEG cameras are supported

frka1204, 2024-08-16:
> `Setting format: 0 NOT found` means your camera has no `MJPEG` format.
> 
> Only MJPEG cameras are supported

Yes, but lsusb shows:
```
VideoStreaming Interface Descriptor:
        bLength                            30
        bDescriptorType                    36
        bDescriptorSubtype                  7 (FRAME_MJPEG)
        bFrameIndex                         4
        bmCapabilities                   0x00
          Still image unsupported
        wWidth                            320
        wHeight                           240
        dwMinBitRate                 36864000
        dwMaxBitRate                 36864000
        dwMaxVideoFrameBufferSize      153600
        dwDefaultFrameInterval         333333
        bFrameIntervalType                  1
        dwFrameInterval( 0)            333333
```

I can save a frame with:
`v4l2-ctl -d /dev/video28 --stream-mmap --stream-count=1 --stream-to=output.jpg`

leeebo, 2024-08-16:
@frka1204 That's because with ESP32-S3 the USB camera can only works in Fullspeed mode. But on your PC it works in Highspeed mode.

The configuration descriptors used by the devices are different for different rate modes

frka1204, 2024-08-16:
Ok, I thought it was enough to set _maximum-speed = "full-speed"_; in device tree to force full-speed but it seems that's not enough. Instead I disabled USB EHCI in device tree so that only USB OHCI is enabled which forces full-speed mode as shown:
```
[root@RV1126_RV1109:/]# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=, Driver=hub/4p, 480M
        |__ Port 3: Dev 3, If 0, Class=, Driver=r8152, 480M
        |__ Port 4: Dev 7, If 1, Class=, Driver=uvcvideo, 480M
        |__ Port 4: Dev 7, If 0, Class=, Driver=uvcvideo, 480M
```
Now the usb descriptors show that MJPEG isn't a supported format for full-speed mode:
```
Bus 001 Device 004: ID 0bda:5844  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 
  bDeviceSubClass         2 
  bDeviceProtocol         1 
  bMaxPacketSize0        64
  idVendor           0x0bda 
  idProduct          0x5844 
  bcdDevice           12.34
  iManufacturer           3 Generic
  iProduct                1 USB Camera
  iSerial                 2 200901010001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          324
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          4 USB Camera
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 
      bFunctionSubClass       3 
      bFunctionProtocol       0 
      iFunction               5 USB Camera
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 
      bInterfaceSubClass      1 
      bInterfaceProtocol      0 
      iInterface              5 USB Camera
      VideoControl Interface Descriptor:
        bLength                13
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.00
        wTotalLength           78
        dwClockFrequency       15.000000MHz
        bInCollection           1
        baInterfaceNr( 0)       1
      VideoControl Interface Descriptor:
        bLength                18
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
[... 258 lines cut ...]
          Usage Type               Data
        wMaxPacketSize     0x030c  1x 780 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       7
      bNumEndpoints           1
      bInterfaceClass        14 
      bInterfaceSubClass      2 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            5
          Transfer Type            Isochronous
          Synch Type               Asynchronous
          Usage Type               Data
        wMaxPacketSize     0x0320  1x 800 bytes
        bInterval               1

```

closed 2024-08-16 by frka1204