espressif/esp-iot-solution #400: iot_usbh_modem运行后esp32s3可以和4g模块通信成功，但是获取不到ip？

stronmind, 2024-09-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

我的环境是vscode，esp-idf relase5.3版本，iot_usbh_modem, version: 0.2.1。硬件是ESP32-S3-DevKitC-1+移远EC800M-CN。我的问题和#341很像。但是我这个模块ec800m是确定支持ppp协议的。这个模块在电脑上也验证可以使用数据流量，但是运行起来前面几个步骤通过AT指令的log可以看到都能通信成功而且返回成功，但是到这一步modem_board: Modem state STAGE_WAIT_IP, Start就没法成功获取ip,加了一些打印，一直打dhcp_coarse_tmr()。请问可能是什么原因？如何排查？下面是完整的开机到异常的log

--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM18 instead...
--- esp-idf-monitor 1.4.0 on \\.\COM18 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1918
load:0x403c8700,len:0x4   
load:0x403c8704,len:0xe5c 
load:0x403cb700,len:0x303c
entry 0x403c893c
I (27) boot: ESP-IDF GIT-NOTFOUND 2nd stage bootloader
I (27) boot: compile time Sep  3 2024 11:38:03        
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) qio_mode: Enabling default flash chip QIO
I (40) boot.esp32s3: Boot SPI Speed : 80MHz     
I (44) boot.esp32s3: SPI Mode       : QIO       
I (49) boot.esp32s3: SPI Flash Size : 8MB        
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot:  3 storage          Unknown data     01 82 00110000 00100000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=3454ch (214348) map
I (145) esp_image: segment 1: paddr=00044574 vaddr=3fc9e200 size=05238h ( 21048) load
I (149) esp_image: segment 2: paddr=000497b4 vaddr=40374000 size=06864h ( 26724) load
I (156) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=a9494h (693396) map
I (265) esp_image: segment 4: paddr=000f94bc vaddr=4037a864 size=138a8h ( 80040) load
I (291) boot: Loaded app from partition at offset 0x10000
I (291) boot: Disabling RNG early entropy source...
I (303) octal_psram: vendor id    : 0x0d (AP)
I (303) octal_psram: dev id       : 0x02 (generation 3)
I (303) octal_psram: density      : 0x03 (64 Mbit)
I (308) octal_psram: good-die     : 0x01 (Pass)
I (313) octal_psram: Latency      : 0x01 (Fixed)
I (318) octal_psram: VCC          : 0x01 (3V)
I (323) octal_psram: SRF          : 0x01 (Fast Refresh)
I (329) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (335) octal_psram: BurstLen     : 0x01 (32 Byte)
I (340) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (347) octal_psram: DriveStrength: 0x00 (1/1)
D (352) MSPI Timing: test nums: 1, test result: [id][good/bad][good_times]:
D (359) MSPI Timing: [0][good][1]
D (362) MSPI Timing: [1][good][1]
D (366) MSPI Timing: [2][bad][0] 
D (369) MSPI Timing: [3][good][1]
D (373) MSPI Timing: [4][good][1]
D (376) MSPI Timing: [5][good][1]
D (380) MSPI Timing: [6][good][1] 
D (383) MSPI Timing: [7][bad][0]
D (386) MSPI Timing: [8][bad][0]
D (390) MSPI Timing: [9][good][1]
D (393) MSPI Timing: [10][good][1] 
[... 792 lines cut ...]
D (77269) esp_netif_lwip: check: local, if=0x3fcc8ee8 fn=0x420a2bb4
D (77276) modem_board: Handling bits = 0100, stage = 7, retry = 5
D (77281) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3fcc8ee8
E (77289) modem_board: Modem state STAGE_WAIT_IP, retry 5, timeout !
D (77295) esp_netif_lwip: call api in lwip: ret=0x0, give sem
W (77302) modem_board: Retry From Start !
D (77307) event: running post IP_EVENT:7 with handler 0x42014e98 and context 0x3fcc8bac on loop 0x3fcb4acc
0x42014e98: _svfiprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp-elf/src/newlib/newlib/libc/stdio/svfiprintf.c:1391 (discriminator 3)

I (77322) modem_board: IP event! 7
I (77326) modem_board: Modem Disconnect from PPP Server
I (77312) modem_board: Modem state STAGE_SYNC, Start
D (77347) esp_modem_dce: esp_modem_dce_generic_command(20): Sending command:AT

W (77347) esp-modem-dte: Not support sending command in ppp mode, please exit ppp mode first
 response timeoutem_dce: esp_modem_dce_generic_command(26): Command:AT
I (77373) modem_board: Current timeouts: 1 and errors: 0
W (77373) modem_board: Try to exit ppp mode........
D (77374) esp_modem_dce: esp_modem_dce_generic_command(20): Sending command:+++

D (77382) USB_HCDC: cdc write itf: 0, buf = 0x3c0b9ee4, len = 3
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()

leeebo, 2024-10-08:
@stronmind 看起来拨号流程都是正常的，但是运营商没有返回 IP，请检查你的 SIM 卡确保有上网功能，另外确认一下是否需要特殊的 APN 配置

CM20201227, 2024-12-08:
你好，请问这个代码可以分享下吗？ESP32S3 USB驱动EC800M还是有很对疑惑，非常希望学习下

lijunru-hub, 2024-12-23:
You can try using the latest 4g_cdc example code and change the interface (itf) to 0x4. If there are issues with this itf communication, you can try other itf values.