espressif/esp-iot-solution #401: ADC based button and low power

mike-2020, 2024-09-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The document only mentions GPIO based button support opportimization for lower power. How about ADC based button?

lijunru-hub, 2024-09-05:
Unfortunately, the ADC peripheral cannot wake up the CPU. You can try increasing the polling interval to reduce power consumption.

mike-2020, 2024-09-06:
got it, thanks.

closed 2024-09-06 by mike-2020