espressif/esp-iot-solution #402: Button supports two instances in one app?

mike-2020, 2024-09-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Many internal variables of "button" are defined as static global variables. So, want to confirm if I can create two button instances in my app? one is for GPIO button, one is for ADC button.

lijunru-hub, 2024-09-06:
you can create any number of button for any types

mike-2020, 2024-09-06:
In such case, multiple "button" instances share the same timer, right?