espressif/esp-iot-solution #403: The macro definitions of IOT solution USB 4G module

JDingOneSelf, 2024-09-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Macro Definition of IOT solution USB 4G module IOT_USBH_MODEM_VER_MAJOR, IOT_USBH_MODEM_VER_MINOR, IOT_USBH_MODEM_VER_minor, IOT_USBH_MODEM_VER_PATCH is not defined in the file but can be used, how to implement it, I put the USB components in the components file of the local project, I need to define, is there a method to implement it like a routine
Error messages:
`components/iot_usbh_modem/src/usbh_modem_board.c:654:80: error: expected expression before ',' tokenESP_LOGI(TAG, "iot_usbh_modem, version: %d.%d.%d", IOT_USBH_MODEM_VER_MAJOR,IOT_USBH_MODEM_VER_MINOR,IOT_USBH_MODEM_VER_PATCH);`

leeebo, 2024-09-18:
@JDingOneSelf the [cmake_utilities](https://github.com/espressif/esp-iot-solution/blob/master/components/usb/iot_usbh_modem/idf_component.yml#L11) provides the CMake API [cu_pkg_define_version](https://github.com/espressif/esp-iot-solution/blob/master/components/usb/iot_usbh_modem/CMakeLists.txt#L18) to parse the component's version from [iot_usbh_modem/idf_component.yml](https://github.com/espressif/esp-iot-solution/blob/master/components/usb/iot_usbh_modem/idf_component.yml#L11) then add version related macros.

JDingOneSelf, 2024-09-20:
如果我把这几个组件移出managed_components，放在components中如何为这几个宏赋值呢？试过-DIOT_USBH_MODEM_VER_MAJOR ,提示宏重复定义

lijunru-hub, 2024-09-20:
直接在组件的 CmakeLists 里面添加 
cu_pkg_define_version(${CMAKE_CURRENT_LIST_DIR})
并在 idf_component.yml 里面添加对 camke_utilities 的依赖
  cmake_utilities: "0.*"
如果只是为了编译成功，可以把这个打印注释掉

JDingOneSelf, 2024-10-29:
谢谢！

closed 2024-10-29 by JDingOneSelf