espressif/esp-iot-solution #405: Video recorder crashes

mmadela, 2024-09-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-S3-Eye and ESP-S3-Korvo2

### Power Supply used.

USB

### What is the expected behavior?

the example works in every light conditions

### What is the actual behavior?

there is a watchdog reset, especially in dark environment

### Steps to reproduce.

1. Compile attached code
2. Run on a dev board
3. Boards restarts

### Debug Logs.

```plain
I (242) esp_image: segment 3: paddr=000e6204 vaddr=3fc9e1I (809) esp_psram: SPI SRAM memory test OK
I (818) cpu_start: Pro cpu start user code
I (818) cpu_start: cpu freq: 240000000 Hz
I (818) cpu_start: Application information:
I (821) cpu_start: Project name:     video_recorder
I (827) cpu_start: App version:      f7f9edd-dirty
I (832) cpu_start: Compile time:     Sep 11 2024 13:32:26
I (838) cpu_start: ELF file SHA256:  619ff5cb1...
I (843) cpu_start: ESP-IDF:          v5.2.2-dirty
I (849) cpu_start: Min chip rev:     v0.0
I (854) cpu_start: Max chip rev:     v0.99 
I (858) cpu_start: Chip rev:         v0.2
I (863) heap_init: Initializing. RAM available for dynamic allocation:
I (870) heap_init: At 3FCA5098 len 00044678 (273 KiB): RAM
I (876) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (883) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (889) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (895) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (903) spi_flash: detected chip: gd
I (907) spi_flash: flash io: qio
W (911) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
W (924) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (934) sleep: Configure to isolate all GPIO pins in sleep state
I (941) sleep: Enable automatic switching of GPIO sleep configuration
I (949) main_task: Started on CPU0
I (953) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (961) main_task: Calling app_main()
I (967) s3 ll_cam: DMA Channel=0
I (969) cam_hal: cam init ok
I (973) sccb: pin_sda 4 pin_scl 5
I (977) sccb: sccb_i2c_port=1
I (991) camera: Detected camera at address=0x30
I (993) camera: Detected OV2640 camera
I (993) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (1075) cam_hal: buffer_size: 16384, half_buffer_size: 1024, node_buffer_size: 1024, node_cnt: 16, total_cnt: 60
I (1075) cam_hal: Allocating 61440 Byte frame buffer in PSRAM
I (1081) cam_hal: Allocating 61440 Byte frame buffer in PSRAM
I (1087) cam_hal: Allocating 61440 Byte frame buffer in PSRAM
I (1093) cam_hal: Allocating 61440 Byte frame buffer in PSRAM
I (1099) cam_hal: Allocating 61440 Byte frame buffer in PSRAM
I (1107) cam_hal: Allocating 61440 Byte frame buffer in PSRAM
I (1113) cam_hal: Allocating 61440 Byte frame buffer in PSRAM
I (1119) cam_hal: Allocating 61440 Byte frame buffer in PSRAM
I (1125) cam_hal: cam config ok
I (1129) ov2640: Set PLL: clk_2x: 0, clk_div: 0, pclk_auto: 0, pclk_div: 8
I (1219) video_recorder: start to test fps
I (1989) video_recorder: fps=25.016026, image_average_size=13090
I (1989) file manager: Initializing SD card
I (1989) file manager: Using SDMMC peripheral
I (1995) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2003) gpio: GPIO[38]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2013) gpio: GPIO[40]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
Name: SD16G
Type: SDHC/SDXC
Speed: 20.00 MHz (limit: 20.00 MHz)
Size: 14772MB
CSD: ver=2, sector_size=512, capacity=30253056 read_bl_len=9
SSR: bus_width=1
I (2065) avi recorder: Starting an avi [/sdcard/recorde.avi]
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x40041a76
0x40041a76: hci_le_ltk_req_neg_reply_cmd_handler in ROM

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1918
load:0x403c9700,len:0x4
load:0x403c9704,len:0xe5c
load:0x403cc700,len:0x3028
entry 0x403c993c
I (26) boot: ESP-IDF v5.2.2-dirty 2nd stage bootloader
I (26) boot: compile time Sep 11 2024 13:32:31
I (26) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (48) boot.esp32s3: SPI Flash Size : 4MB
W (53) boot.esp32s3: PRO CPU has been reset by WDT.
W (59) boot.esp32s3: APP CPU has been reset by WDT.
I (64) boot: Enabling RNG early entropy source...
I (70) boot: Partition Table:
I (73) boot: ## Label            Usage          Type ST Offset   Length
I (81) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (88) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  2 factory          factory app      00 00 00010000 00100000
```

### More Information.

I’ve tried the video recorder example, the code crashes in a strange way, there is a watchdog reset, but typical information (like backtrace or coredump, if enabled) is not available.
I’ve tested the S3 Eye and S3 Korvo 2 board with 3 OV2640 cameras. Always the same problem, the crashes are somehow related to light conditions, if bright, then sometimes works, but if darken (or with lens cover), then fails every time.
I’ve tried also different SD cards, clock speeds, camera resolutions, etc. No changes.
The complete source (even with managed components) attached, however the alterations from original are just adjustments for the boards I have.
[video_recorder_full_source.zip](https://github.com/user-attachments/files/16963242/video_recorder_full_source.zip)

mmadela, 2024-09-14:
so the problem is inside 
`static int jpeg2avi_add_frame(jpeg2avi_data_t *j2a, uint8_t *data, uint32_t len)`

adding console output
```diff
--- a/components/avi_video_process/avi_recorder.c
+++ b/components/avi_video_process/avi_recorder.c
@@ -152,6 +152,7 @@ static int jpeg2avi_add_frame(jpeg2avi_data_t *j2a, uint8_t *data, uint32_t len)
             memcpy(&j2a->buffer[j2a->write_len], &frame_head, sizeof(AVI_CHUNK_HEAD)); // Copy the header data to the back of the current data to be processed
             j2a->write_len += sizeof(AVI_CHUNK_HEAD);
             int _len = CHUNK_SIZE - last_remain - sizeof(AVI_CHUNK_HEAD);
+            printf("_len %d\n", _len);
             memcpy(&j2a->buffer[j2a->write_len], data, _len);
             j2a->write_len += _len;
             data += _len;
```
gives after a while a **negative length**

and this is not the only problem in this function looking further
```diff
--- a/components/avi_video_process/avi_recorder.c
+++ b/components/avi_video_process/avi_recorder.c
@@ -175,6 +176,7 @@ static int jpeg2avi_add_frame(jpeg2avi_data_t *j2a, uint8_t *data, uint32_t len)
             // the remain is [last remain data + chunk_head + new frame data]
             memcpy(&j2a->buffer[j2a->write_len], &frame_head, sizeof(AVI_CHUNK_HEAD)); // add the avi_chunk_header
             j2a->write_len += sizeof(AVI_CHUNK_HEAD);
+            // see line 139,  var **remain** includes already j2a->write_len, so the line below might overflow the buffer twice in worst case
             memcpy(&j2a->buffer[j2a->write_len], data, remain - sizeof(AVI_CHUNK_HEAD)); // add the new frame data
             j2a->write_len += remain - sizeof(AVI_CHUNK_HEAD);
             remain = 0;
```

WangYuxin-esp, 2024-09-23:
This may be a hardware power supply issue. When the brightness of the environment decreases, the sensor requires a larger current to achieve greater gain.

mmadela, 2024-09-23:
The problem is clearly software problem, see the comment above
the only HW correlation is that the dark images are smaller and causes the bug to rise faster