espressif/esp-iot-solution #406: bootloader_support_plus @esp32s3 Not functioning properly

greentd, 2024-09-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.4-dev-2744-g59e1838270

### Espressif SoC revision.

ESP32S3FH4R2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32S3FH4R2

### Power Supply used.

External 3.3V

### What is the expected behavior?

I want to implement differential compression OTA on esp32s2fh4r2 to reduce the flash usage of OTA partition

### What is the actual behavior?

https://github.com/espressif/esp-iot-solution/tree/master/examples/ota/simple_ota_example       

Using this routine, the uncompressed bin file can be upgraded normally, but the compressed file will prompt after the upgrade and restart.     

E (122) esp_image: Checksum failed. Calculated 0xef read 0x0
E (128) boot: OTA app partition slot 1 is not bootable

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
I (27) boot: ESP-IDF v5.4-dev-2744-g59e1838270 2nd stage bootloader
I (27) boot: compile time Sep 13 2024 14:51:33
I (27) boot: Multicore bootloader
I (32) boot: chip revision: v0.1
I (35) boot: efuse block revision: v1.2
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (45) boot.esp32s3: SPI Mode       : DIO
I (49) boot.esp32s3: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 phy_init         RF data          01 01 0000c000 00001000
I (78) boot:  1 nvs              WiFi data        01 02 0000d000 00007000
[... 124 lines cut ...]
I (40) boot: efuse block revision: v1.2
I (44) boot.esp32s3: Boot SPI Speed : 80MHz
I (49) boot.esp32s3: SPI Mode       : DIO
I (54) boot.esp32s3: SPI Flash Size : 4MB
I (59) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 phy_init         RF data          01 01 0000c000 00001000
I (82) boot:  1 nvs              WiFi data        01 02 0000d000 00007000
I (90) boot:  2 otadata          OTA data         01 00 00014000 00002000
I (97) boot:  3 ota_0            OTA app          00 10 00020000 00130000
I (105) boot:  4 ota_1            OTA app          00 11 00150000 000a0000
I (112) boot: End of partition table
[boot] Custom bootloader message defined in the KConfig file.
E (122) esp_image: Checksum failed. Calculated 0xef read 0x0
E (128) boot: OTA app partition slot 1 is not bootable
I (134) esp_image: segment 0: paddr=00020020 vaddr=3c0b0020 size=2bd50h (179536) map
I (174) esp_image: segment 1: paddr=0004bd78 vaddr=3fc99c00 size=042a0h ( 17056) load
I (178) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=a08e8h (657640) map
I (296) esp_image: segment 3: paddr=000f0910 vaddr=3fc9dea0 size=0070ch (  1804) load
I (297) esp_image: segment 4: paddr=000f1024 vaddr=40374000 size=15b3ch ( 88892) load
I (330) boot: Loaded app from partition at offset 0x20000
I (330) boot: Disabling RNG early entropy source...
I (342) cpu_start: Multicore app
```

greentd, 2024-09-18:
I bought an ESP32C3 development board and ran it with esp-idf v5.07. I still get the error: 

E (6255) esp_image: Checksum failed. Calculated 0xef read 0x0
E (6255) simple_ota_example: Firmware upgrade failed

I (4525) example_connect: Got IPv6 event: Interface "example_netif_sta" address: fe80:0000:0000:0000:9e9e:6eff:fe09:ac68, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (4975) esp_netif_handlers: example_netif_sta ip: 192.168.113.129, mask: 255.255.255.0, gw: 192.168.113.1
I (4975) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.113.129
I (4985) example_common: Connected to example_netif_sta
I (4985) example_common: - IPv4 address: 192.168.113.129,
I (4995) example_common: - IPv6 address: fe80:0000:0000:0000:9e9e:6eff:fe09:ac68, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (5005) wifi:Set ps type: 0, coexist: 0

I (5005) simple_ota_example: Starting OTA example task
I (5015) simple_ota_example: Attempting to download update from https://192.168.113.118:8000/upload/test_hello.bin.xz.packed
I (5025) main_task: Returned from app_main()
I (5125) esp-x509-crt-bundle: Certificate validated
I (5465) esp_https_ota: Starting OTA...
I (5465) esp_https_ota: Writing to partition subtype 17 at offset 0x150000
E (6255) esp_image: Checksum failed. Calculated 0xef read 0x0
E (6255) simple_ota_example: Firmware upgrade failed

I also tested it on esp32c3 using esp-idf v5.14, and it also cannot be compressed and upgraded.

WangYuxin-esp, 2024-09-23:
Thank you for reporting the issue. Please check if option `CONFIG_BOOTLOADER_COMPRESSED_ENABLED` is enabled in the configuration menu.

closed 2024-09-23 by zhanzhaocheng

greentd, 2024-12-11:
我启用了CONFIG_BOOTLOADER_COMPRESSED_ENABLED=y，在C3上问题仍无法解决

WangYuxin-esp, 2024-12-11:
Please upgrade your esp-idf version to v5.1 and above. A normally linked program will not print checksum errors starting with `esp_image:`.