espressif/esp-iot-solution #408: add Wifi to my Stm32 USB device

maaamcube, 2024-09-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello everyone,
I want to add wifi to my Stm32 USB device and my PC(see image below), can i use this solution ?
Thanks.

[image]

leeebo, 2024-09-18:
@maaamcube Please refer the [esp-hosted](https://github.com/espressif/esp-hosted) solution

maaamcube, 2024-09-18:
@leeebo it can't work with usb