espressif/esp-iot-solution #410: ST7701 example bootloop

jonodav, 2024-10-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.1

### Espressif SoC revision.

esp32s3-20210327

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32S3 devkit module

### Power Supply used.

USB

### What is the expected behavior?

Build and run the example provided with ST7701 library from component registry.

### What is the actual behavior?

It crashes and bootloops.

### Steps to reproduce.

1. Build the example from the ST7701 library from component registry.
2. Flash the ESP32S3
3. It crashes

### Debug Logs.

```plain
I (872) lcd_panel.io.3wire_spi: Panel IO create success, version: 1.0.1
I (879) example: Install ST7701 panel driver
I (884) st7701: version: 1.1.0
I (1143) lv_port: LVGL buffer size: 93KB
I (1145) lv_port: Create LVGL task
I (1145) lv_port: Starting LVGL task
I (1160) lv_port: Success creating LVGL task
I (1160) example: Turn on LCD backlight
I (1160) example: Display LVGL demos
I (1348) main_task: Returned from app_main()
Guru Meditation Error: Core  / panic'ed (Cache disabled but cached memory region accessed).
Core  0 register dump:
PC      : 0x40056fd5  PS      : 0x00060634  A0      : 0x8203607a  A1      : 0x3fcc6c30
--- 0x40056fd5: memcpy in ROM
A2      : 0x3c11d066  A3      : 0x4036fff0  A4      : 0xffffffb8  A5      : 0x3c7dfca8
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000001  A11     : 0x00000002  A12     : 0x000001df  A13     : 0x3fcc6bb0
A14     : 0x000001b1  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x00000007 
EXCVADDR: 0x00000000  LBEG    : 0x40056fc5  LEND    : 0x40056fe7  LCOUNT  : 0x0ff93d36 
--- 0x40056fc5: memcpy in ROM
0x40056fe7: memcpy in ROM
Backtrace: 0x40056fd2:0x3fcc6c30 0x42036077:0x3fcc6c40 0x42035abb:0x3fcc6c90 0x42009c76:0x3fcc6cc0 0x42010b93:0x3fcc6ce0 0x4201120f:0x3fcc6dc0 0x4201556f:0x3fcc6e50 0x42009d29:0x3fcc6e80
--- Error: GetOverlappedResult failed (PermissionError(13, 'Access is denied.', None, 5))
--- Waiting for the device to reconnect--- 0x40056fd2: memcpy in ROM
0x42036077: rgb_panel_draw_bitmap at C:/Users/user/esp/v5.3.1/esp-idf/components/esp_lcd/rgb/esp_lcd_panel_rgb.c:773 (discriminator 12)
0x42035abb: esp_lcd_panel_draw_bitmap at C:/Users/user/esp/v5.3.1/esp-idf/components/esp_lcd/src/esp_lcd_panel_ops.c:37
0x42009c76: flush_callback at C:/Users/user/esp/rgb_avoid_tearing/main/lvgl_port.c:368
0x42010b93: call_flush_cb at C:/Users/user/esp/rgb_avoid_tearing/managed_components/lvgl__lvgl/src/core/lv_refr.c:1322
(inlined by) draw_buf_flush at C:/Users/user/esp/rgb_avoid_tearing/managed_components/lvgl__lvgl/src/core/lv_refr.c:1297
(inlined by) refr_area_part at C:/Users/user/esp/rgb_avoid_tearing/managed_components/lvgl__lvgl/src/core/lv_refr.c:796
0x4201120f: refr_area at C:/Users/user/esp/rgb_avoid_tearing/managed_components/lvgl__lvgl/src/core/lv_refr.c:678
(inlined by) refr_invalid_areas at C:/Users/user/esp/rgb_avoid_tearing/managed_components/lvgl__lvgl/src/core/lv_refr.c:616
(inlined by) _lv_disp_refr_timer at C:/Users/user/esp/rgb_avoid_tearing/managed_components/lvgl__lvgl/src/core/lv_refr.c:324
0x4201556f: lv_timer_exec at C:/Users/user/esp/rgb_avoid_tearing/managed_components/lvgl__lvgl/src/misc/lv_timer.c:313 (discriminator 2)
(inlined by) lv_timer_handler at C:/Users/user/esp/rgb_avoid_tearing/managed_components/lvgl__lvgl/src/misc/lv_timer.c:109 (discriminator 2)
0x42009d29: lvgl_port_task at C:/Users/user/esp/rgb_avoid_tearing/main/lvgl_port.c:506
```

Horion0415, 2024-10-08:
Hello. The tests on my side are running normally without any crashes. I’d like to confirm whether your pin configuration is correct. The crash issue seems quite unusual. Have you made any changes to the configuration or adjustments to the code?

jonodav, 2024-10-08:
My apologies, it appears that the issue may have been that there are two locations to set resolution - driver and LVGL. Display is a different resolution to the default and I hadn't changed resolution for LVGL. Seems to run fine now.

closed 2024-10-08 by jonodav