espressif/esp-iot-solution #415: esp-iot-solution/examples/usb/device /usb_lcd_display/ in esp32p4

Kwon5176, 2024-10-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x400455b6  RA      : 0x40044582  SP      : 0x4ff272e0  GP      : 0x4ff20180  
TP      : 0x4ff27420  T0      : 0x4fc19b4c  T1      : 0x0000000f  T2      : 0x00000000  
S0/FP   : 0x50000000  S1      : 0x00000003  A0      : 0x00000001  A1      : 0x00000000  
A2      : 0x00000000  A3      : 0xffff0000  A4      : 0x4f540000  A5      : 0x00000000  
A6      : 0x4003fefe  A7      : 0x4003fd26  S2      : 0x00000001  S3      : 0x00000000  
S4      : 0x4ff24000  S5      : 0x40043000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x482b11c0  T4      : 0x002a3000  T5      : 0x4ff306b8  T6      : 0x00000000  
MSTATUS : 0x00011880  MTVEC   : 0x4ff00003  MCAUSE  : 0x00000005  MTVAL   : 0x50000040  
MHARTID : 0x00000000  

Stack memory:
4ff272e0: 0x4003fefe 0x4003fd26 0x002a3000 0x00000054 0x00000001 0x00000000 0x00000001 0x40044582
4ff27300: 0x00000000 0x00000000 0x00000001 0x00000000 0x002a3000 0x00000003 0x4ff306b8 0x4004519c
4ff27320: 0x00000000 0x00000046 0x00000400 0x400429a8 0x4ff273cc 0x4ff284f0 0x4ff284a4 0x4003fd9c
4ff27340: 0x00000000 0x482b11c0 0x002a3000 0x4003f326 0x4003fefe 0x4003fd26 0x4003f2e0 0x4ff306b8
4ff27360: 0x002a3000 0x00000003 0x40064000 0x4000b54e 0x00000000 0x00000000 0x4ff306b8 0x00000000
4ff27380: 0x00000000 0x00000000 0x00000000 0x00000000 0x4ff273cc 0x050002d0 0x00000118 0x002a3000
4ff273a0: 0x00000000 0x03000002 0x00000450 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000
4ff273c0: 0x00000000 0x04000000 0x00000000 0x00000000 0x00000000 0x00000003 0x40062000 0x400582dc
4ff273e0: 0x00000000 0x00001388 0x00000003 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff27400: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff27420: 0x00000000 0x4ff2644c 0x4ff2644c 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5
4ff27440: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000160 0x4ff27240 0x00000000 0x4ff220d4 0x4ff220d4
4ff27460: 0x4ff27450 0x4ff220cc 0x00000018 0x00000000 0x00000000 0x4ff27450 0x00000000 0x00000001
4ff27480: 0x4ff2644c 0x6e69616d 0x00000000 0x00000000 0x00000000 0x00000000 0x4ff27440 0x00000001
4ff274a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x4ff24474 0x4ff244dc 0x4ff24544 0x00000000
4ff274c0: 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x4fc0aa82 0x00000000 0x00000000
4ff274e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff27500: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff27520: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff27540: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff27560: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff27580: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff275a0: 0x00000000 0xf25e874a 0x5ba750bb 0x4ff27448 0x00000600 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff275c0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff275e0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff27600: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff27620: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff27640: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: 968d59a6d

can i fix this?
i got esp32-p4-function-ev-board LCD set

lijunru-hub, 2024-10-25:
Could you please provide a detailed description of the steps to reproduce the issue and also specify your Windows version? It would be best to use the `idf.py` tool for flashing and monitoring, as this will help pinpoint the exact line of code where the issue occurs.

leeebo, 2024-12-24:
@Kwon5176 The issue will be closed, as it can not be reproduced. Feel free to reopen if you can provide a reproduction process

closed 2024-12-24 by leeebo