espressif/esp-iot-solution #416: Support for OpenAI Realtime API

nischalj10, 2024-10-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Voice powered edge devices are becoming more and more prevalant. Voice-to-Voice is a very powerful modality to interact with AI for masses. 

I would love if the support for OpenAI's Realtime Voice API can be added in esp-iot-solution

duketxl, 2024-10-29:
I'm adding realtime support based on esp-adf

nischalj10, 2024-11-04:
hey @duketxl! just curious if you were able to work on the direction. would love to see a reference implementation as I dive in to it as well!

baapar1101, 2024-12-29:
when done it ?

duketxl, 2024-12-29:
It is now possible to directly use OpenAI's demo.

duketxl, 2024-12-29:
https://github.com/openai/openai-realtime-embedded-sdk

duketxl, 2024-12-29:
[image]
We finished this audio board a year ago and are waiting for OpenAI to open the interface.