espressif/esp-iot-solution #420: The download speed of large files ( >=1.7MB ) based on 'ESP32S3 & USB-4G & PPP' Internet access scheme is very slow.

sky-gan, 2024-11-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

基于ESP32S3实现的USB 4G Modem PPP 拨号上网方案，遇到以下问题：
1、关闭WIFI路由功能，在esp32s3上实现了一个HTTP OTA升级功能，测试过程中发现HTTP 下载文件速度极其慢，中途容易触发接收超时.
2、开启WIFI路由功能，使用手机WIFI连接ESP32S3，然后手机下载问题1中同样大小文件，速度比较快.
是否是因为esp32s3同时处理PPP和TCP IP协议栈的调度太繁忙，导致卡顿？
能否提供一个解决方案或者建议？

Thanks.

lijunru-hub, 2024-11-12:
Are you referring to the following two scenarios?

1. Connecting to a 4G module and downloading a file via the 4G connection.
2. Connecting to a mobile hotspot and downloading a file through the hotspot.

If in the first scenario the speed is limited by the USB connection, then Wi-Fi should be significantly faster than USB. So, in theory, the second scenario should be faster than the first. From our tests, the download speed for the first scenario is around 4.2 Mbps down and 2.2 Mbps up.

sky-gan, 2024-11-12:
> Are you referring to the following two scenarios?
>
>     1. Connecting to a 4G module and downloading a file via the 4G connection.
>
>     2. Connecting to a mobile hotspot and downloading a file [...]
My test result  is lower than yours , could you provide your sample code ? 
Thanks !