espressif/esp-iot-solution #422: idf版本是5.1.4，编译esp-iot-solution\examples\get-started里的blink、button_power_save、knob_power_save都会报错

yangjinyiyiyi, 2024-11-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.4

### Espressif SoC revision.

ESP32S3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

esp32s3模组

### Steps to reproduce.

[image]

[image]

### More Information.

idf-component-manager 1.5.2和1.4.2都尝试过，都不行，还是报错

YanKE01, 2024-11-11:
Hi， @yangjinyiyiyi .我在idf 5.1.4与idf-component-manager 1.5.2下编译esp-iot-solution里的例程都是可以正常编译，你可以提供下你的esp-iot-solution的commit号，以便测试。
```shell
➜  knob_power_save git:(master) pip list
Package               Version
--------------------- ----------
annotated-types       0.7.0
astyle                3.4.10
bitarray              2.8.4
bitstring             4.1.4
CacheControl          0.13.1
certifi               2023.11.17
cffi                  1.16.0
cfgv                  3.4.0
charset-normalizer    3.3.2
click                 8.1.7
colorama              0.4.6
construct             2.10.69
contextlib2           21.6.0
cryptography          39.0.2
distlib               0.3.8
ecdsa                 0.18.0
esp-coredump          1.12.0
esp-debug-backend     1.0.3
esp-idf-kconfig       1.4.3
esp-idf-monitor       1.5.0
esp-idf-panic-decoder 1.2.1
esp-idf-size          1.6.0
esptool               4.7.0
filelock              3.13.1
freertos-gdb          1.0.3
identify              2.5.36
idf-component-manager 1.5.2
idna                  3.6
intelhex              2.3.0
jsonref               1.1.0
kconfiglib            14.1.0
markdown-it-py        3.0.0
mdurl                 0.1.2
mpremote              1.23.0
msgpack               1.0.7
nodeenv               1.9.1
packaging             24.1
pip                   24.2
platformdirs          4.2.2
pre-commit            3.7.1
pycparser             2.21
pydantic              2.8.2
pydantic_core         2.20.1
pydantic-settings     2.4.0
pyelftools            0.29
pygdbmi               0.9.0.2
Pygments              2.17.2
pyparsing             3.0.9
pyserial              3.5
python-dotenv         1.0.1
PyYAML                6.0.1
reedsolo              1.7.0
requests              2.32.3
requests-file         1.5.1
requests-toolbelt     1.0.0
rich                  13.7.0
schema                0.7.5
setuptools            71.0.0
six                   1.16.0
tqdm                  4.66.1
typing_extensions     4.12.2
urllib3               1.26.18
virtualenv            20.26.2
```
```shell
[... 97 lines cut ...]
-- Adding linker script /home/yanke/esp/esp-idf-51/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /home/yanke/esp/esp-idf-51/esp-idf/components/bootloader/subproject/main/ld/esp32s3/bootloader.ld
-- Adding linker script /home/yanke/esp/esp-idf-51/esp-idf/components/bootloader/subproject/main/ld/esp32s3/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: /home/yanke/esp/esp-idf-51/esp-idf/components/bootloader /home/yanke/esp/esp-idf-51/esp-idf/components/bootloader_support /home/yanke/esp/esp-idf-51/esp-idf/components/efuse /home/yanke/esp/esp-idf-51/esp-idf/components/esp_app_format /home/yanke/esp/esp-idf-51/esp-idf/components/esp_common /home/yanke/esp/esp-idf-51/esp-idf/components/esp_hw_support /home/yanke/esp/esp-idf-51/esp-idf/components/esp_rom /home/yanke/esp/esp-idf-51/esp-idf/components/esp_system /home/yanke/esp/esp-idf-51/esp-idf/components/esptool_py /home/yanke/esp/esp-idf-51/esp-idf/components/freertos /home/yanke/esp/esp-idf-51/esp-idf/components/hal /home/yanke/esp/esp-idf-51/esp-idf/components/log /home/yanke/esp/esp-idf-51/esp-idf/components/bootloader/subproject/main /home/yanke/esp/esp-idf-51/esp-idf/components/bootloader/subproject/components/micro-ecc /home/yanke/esp/esp-idf-51/esp-idf/components/newlib /home/yanke/esp/esp-idf-51/esp-idf/components/partition_table /home/yanke/esp/esp-idf-51/esp-idf/components/soc /home/yanke/esp/esp-idf-51/esp-idf/components/spi_flash /home/yanke/esp/esp-idf-51/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yanke/project/esp-iot-solution/examples/get-started/knob_power_save/build/bootloader
[109/110] Generating binary image from built executableesptool.py v4.7.0
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Generated /home/yanke/project/esp-iot-solution/examples/get-started/knob_power_save/build/bootloader/bootloader.bin
[110/110] cd /home/yanke/project/esp-iot-solution/examples/get-started/knob_power_save/build/bootload...e/yanke/project/esp-iot-solution/examples/get-started/knob_power_save/build/bootloader/bootloader.binBootloader binary size 0x51c0 bytes. 0x2e40 bytes (36%) free.
[946/947] Generating binary image from built executableesptool.py v4.7.0
Creating esp32s3 image...
Merged 3 ELF sections
Successfully created esp32s3 image.
Generated /home/yanke/project/esp-iot-solution/examples/get-started/knob_power_save/build/knob_power_save.bin
[947/947] cd /home/yanke/project/esp-iot-solution/examples/get-started/knob_power_save/build/esp-idf/...n /home/yanke/project/esp-iot-solution/examples/get-started/knob_power_save/build/knob_power_save.binknob_power_save.bin binary size 0x38240 bytes. Smallest app partition is 0x100000 bytes. 0xc7dc0 bytes (78%) free.

Project build complete. To flash, run this command:
/home/yanke/.espressif/python_env/idf5.1_py3.10_env/bin/python ../../../../../esp/esp-idf-51/esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32s3  write_flash --flash_mode dio --flash_size 2MB --flash_freq 80m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/knob_power_save.bin
or run 'idf.py -p (PORT) flash'
```

yangjinyiyiyi, 2024-11-12:
我的esp-iot-solution的commit号是6f426e5d3a4ca3130c540ffa26f4b72c7c39156e。你是不是使用代理了，你关闭代理试试，我没有代理，如果关闭代理也不行，麻烦给个方法，谢谢。
[image]
[image]

YanKE01, 2024-11-12:
Hi  @yangjinyiyiyi 。我在关闭代理且切换到这个comnmit也是正常编译通过。有确认esp_iot_solution无改动吗？是否执行`idf.py set-target xxx`获取组件。

查看工程目录下是否包含组件：
[image]

[image]

```shell
➜  blink git:(6f426e5d) pip list
Package               Version
--------------------- ----------
annotated-types       0.7.0
astyle                3.4.10
bitarray              2.8.4
bitstring             4.1.4
CacheControl          0.13.1
certifi               2023.11.17
cffi                  1.16.0
cfgv                  3.4.0
charset-normalizer    3.3.2
click                 8.1.7
colorama              0.4.6
construct             2.10.69
contextlib2           21.6.0
cryptography          39.0.2
distlib               0.3.8
ecdsa                 0.18.0
esp-coredump          1.12.0
esp-debug-backend     1.0.3
esp-idf-kconfig       1.4.3
esp-idf-monitor       1.5.0
esp-idf-panic-decoder 1.2.1
esp-idf-size          1.6.0
esptool               4.7.0
filelock              3.13.1
freertos-gdb          1.0.3
identify              2.5.36
idf-component-manager 1.5.2
idna                  3.6
intelhex              2.3.0
jsonref               1.1.0
kconfiglib            14.1.0
markdown-it-py        3.0.0
mdurl                 0.1.2
mpremote              1.23.0
msgpack               1.0.7
nodeenv               1.9.1
packaging             24.1
pip                   24.2
platformdirs          4.2.2
pre-commit            3.7.1
pycparser             2.21
pydantic              2.8.2
pydantic_core         2.20.1
pydantic-settings     2.4.0
pyelftools            0.29
pygdbmi               0.9.0.2
Pygments              2.17.2
pyparsing             3.0.9
pyserial              3.5
python-dotenv         1.0.1
PyYAML                6.0.1
reedsolo              1.7.0
requests              2.32.3
requests-file         1.5.1
requests-toolbelt     1.0.0
rich                  13.7.0
schema                0.7.5
setuptools            71.0.0
six                   1.16.0
[... 163 lines cut ...]
      |  ^~~~~~~
[937/962] Building C object esp-idf/boards/CMakeFiles/__idf_boards.dir/esp32s3_usb_otg_ev/board_esp32s3_usb_otg_ev.c.objIn file included from /home/yanke/project/github/esp-iot-solution/examples/common_components/boards/esp32s3_usb_otg_ev/board_esp32s3_usb_otg_ev.c:9:
/home/yanke/esp/esp-idf-51/esp-idf/components/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
In file included from /home/yanke/project/github/esp-iot-solution/examples/common_components/boards/esp32s3_usb_otg_ev/board_esp32s3_usb_otg_ev.c:10:
/home/yanke/esp/esp-idf-51/esp-idf/components/esp_adc/deprecated/include/esp_adc_cal.h:17:2: warning: #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h" [-Wcpp]
   17 | #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h"
      |  ^~~~~~~
[949/962] Building C object esp-idf/bus/CMakeFiles/__idf_bus.dir/8080_lcd_esp32s3.c.objIn file included from /home/yanke/project/github/esp-iot-solution/components/bus/include/i2s_lcd_driver.h:10,
                 from /home/yanke/project/github/esp-iot-solution/components/bus/8080_lcd_esp32s3.c:29:
/home/yanke/esp/esp-idf-51/esp-idf/components/driver/deprecated/driver/i2s.h:27:2: warning: #warning "This set of I2S APIs has been deprecated, please include 'driver/i2s_std.h', 'driver/i2s_pdm.h' or 'driver/i2s_tdm.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress leagcy driver deprecated warning' option under 'I2S Configuration' menu in Kconfig" [-Wcpp]
   27 | #warning "This set of I2S APIs has been deprecated, \
      |  ^~~~~~~
[961/962] Generating binary image from built executableesptool.py v4.7.0
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated /home/yanke/project/github/esp-iot-solution/examples/get-started/blink/build/blink.bin
[962/962] cd /home/yanke/project/github/esp-iot-solution/examples/get-started/blink/build/esp-idf/esptool_py && /home/y...tition_table/partition-table.bin /home/yanke/project/github/esp-iot-solution/examples/get-started/blink/build/blink.binblink.bin binary size 0x39110 bytes. Smallest app partition is 0x100000 bytes. 0xc6ef0 bytes (78%) free.

Project build complete. To flash, run this command:
/home/yanke/.espressif/python_env/idf5.1_py3.10_env/bin/python ../../../../../../esp/esp-idf-51/esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32s3  write_flash --flash_mode dio --flash_size 2MB --flash_freq 80m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/blink.bin
or run 'idf.py -p (PORT) flash'
```

yangjinyiyiyi, 2024-11-12:
我的工程目录下没有包含这个组件，esp_iot_solution是clone下来完整的没有改动过。执行idf.py set-target esp32s3获取组件就会出现问题，跟执行idf.py menuconfig一样。我是在win10环境下，我看你是Linux系统下编译的，环境是否不一样。
[image]
[image]
[image]

leeebo, 2024-11-12:
@yangjinyiyiyi 电脑没有联网？需要在网络环境下编译，因为包管理器会在编译过程中下载依赖的软件包

closed 2024-12-02 by leeebo