espressif/esp-iot-solution #425: [esp32-p4] uas-uart-bridge example

chegewara, 2024-11-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi team,

this example works good with S3, how about P4?

https://github.com/espressif/esp-iot-solution/blob/master/examples/usb/device/usb_uart_bridge

In theory it should be straightforward, assign pins and let it run, but:
- code seems to run correctly
- monitor other devices, all kind of esp32 devkits, works good
- there is problem to flash devkit or our custom PCB with it
- the same setup (wires, PCB etc) just with esp32-S3 instead of P4 works fine, i can flash and monitor

Thanks

lijunru-hub, 2024-11-19:
this example should update esp_tinyusb version which support p4.

lijunru-hub, 2024-11-19:
https://components.espressif.com/components/espressif/esp_tinyusb/versions/1.4.5

chegewara, 2024-11-19:
I am using esp_tinyusb which supports P4.