espressif/esp-iot-solution #427: Using the M307A USB4G module and the module has successfully registered for network access, but my phone cannot connect to the ESP32S3 hotspot for internet access

Rou-sir, 2024-11-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

cpu: esp32s3
modem: ML307A DCLN( usb 4g modem )

I am using the ESP32S3+M307A USB4G module and the module has successfully registered for network access, but my phone cannot connect to the ESP32S3 hotspot for internet access
log：
```
I (630) led_indicator: Indicator create successfully. type:GPIO mode, hardware_data:0x10, is_active_level_high:1, blink_lists:default
I (651) 4g_main: ====================================
I (652) 4g_main:      ESP 4G Cat.1 Wi-Fi Router
I (653) 4g_main: ====================================
I (659) modem_board: iot_usbh_modem, version: 0.2.1
I (665) modem_board: Force reset modem board....
I (670) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (679) modem_board: Resetting modem using io=13, level=0
I (885) modem_board: Waiting for modem initialize ready
I (5885) USB_HCDC: iot_usbh_cdc, version: 0.2.2
I (5885) IOT_USBH: iot_usbh, version: 0.2.1
I (5915) IOT_USBH: USB Port=1 init succeed, fifo strategy=0
I (5915) IOT_USBH: USB Processing Start
I (5915) IOT_USBH: usb port start succeed
I (5915) IOT_USBH: Waiting USB Connection
I (5919) USB_HCDC: usbh cdc driver install succeed
I (5924) IOT_USBH: Port power: ON Succeed
I (5929) USB_HCDC: Waiting CDC Device Connection
I (5915) USB_HCDC: CDC task start
I (6184) IOT_USBH: line 131 PORT_EVENT_CONNECTION
I (6284) IOT_USBH: Resetting Port
I (6344) IOT_USBH: Port reset succeed
I (6344) IOT_USBH: Getting Port Speed
I (6344) IOT_USBH: USB Speed: full-speed
I (6346) IOT_USBH: Pipe init succeed, addr: 00
I (6351) IOT_USBH: Set Device Addr = 1
I (6356) IOT_USBH: Set Device Addr Done
I (6360) IOT_USBH: get device desc
I (6365) IOT_USBH: get device desc, actual_num_bytes:26
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0xef
bDeviceSubClass 0x2
bDeviceProtocol 0x1
bMaxPacketSize0 64
idVendor 0x2ecc
idProduct 0x3012
bcdDevice 1.00
iManufacturer 1
iProduct 2
iSerialNumber 3
bNumConfigurations 1
I (6394) IOT_USBH: get short config desc
I (6399) IOT_USBH: get config desc, actual_num_bytes:16
I (6404) IOT_USBH: get full config desc
I (6409) IOT_USBH: get full config desc, actual_num_bytes:204
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 196
bNumInterfaces 5
bConfigurationValue 1
iConfiguration 0
bmAttributes 0xc0
[... 264 lines cut ...]
I (911718) modem_wifi: Get IP addr
E (913126) 4g_router_server: No auth header received
E (913128) 4g_router_server: rest_common_get_handler(901): not login yet
I (914494) 4g_main: Ping addr 110.242.68.66 Restart..
W (919494) 4g_main: From 110.242.68.66 icmp_seq=1 timeout

I (924494) 4g_main: Ping addr 110.242.68.66 Restart..
W (927403) USB_HCDC: The out buffer is too small, the data has been lost 65
W (927603) USB_HCDC: The out buffer is too small, the data has been lost 85
W (927803) USB_HCDC: The out buffer is too small, the data has been lost 90
W (928003) USB_HCDC: The out buffer is too small, the data has been lost 66
W (928203) USB_HCDC: The out buffer is too small, the data has been lost 65
W (928403) USB_HCDC: The out buffer is too small, the data has been lost 65
W (928603) USB_HCDC: The out buffer is too small, the data has been lost 71
W (928803) USB_HCDC: The out buffer is too small, the data has been lost 65
W (929003) USB_HCDC: The out buffer is too small, the data has been lost 65
W (929203) USB_HCDC: The out buffer is too small, the data has been lost 71
W (929403) USB_HCDC: The out buffer is too small, the data has been lost 80
W (929494) 4g_main: From 110.242.68.66 icmp_seq=1 timeout

W (929603) USB_HCDC: The out buffer is too small, the data has been lost 65
W (929803) USB_HCDC: The out buffer is too small, the data has been lost 65
W (930003) USB_HCDC: The out buffer is too small, the data has been lost 66
```
I suspect that the USB CDC endpoint address selection may be incorrect. I have tested several sets of scanned addresses, but none of them work. However, I do not have the ability to solve them and need your help

lijunru-hub, 2024-11-19:
Please confirm whether this 4G module supports the PPPoS communication method, and try testing whether the bulk IN and OUT endpoints on the same interface can communicate.

Rou-sir, 2024-11-19:
@lijunru-hub ML307A supports PPP mode, with USB AT and PPP endpoints being 0x81 and 0x0a,The module has successfully connected to the PPP server

tienat, 2024-11-26:
@Rou-sir I have the same problem with ESP32S3 and ML307A. Could you share me your sdkconfig file?

Rou-sir, 2024-12-07:
[sdkconfig.zip](https://github.com/user-attachments/files/18046363/sdkconfig.zip)
@tienat

lijunru-hub, 2024-12-09:
I have updated the relevant code for this example. Please try using the latest version. Note that the new version no longer requires specifying the USB port number; you only need to configure the correct interface number.
https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_cdc_4g_module

Rou-sir, 2024-12-23:
> I have updated the relevant code for this example. Please try using the latest version. Note that the new version no longer requires specifying the USB port number; you only need to configure the [...]

oh that's great!

lijunru-hub, 2024-12-23:
Test with the latest code. Can modifying the interface number make the 4G module work properly?

closed 2025-06-04 by leeebo

yxmstar, 2025-06-11:
> > 我已经更新了此示例的相关代码。请尝试使用最新版本。请注意，新版本不再需要指定 USB 端口号;您只需配置正确的接口编号。https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_cdc_4g_module
> 
> 哦，那太好了！

@Rou-sir 请问您最后配置的interface是多少？