espressif/esp-iot-solution #428: [Feature request] elf-loader for RISC-V based ESP32 (like C6) (IDFGH-14086)

martinssense, 2024-11-18:
### Is your feature request related to a problem?

Dynamically loading code during runtime can be useful.
There already is an [elf-loader for Xtensa](https://components.espressif.com/components/espressif/elf_loader) based ESP32, I cannot find a working solution for RISC-V based ESP32 (like C6 or H2) though.

### Describe the solution you'd like.

A working elf-loader for RISC-V based ESP32 would be awesome.
This way we could load "program modules" on RISC-V based ESP32 during runtime, without having to flash the firmware.

### Describe alternatives you've considered.

Zephyr has something called [LLEXT](https://docs.zephyrproject.org/latest/services/llext/index.html) which is basically an elf-loader. Using Zephyr is not the same as having the luxury of FreeRTOS though - as I understand there are also limitations of using Zephyr on ESP32.
There is also embedded wasm, but its overhead is huge, so it's not a viable solution for concurrent dynamically loaded code.
There are always interpreters like MicroPython, but be real: it's not the same.

metarutaiga, 2024-11-22:
I finish it this week for ESP32-C3.

https://github.com/NyankoLab/esp32c3-elf/blob/extra/elf_loader/src/esp_elf.c
https://github.com/NyankoLab/esp32c3-elf/blob/extra/elf_loader/src/arch/esp_elf_riscv.c

I changed some function that I can run it from file.

martinssense, 2024-11-25:
That's awesome! I'll take it as reference and try it out on C6 the upcoming days. Thanks a lot for the work :)