espressif/esp-iot-solution #429: esp_msc_ota api

fink-at-trmc-dk, 2024-11-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,
Regarding the esp_msc_ota API:
I have a project that already uses the MSC class, so I have had to make some modifications to make it work:
1 - Since I already have an MSC instance up and running, I had to create another task for the esp_msc_ota to live in, so I could send it events ( MSC_CONNECT).
2 - I had to the modify the esp_msc_ota_begin code so it initializes the handle BEFORE it waits for the MSC_CONNECT event, because otherwise I had not handle to which I could send the event.
3 - I also needed to create a semaphore (msc_read_semaphore) for the reader.

The above means that I have to do modifications in the module which is sub-optimal.

I would like to know if you have an API somewhere, that is separate from the MSC implementation? 
The current API ties the MSC to the OTA very close together and extra steps (see above) needs to be taken in order for it to work in an application that already has an MSC...

BR
Fink

lijunru-hub, 2024-11-19:
You can use the API `esp_msc_ota_set_msc_connect_state` to inform `msc_ota` that the USB drive has been inserted and the VFS file system has been initialized.  
This component supports separating the OTA part and the `msc_host` part.

Rasmus-Fink, 2024-11-19:
Hi and thanks for fast response.
That method `esp_msc_ota_set_msc_connect_state ` requires a valid handle to the esp_msc_ota instance.
That handle get initialized in esp_msc_ota_begin() - but only AFTER this line:
`EventBits_t bits = xEventGroupWaitBits(msc_ota->mscEventGroup, MSC_CONNECT, pdFALSE, pdFALSE, config->wait_msc_connect);`

- and I cannot use `esp_msc_ota_set_msc_connect_state`  to set the eventbit `MSC_CONNECT` until I have a valid handle.

It's the chicken and egg problem - unless I have misunderstood something and is using the API wrongly - which is entirely possible!

BR
Fink

lijunru-hub, 2024-12-24:
Yes, you are correct. I will make the necessary changes to this API promptly.

closed 2025-01-06 by zhanzhaocheng