espressif/esp-iot-solution #430: Trouble compiling   Esp_lcd_st7701  test_apps

Maxim88S, 2024-11-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Version: 3.1.0 Build id: 20241007-0706

### Espressif SoC revision.

ESP32-P4  v1.4

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-P4  v1.4 Fuction EV board

### Steps to reproduce.

Hi,

I'm having trouble compiling the example from this link: [esp_lcd_st7701](https://github.com/espressif/esp-iot-solution/tree/master/components/display/lcd/esp_lcd_st7701/test_apps)

I get the following error:

```
CMake Error at CMakeLists.txt:1 (idf_component_register):
  Unknown CMake command "idf_component_register".
```

Also, the compiler is complaining about the CMake version, even though I'm using a recent version. Here's the warning:

```
CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present. A line of code such as

    cmake_minimum_required(VERSION 3.24)
```

Additionally, I wanted to ask about the display initialization code in this example. It looks like this:

`{0xB0, (uint8_t []){0x00, 0x0B, 0x12, 0x0D, 0x10, 0x06, 0x02, 0x08, 0x07, 0x1F, 0x04, 0x11, 0x0F, 0x29, 0x31, 0x1E}, 16, 0}`

How is this line constructed? It doesn't seem to match either the common ESP32 LCD panel commands or the MIPI DSI standard, which requires lines to start with `0x39 (DCS Long Write)` or `0x05 (DCS Short Write)`. Can someone provide some insight into this?

### Build Logs.

```plain
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=D:\Espressif\frameworks\esp-idf-v5.3.1\tools\cmake\toolchain-esp32p4.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCCACHE_ENABLE=1 -DIDF_TARGET=esp32p4 D:\PRJ_2024\KB-RUEL\ESP32\Test_11\st7701
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at D:/Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/utilities.cmake:374 (list):
  Policy CMP0007 is not set: list command no longer ignores empty elements.
  Run "cmake --help-policy CMP0007" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.  List has value =
  [-march=rv32imafc_zicsr_zifencei_xesppie;-mabi=ilp32f;].
Call Stack (most recent call first):
  D:/Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/toolchain-esp32p4.cmake:10 (remove_duplicated_flags)
  D:/Espressif/tools/cmake/3.24.0/share/cmake-3.24/Modules/CMakeDetermineSystem.cmake:124 (include)
  CMakeLists.txt
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at D:/Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/utilities.cmake:374 (list):
  Policy CMP0007 is not set: list command no longer ignores empty elements.
  Run "cmake --help-policy CMP0007" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.  List has value =
  [-march=rv32imafc_zicsr_zifencei_xesppie;-mabi=ilp32f;].
Call Stack (most recent call first):
  D:/Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/toolchain-esp32p4.cmake:12 (remove_duplicated_flags)
  D:/Espressif/tools/cmake/3.24.0/share/cmake-3.24/Modules/CMakeDetermineSystem.cmake:124 (include)
  CMakeLists.txt
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at D:/Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/utilities.cmake:374 (list):
  Policy CMP0007 is not set: list command no longer ignores empty elements.
  Run "cmake --help-policy CMP0007" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.  List has value =
  [-march=rv32imafc_zicsr_zifencei_xesppie;-mabi=ilp32f;].
Call Stack (most recent call first):
  D:/Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/toolchain-esp32p4.cmake:14 (remove_duplicated_flags)
  D:/Espressif/tools/cmake/3.24.0/share/cmake-3.24/Modules/CMakeDetermineSystem.cmake:124 (include)
  CMakeLists.txt
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at D:/Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/utilities.cmake:374 (list):
  Policy CMP0007 is not set: list command no longer ignores empty elements.
  Run "cmake --help-policy CMP0007" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.  List has value =
  [-nostartfiles;-march=rv32imafc_zicsr_zifencei_xesppie;-mabi=ilp32f;--specs=nosys.specs;;;;;;;;;;;;;;;;;;;;;;;;;].
Call Stack (most recent call first):
  D:/Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/toolchain-esp32p4.cmake:17 (remove_duplicated_flags)
  D:/Espressif/tools/cmake/3.24.0/share/cmake-3.24/Modules/CMakeDetermineSystem.cmake:124 (include)
  CMakeLists.txt
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at D:/Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/utilities.cmake:374 (list):
  Policy CMP0007 is not set: list command no longer ignores empty elements.
  Run "cmake --help-policy CMP0007" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.  List has value =
  [-nostartfiles;-march=rv32imafc_zicsr_zifencei_xesppie;-mabi=ilp32f;--specs=nosys.specs;;;;;;;;;;;;;;;;;;;;;;;;;-nostartfiles;-march=rv32imafc_zicsr_zifencei_xesppie;-mabi=ilp32f;--specs=nosys.specs].
Call Stack (most recent call first):
  D:/Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/toolchain-esp32p4.cmake:17 (remove_duplicated_flags)
  build/CMakeFiles/3.24.0/CMakeSystem.cmake:6 (include)
  CMakeLists.txt
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:1 (idf_component_register):
  Unknown CMake command "idf_component_register".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.24)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
cmake --build . -- -v
ninja: error: loading 'build.ninja'
```

Horion0415, 2024-11-20:
Thank you for your question. You should compile the component's test_case instead of the component itself. Specifically, you should compile: https://github.com/espressif/esp-iot-solution/tree/master/components/display/lcd/esp_lcd_st7701/test_apps.

Regarding the initialization parameters, you are correct that in DCS, a Long Write is 0x39, and a Short Write is 0x05. However, in ESP's MIPI-DSI driver, users do not need to specify these commands manually. The driver will automatically send 0x39 or 0x05 based on the length of your parameters. You only need to write the registers and their corresponding parameters in the initialization settings. For further details, you can consult the display manufacturer.

Horion0415, 2024-11-21:
Users do not need to worry about the specific implementation of the driver. You only need to update the initialization parameters and timing parameters of your display. Of course, if you want to learn more details, you can refer to the ST7701 datasheet to understand the functionality of some display registers (https://dl.espressif.com/AE/esp-iot-solution/ST7701S_SPEC_%20V1.4.pdf). As for the MIPI-DSI driver, you can refer to the DSI driver section in ESP-IDF (https://github.com/espressif/esp-idf/tree/master/components/esp_lcd/dsi).

Maxim88S, 2024-11-26:
Thank you for your reply, you helped me a lot.
The display worked.

closed 2024-11-26 by Maxim88S