espressif/esp-iot-solution #432: ESP_SimpleFOC 6PWM driver

HirenDayro017, 2024-11-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

How soon can we get esp_hal_bldc_6pwm driver for a bldc driver that needs 6pwm?

YanKE01, 2024-11-24:
Support for the 6PWM driver will begin soon. The 6PWM driver can only run on chips that support `MCPWM`, and it will take some time to test it.

HirenDayro017, 2024-11-25:
ok, thanks. That's great to hear.

dayro-ram, 2024-11-25:
@YanKE01 , I have tried to make a library for 6PWM but not sure why it doesn't work. If you could debug the code, that would be great, as we are running late on key milestones. Kindly note this is based on esp_hal_bldc_3pwm library files.
Here is my esp_hal_bldc_6pwm.h
/*
 * SPDX-FileCopyrightText: 2023-2024 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#pragma once

#include <vector>
#include "BLDCDriver.h"
#include "foc_utils.h"
#include "time_utils.h"
#include "defaults.h"
#include "drivers/hardware_api.h"
#include "driver/ledc.h"

#ifdef CONFIG_SOC_MCPWM_SUPPORTED
#include "driver/mcpwm_prelude.h"
#endif

#define _PWM_FREQUENCY 20000                           /*!< default frequency of MCPWM */
#define _PWM_FREQUENCY_MAX 50000                       /*!< Max frequency of MCPWM */
#define _PWM_TIMEBASE_RESOLUTION_HZ (10 * 1000 * 1000) /*!< Resolution of MCPWM */
#define _LEDC_TIMER LEDC_TIMER_0                       /*!< Timer od LEDC */
#define _LEDC_MODE LEDC_LOW_SPEED_MODE                 /*!< Speed Mode of LEDC */
#define _LEDC_DUTY_RES LEDC_TIMER_9_BIT                /*!< Resolution of LEDC */
#define _LEDC_DUTY (511)                               /*!< Max duty of LEDC */
#define _LEDC_FREQUENCY (20 * 1000)                    /*!< Frequency of LEDC */

enum class DriverMode {
    mcpwm = 0, /*!< Foc hardware drive mode：MCPWM */
    ledc = 1,  /*!< Foc hardware drive mode：LEDC */
};

class BLDCDriver6PWM : public BLDCDriver {
public:
    /**
     * @brief Construct a new BLDCDriver6PWM object
     * @param phA_h A phase pwm pin
     * @param phA_l A phase pwm pin
     * @param phB_h B phase pwm pin
     * @param phB_l A phase pwm pin
     * @param phC_h C phase pwm pin
     * @param phC_l A phase pwm pin
     * @param en1 enable pin (optional input)
     * @param en2 enable pin (optional input)
     * @param en3 enable pin (optional input)
     */
    BLDCDriver6PWM(int phA_h,int phA_l, int phB_h,int phB_l, int phC_h, int phC_l, int en1 = NOT_SET);

    /**
     * @brief Motor hardware init function.
     *
     * @return
     *     - 0 Failed
     *     - 1 Success
     */
    int init() override;

#ifdef CONFIG_SOC_MCPWM_SUPPORTED
    /**
     * @brief Motor hardware init function, using MCPWM.
     *
     * @param _mcpwm_group mcpwm group, such as 0
     * @return
     *     - 0 Failed
     *     - 1 Success
     */
[... 682 lines cut ...]
    if (!initialized) {
        return;
    }

    switch (driverMode) {
    case DriverMode::ledc:
        ESP_ERROR_CHECK(ledc_set_duty(_LEDC_MODE, static_cast<ledc_channel_t>(ledc_channels[0]), (uint32_t)((mcpwm_period * dc_a))));
        ESP_ERROR_CHECK(ledc_update_duty(_LEDC_MODE, static_cast<ledc_channel_t>(ledc_channels[0])));
        ESP_ERROR_CHECK(ledc_set_duty(_LEDC_MODE, static_cast<ledc_channel_t>(ledc_channels[1]), (uint32_t)((mcpwm_period * dc_b))));
        ESP_ERROR_CHECK(ledc_update_duty(_LEDC_MODE, static_cast<ledc_channel_t>(ledc_channels[1])));
        ESP_ERROR_CHECK(ledc_set_duty(_LEDC_MODE, static_cast<ledc_channel_t>(ledc_channels[2]), (uint32_t)((mcpwm_period * dc_c))));
        ESP_ERROR_CHECK(ledc_update_duty(_LEDC_MODE, static_cast<ledc_channel_t>(ledc_channels[2])));
        break;

    case DriverMode::mcpwm:
        // printf("\n Duty Cycle Ua : %ld Ub : %ld Uc : %ld", (uint32_t)((mcpwm_period * dc_a)), (uint32_t)((mcpwm_period * dc_b)), (uint32_t)((mcpwm_period * dc_c)));
        ESP_ERROR_CHECK(mcpwm_comparator_set_compare_value(comparator[0], (uint32_t)((mcpwm_period * dc_a))));
        ESP_ERROR_CHECK(mcpwm_comparator_set_compare_value(comparator[1], (uint32_t)((mcpwm_period * dc_b))));
        ESP_ERROR_CHECK(mcpwm_comparator_set_compare_value(comparator[2], (uint32_t)((mcpwm_period * dc_c))));
        break;
        
    default:
        break;
    }
}

dayro-ram, 2024-11-26:
@YanKE01 Thanks for your quick response. We will try following your suggestion to keep the 6PWM library in a separate component folder. For now we got 6PWM working by using Arduino as component. We look forward to get 6PWM working library soon. I have shared the code with a hope that the code help expedite the development. Thanks again.

closed 2024-11-28 by zhanzhaocheng

YanKE01, 2024-11-28:
Hi @dayro-ram @HirenDayro017 . Thank you for your attention to [esp_simplefoc](https://components.espressif.com/components/espressif/esp_simplefoc/versions/1.1.0?language=en). Support for the 6PWM mode has been added in version `1.1.0`. Our next step will be to synchronize esp_simplefoc with the upstream version `2.3.4`. We welcome your feedback on usage.

dayro-ram, 2024-11-29:
HI @YanKE01 Thanks for quick support. We have tested foc_velocity_control example by changing necessary object instantiations & tested. The code is working. We will explore more and provide update if we encounter any issues.

HirenDayro017, 2024-12-06:
Hi @YanKE01 ,I tried the foc_knob_example with my "SparkFun IoT Brushless Motor Driver" board. Here is the code snippet:
Can you help me with what is the issue with this? The same motor with similar configuration works fine with Arduino Simple FOC Library. 

/**********************************************************************************************************************/

/*update motor parameters based on hardware design*/
BLDCDriver6PWM driver = BLDCDriver6PWM(16, 17, 18, 23, 19,33,32,0);
BLDCMotor motor = BLDCMotor(7);
TMAG5273 tmag = TMAG5273(I2C_NUM_0, GPIO_NUM_22, GPIO_NUM_21);
uint8_t conversionAverage = TMAG5273_X32_CONVERSION;
uint8_t magneticChannel = TMAG5273_XYX_ENABLE;
uint8_t angleCalculation = TMAG5273_XY_ANGLE_CALCULATION;

void tmag_init(void)
{
    esp_err_t ret;
    tmag.init();

    ret = tmag.setConvAvg(conversionAverage);
    if(ret != ESP_OK){
        ESP_LOGE(TAG, "Conversion Average not set");
    }else{
        ESP_LOGI(TAG, "Conversion Average set");
    }
    
    ret = tmag.setMagneticChannel(magneticChannel);
    if(ret != ESP_OK){
        ESP_LOGE(TAG,"Magnetic Channel not set");
    }else{
        ESP_LOGI(TAG, "Magnetic Channel set");
    }
    
    ret = tmag.setAngleEn(angleCalculation);
    if(ret != ESP_OK){
        ESP_LOGE(TAG,"Angle reading not enabled");
    }else{
        ESP_LOGI(TAG, "Angle reading enabled");
    }    
}

/*Motor initialization*/
void motor_init(void)
{
    SimpleFOCDebug::enable();
    Serial.begin(115200);

    tmag_init();
    motor.LPF_angle = LowPassFilter(0.02);
    motor.linkSensor(&tmag);
    driver.voltage_power_supply = 5;
    driver.voltage_limit = 4;

#if USING_MCPWM
    driver.init();
#else
    driver.init({LEDC_CHAN_0, LEDC_CHAN_1, LEDC_CHAN_2});
#endif

    motor.linkDriver(&driver);

    motor.foc_modulation = SpaceVectorPWM;
    motor.controller = MotionControlType::torque;

    motor.PID_velocity.P = 4.0;
    motor.PID_velocity.I = 0.0;
    motor.PID_velocity.D = 0.04;
    motor.PID_velocity.output_ramp = 10000;
    motor.PID_velocity.limit = 10;

    motor.useMonitoring(Serial);
[... 152 lines cut ...]
I (332215) FOC_Knob_Example: foc_knob_l_lim_cb
I (332423) FOC_Knob_Example: foc_knob_h_lim_cb
I (332537) FOC_Knob_Example: foc_knob_l_lim_cb
I (332641) FOC_Knob_Example: foc_knob_h_lim_cb
I (332723) FOC_Knob_Example: foc_knob_l_lim_cb
I (333521) FOC_Knob_Example: mode: 5
I (334179) FOC_Knob_Example: foc_knob_h_lim_cb
I (335549) FOC_Knob_Example: foc_knob_h_lim_cb
I (335559) FOC_Knob_Example: foc_knob_h_lim_cb
I (335605) FOC_Knob_Example: foc_knob_l_lim_cb
I (335729) FOC_Knob_Example: foc_knob_h_lim_cb
I (336826) FOC_Knob_Example: mode: 6
I (336851) FOC_Knob_Example: foc_knob_h_lim_cb
I (336857) FOC_Knob_Example: foc_knob_l_lim_cb
I (337379) FOC_Knob_Example: foc_knob_h_lim_cb
I (337485) FOC_Knob_Example: foc_knob_l_lim_cb
I (337899) FOC_Knob_Example: foc_knob_h_lim_cb
I (337949) FOC_Knob_Example: foc_knob_l_lim_cb
I (339166) FOC_Knob_Example: mode: 0

/**********************************************************************************************************************/

Here is how it behaves, the motor keeps hunting.

https://github.com/user-attachments/assets/ea3bc311-0d61-4d53-b339-4af14dd18be7

YanKE01, 2024-12-06:
Hi, @HirenDayro017 .From the log, it seems to be an error in the pole number estimation. Please fill in the correct pole number based on the specific motor parameters. I have encountered a situation before where the distance between the magnetic encoder and the magnetic ring was too far, which led to an incorrect pole number estimation. In any case, you can check whether the magnetic encoder is functioning properly.

HirenDayro017, 2024-12-06:
Hi @YanKE01 , have a look at the log from the code in which I have added **SimpleFOC as Arduino Component**. The code works pretty fine in this case.

Simple FOC as Arduino Component:

"**components/arduino/libraries/Simple_FOC/src/**"

Here is the same configuration:

#define DRIVER_FAULT_GPIO       12
#define DRIVER_UH_GPIO          16
#define DRIVER_UL_GPIO          17
#define DRIVER_VH_GPIO          18
#define DRIVER_VL_GPIO          23
#define DRIVER_WH_GPIO          19
#define DRIVER_WL_GPIO          33

using namespace std;
static const char *TAG = "Knob_demo";

BLDCMotor motor = BLDCMotor(7);
BLDCDriver6PWM driver = BLDCDriver6PWM(DRIVER_UH_GPIO, DRIVER_UL_GPIO, DRIVER_VH_GPIO, DRIVER_VL_GPIO, DRIVER_WH_GPIO, DRIVER_WL_GPIO);

/**********************************************************************************************************************/

PS D:\Dayro\Knob> & set IDF_PATH='C:/Espressif/frameworks/esp-idf-v5.1.5/'
PS D:\Dayro\Knob> & 'C:/Espressif/python_env/idf5.1_py3.11_env/Scripts/python.exe' 'C:\Espressif\frameworks\esp-idf-v5.1.5\tools\idf_monitor.py' -p COM11 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 'd:\Dayro\Knob\build\dayro_knob.elf'
--- Warning: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM11 instead...
--- esp-idf-monitor 1.5.0 on \\.\COM11 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7116
ho 0 tail 12 room 4
load:0x40078000,len:15624
ho 0 tail 12 room 4
load:0x40080400,len:4
--- 0x40080400: _init at ??:?

load:0x40080404,len:3876
entry 0x4008064c
I (33) boot: ESP-IDF v5.1.5-dirty 2nd stage bootloader
I (33) boot: compile time Dec  6 2024 12:38:16
I (33) boot: Multicore bootloader
I (38) boot: chip revision: v3.1
I (42) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 16MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00177000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=3c19ch (246172) map
I (196) esp_image: segment 1: paddr=0004c1c4 vaddr=3ffb0000 size=03e54h ( 15956) load
I (202) esp_image: segment 2: paddr=00050020 vaddr=400d0020 size=dc130h (901424) map
I (528) esp_image: segment 3: paddr=0012c158 vaddr=3ffb3e54 size=00104h (   260) load
I (528) esp_image: segment 4: paddr=0012c264 vaddr=40080000 size=17c5ch ( 97372) load
I (586) boot: Loaded app from partition at offset 0x10000
I (586) boot: Disabling RNG early entropy source...
I (598) cpu_start: Multicore app
I (599) cpu_start: Pro cpu up.
I (599) cpu_start: Starting app cpu, entry point is 0x400814cc
--- 0x400814cc: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v5.1.5/components/esp_system/port/cpu_start.c:159
[... 169 lines cut ...]
MOT: Align sensor.
I (6015) Knob_demo: MQTT_EVENT_SUBSCRIBED, msg_id=2615
I (6019) Knob_demo: sent publish successful, msg_id=0
MOT: sensor_direction==CCW
MOT: PP check: OK!
MOT: No current sense.
MOT: Ready.
intiailized....
Command Received, moving the knob
New Position=0
Target position reached,motor stopped
Current Angle: 0.00°
MOT: Init
MOT: Enable driver.
MOT: Align sensor.
MOT: Skip dir calib.
MOT: Skip offset calib.
MOT: No current sense.
MOT: Ready.

/**********************************************************************************************************************/

Here is the behaviour:

https://github.com/user-attachments/assets/22d0f315-adb1-4b40-bd65-7c463c434468

YanKE01, 2024-12-06:
It looks quite fascinating. Perhaps we could add print statements individually in the angle sensor or driver to investigate further. I usually use the 3PWM driver, so the 6PWM driver was only tested for feasibility with the waveform. I will try prototyping a board to test it out. You could also try investigating and submit a PR.

HirenDayro017, 2024-12-06:
sure @YanKE01 , I will let you know if I find anything useful. Thanks.