espressif/esp-iot-solution #433: Will Touch Proximity Sensor eventually become production ready? (touch_proximity_sensor)

B-l-u-e-f-i-r-e, 2024-11-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Will Touch Proximity Sensor eventually become production ready?

The [Documentation](https://docs.espressif.com/projects/esp-iot-solution/en/latest/touch/touch_proximity_sensor.html) states the following:

ESP32/ESP32-S2/ESP32-S3 touch-related components are intended for testing or demo purposes only. Due to the poor anti-interference capability of the touch functionality, it may not pass EMS testing, and therefore, it is not recommended for mass production products.

Will this statement change eventually after more testing or software fixes, or is the poor anti-interference caused by the Esp32s3 chip's design itself?

leeebo, 2024-11-26:
For ESP32-S3, as the component's readme mentioned "Due to the poor anti-interference capability of the touch functionality, it may not pass EMS testing, and therefore, it is not recommended for mass production products." It is a hardware issue (there is also notice in datasheet), which could not be workaround by software.

For chips like ESP32-P4 and later chips with touch sensor, it would be possiable

B-l-u-e-f-i-r-e, 2024-11-26:
Okay, Thanks

closed 2024-11-26 by B-l-u-e-f-i-r-e